espressif/esp-idf #15703: SPI2 and SPI3 of ESP32-S3 perform differently when driving the ST7789 screen

SuhaylZhao, 2025-04-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using the ESP32-S3 development board to drive a 1.14-inch screen, with an IC of ST7789 and hardware connections: 
CS--GND， 
RST--(ESP32-S3' EN), 
SCK--47, 
SDA--21, 
DC--45.

wanckl, 2025-04-16:
@SuhaylZhao I'm tring the example code on SPI2 and SPI3 with mode3, don't see the additional pulse on CLK pin.

And translate your arduino code to esp-idf, and test on SPI2 and SPI3 with mode3, also no additional pluse found.   
```c
#define G_PIN_NUM_MISO 15
#define G_PIN_NUM_MOSI 21
#define G_PIN_NUM_CLK  47
#define G_PIN_NUM_CS   16

#define G_LCD_HOST  SPI3_HOST   // or SPI3_HOST
TEST_CASE("Test spi2/3 wave", "[spi]")
{
    spi_device_handle_t spi;
    spi_bus_config_t buscfg = {
        .miso_io_num = G_PIN_NUM_MISO,
        .mosi_io_num = G_PIN_NUM_MOSI,
        .sclk_io_num = G_PIN_NUM_CLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
    };

    spi_device_interface_config_t devcfg = {
        .clock_speed_hz = 20000000,
        .mode = 3,
        .spics_io_num = G_PIN_NUM_CS,
        .queue_size = 1,
    };
    ESP_ERROR_CHECK(spi_bus_initialize(G_LCD_HOST, &buscfg, SPI_DMA_CH_AUTO));
    ESP_ERROR_CHECK(spi_bus_add_device(G_LCD_HOST, &devcfg, &spi));

    spi_transaction_t t = {
        .length = 8,
        .tx_data[0] = 0xcc,
        .flags = SPI_TRANS_USE_TXDATA,
    };
    if (ESP_OK != spi_device_transmit(spi, &t)) {
        ESP_LOGE("TAG", "SPI transmit failed");
    }
}
```

Could you clean your actual issue code and paste here

SuhaylZhao, 2025-04-17:
@wanckl Hi, thankns a lot!

As you said, there is indeed no problem with the Arduino code, but the problem lies in the example of LCD driver in the IDF code. 
[tjpgd](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/tjpgd)
[xiaozhi-esp32](https://github.com/78/xiaozhi-esp32)

No need to connect any LCD, just run these two example codes on the chip and monitor the MOSI and SCK pins. You may find that SPI2 and SPI3 are different, and one SPI will send an extra pulse after initialization, causing timing disorder.

I tested six brand new ESP32-S3-WROOM-1-N8R8 modules, and three of them, both SPI2 and SPI3  performed normally. Two modules, SPI2 were normal while SPI3 had extra pulses, and one module, SPI2 had extra pulses while SPI3 was normal.

However, when these 6 modules run Arduino code, all modules, SPI2 and SPI3, perform normally. Therefore, I suspect that there may be a bug in the IDF code. 

After my testing, I found that the bug occurred in the [spi_bus_initialize](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/tjpgd/main/lcd_tjpgd_example_main.c#L111) function. I added `vTaskDelay()` after this function and observed that the extra pulse width became wider, consistent with the delay time.  But I didn't find any code in this function that could cause extra pulses, so I'm very confused. 

Returning to my actual project, I had to modify the pin connections between ESP and LCD, connecting the LCD-RST pin to the GPIO of ESP. After SPI initialization was completed, I reset the SPI communication timing through hardware reset in order to restore normal operation. But I think this is a potential issue that needs to be resolved.  If you need any more relevant information, please contact me. 

Thanks again!

SuhaylZhao, 2025-04-17:
@wanckl I tested your code and the problem was reproduced. ( I just changed the frequency to 1M, so that the data collected by the logic analyzer is more accurate. )
```
#include <driver/spi_master.h>
#include <inttypes.h>
#include <stdio.h>

#include "driver/gpio.h"
#include "esp_chip_info.h"
#include "esp_flash.h"
#include "esp_log.h"
#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h"

#define G_PIN_NUM_MISO 15
#define G_PIN_NUM_MOSI 21
#define G_PIN_NUM_CLK 47
#define G_PIN_NUM_CS 16

#define G_LCD_HOST SPI2_HOST  // or SPI3_HOST
// TEST_CASE("Test spi2/3 wave", "[spi]")
void test_spi_wave(void) {
  spi_device_handle_t spi;
  spi_bus_config_t buscfg = {
      .miso_io_num = G_PIN_NUM_MISO,
      .mosi_io_num = G_PIN_NUM_MOSI,
      .sclk_io_num = G_PIN_NUM_CLK,
      .quadwp_io_num = -1,
      .quadhd_io_num = -1,
  };

  spi_device_interface_config_t devcfg = {
      .clock_speed_hz = 1000000,
      .mode = 3,
      .spics_io_num = G_PIN_NUM_CS,
      .queue_size = 1,
  };
  ESP_ERROR_CHECK(spi_bus_initialize(G_LCD_HOST, &buscfg, SPI_DMA_CH_AUTO));
  ESP_ERROR_CHECK(spi_bus_add_device(G_LCD_HOST, &devcfg, &spi));

  spi_transaction_t t = {
      .length = 8,
      .tx_data[0] = 0xcc,
      .flags = SPI_TRANS_USE_TXDATA,
  };
  if (ESP_OK != spi_device_transmit(spi, &t)) {
    ESP_LOGE("TAG", "SPI transmit failed");
  }
}

void app_main(void) {
  printf("Hello world!\n");
  test_spi_wave();
}
```
When G_LCD_HOST is SPI2_HOST, everything is normal.
[image]

[image]

When G_LCD_HOST is SPI3_SOST, there is an abnormality in the waveform of SCK.

[image]

[image]

Their waveforms are completely consistent during data transmission, but they are different during initialization.

But when I use Arduino code, the waveforms of the two are exactly the same.

wanckl, 2025-04-17:
@SuhaylZhao  I can't build your `xiaozhi-esp32` with esp32s3,
```
xiaozhi-esp32/managed_components/espressif__esp-sr/lib/esp32s3/libc_speech_features.a(esp_speech_features.c.obj): in function `esp_fftr':
/home/sunxiangyu/workspace/esp_sr_lib/components/c_speech_features/esp_speech_features/esp_speech_features.c:294:(.text.esp_fftr+0x31): undefined reference to `dsps_fft4r_fc32_ae32_'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

And the `tpjgd` example using mode 3 only when Octal-SPI, guess you are not using octal-spi. I change to using mode 3, and with your gpio 47/21/15/16, don't see additional pulse on SPI3 clock.   

Here is the first edge of mode 3, clock switch from `0`->`1`   

[image]

SuhaylZhao, 2025-04-17:
@wanckl Yes, I didn't use Octal-SPI.  Strangely, Even if using your code, some of the modules I tested were normal, while others were abnormal.  But these modules all work properly when running Arduino code. These modules come from the official Espressif and are brand new, so I don't think there is a hardware problem.

wanckl, 2025-04-17:
@SuhaylZhao Okey, you say you can reproduce with my code on part of chips, I should try more chips

SuhaylZhao, 2025-04-18:
@wanckl Hi, Yes, they have the same revision number. All of them are 
```
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
```

wanckl, 2025-04-18:
@SuhaylZhao  I have a patch here on `master` branch, can you help trying if it shuting your issue or not?

[spi_fix_gpio_cfg.patch.txt](https://github.com/user-attachments/files/19810431/spi_fix_gpio_cfg.patch.txt)

SuhaylZhao, 2025-04-18:
@wanckl Hi, I encountered an error while applying this patch file.

[image]

wanckl, 2025-04-18:
@SuhaylZhao Which commit you are working at? just tried on current `master` [d930a38](https://github.com/espressif/esp-idf/commit/d930a386dae78cfab75f313af3df67921e748fc4) it is OK

SuhaylZhao, 2025-04-19:
@wanckl  
After applying your patch, I retested the your test [code](https://github.com/espressif/esp-idf/issues/15703#issuecomment-2809081124) . When using SPI3, the extra pulse disappeared, but its waveform was still different from SPI2, as shown in the following figure: (The waveform comparison in each image has the same time division)

[image]

[image]

Next, I tested the [tjpgd](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/tjpgd) code again. Although the waveform has changed, SPI2 is still usable and SPI3 is not.

[image]

wanckl, 2025-04-21:
@SuhaylZhao Thank you for detailing test, But for now the different wave after patched seems shouldn't change anything, I don't have idea why LCD still don't work on SPI3,

Can you help seeing if anything else different, BTW, you can incrase logic analyser sample rate to grab smaller pulse.   
And from the only different we can see now, can you check dose your LCD has requirement on SCK edge time refer to the MOSI line?

Thanks

SuhaylZhao, 2025-04-21:
@wanckl 
Thanks for your help!
It's really puzzling! 
The waveform of SPI2 and SPI3, even though they appear very close, the performance of the LCD is still the same as before. Some modules can work normally using SPI2, but not using SPI3. Some other modules can work normally using SPI3, while SPI2 does not. There are also some SPI2 and SPI3 that can work normally, and their codes are completely identical. The LCD device is also the same, only the module has been replaced. 
Although I have solved the problem of the screen not being able to be used under SPI3 through other methods, I still really want to figure it out and solve this problem. So far, I don't know what to do. If you are also interested, I can send you my device and LCD.

eriksl, 2026-03-25:
My $0.02.

I am running SPI LCD's on both SPI2 and SPI3, no problems there, none. The LCD's I am using various, the controllers being from both well-known manufacturers and the RA8875, which works completely differently, but also talks SPI.

Two notes:
- I am not using any of the ESP-IDF LCD "driver" code. IMHO controlling a SPI LCD controller is not a "driver", it does not control hardware using registers etc., it only "drives" the SPI code. So I prefer to maintain that code myself.
- the SPI2 controller can bypass the IO-MUX if the IO pins are chosen well (default). The SPI3 controller can't. The IO-MUX introduces a small latency which causes (according to the documentation) it not being able to run at 80 MHz, 40 MHz max. Maybe that's the cause of your issue?

And yes, I am using 40 MHz only. Because most controllers can't handle 80 Mhz (the first available higher rate) anyway.