espressif/esp-idf #15704: ESP32S3 Wifi Driver hangs on reconnect

ShawnLord-Tovala, 2025-04-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-s3 (QFN52) (revision v0.2) 

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

WROOM-1 

### Power Supply used.

External 3.3V

### What is the expected behavior?

I expect wifi component to connect to an available access point on access point returning.

### What is the actual behavior?

I am seeing the wifi component fail to connect to an access point in two different manners:

1.  After a successful call to `esp_wifi_connect();`  No error is returned, and there are no events posted to the `WIFI_EVENT` base of the default event handler.  looking at logs, the wifi component seems stuck in a loop after `profile match:`
`[0;32mI (521496) 🛜 TOVEN_WIFI: Connecting with credentials: ssid=`LordyDev`, pwd=`microwave`[0m
D (521446) wifi:connect status 6 -> 0
D (521506) wifi:Start wifi connect
D (521506) wifi:connect status 0 -> 0
D (521516) wifi:connect chan=0
D (521516) wifi:first chan=1
D (521516) wifi:connect status 0 -> 1
D (521516) wifi:filter: set rx policy=3
D (521526) wifi:clear scan ap list
D (521526) wifi:start scan: type=0x50f, priority=2, cb=0x42063228, arg=0x0, ss_state=0x1, time=521531030, index=0
D (521546) wifi:perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
D (521546) event: running post TOVEN_WIFI_EVENT:1 with handler 0x42010024 and context 0x3fcce218 on loop 0x3fcc9780[0m
D (521546) wifi:rsn valid: gcipher=1 ucipher=3 akm=8 mac=00:c0:ca:b0:97:cc
D (521556) wifi:profile match: ss_state=0x7
D (521626) wifi:rsn valid: gcipher=1 ucipher=3 akm=8 mac=00:c0:ca:b0:97:cc
D (521626) wifi:profile match: ss_state=0x7
D (521726) wifi:rsn valid: gcipher=1 ucipher=3 akm=8 mac=00:c0:ca:b0:97:cc
D (521726) wifi:profile match: ss_state=0x7
D (521826) wifi:rsn valid: gcipher=1 ucipher=3 akm=8 mac=00:c0:ca:b0:97:cc
D (521826) wifi:profile match: ss_state=0x7`

2. After a disconnect, the access point returns, is seen but the wifi continues to fail due to auth expire:
'[0;32mI (1381685) NETWORK_MANAGER: WiFi connected[0m
[0;32mI (1385055) NETWORK_MANAGER: Time set to 1742926049.961498: CPU time=1384184, wall clock=1742926049[0m
D (1438785) wifi:recv unprotected deauth, reason=0x3
D (1453845) wifi:recv unprotected deauth, reason=0x7
[0;32mI (1453845) wifi:starting SA query procedure with AP(00:c0:ca:b0:97:cc)[0m
[... 166 lines cut ...]
D (1456505) wifi:cancel previous scan, ss_state=0x3, just_scan=0
D (1456505) wifi:scan end: arg=0x3c111966, status=4, ss_state=0x0
D (1456505) wifi:stop scan timer and scan, ss_state=0x0
D (1456515) wifi:filter: set rx policy=4
D (1456515) wifi:first chan=1
D (1456515) wifi:handoff_cb: status=1
D (1456525) wifi:clear scan ap list
D (1456525) wifi:filter: set rx policy=8
[0;32mI (1456525) 🛜 TOVEN_WIFI: Did disconnect from WiFi[0m
[0;32mI (1456535) 🛜 TOVEN_WIFI: Connecting with credentials: ssid=`LordyDev`, pwd=`microwave`[0`

### Steps to reproduce.

1. With esp in station mode, connect to access point on router.
2. Cycle router access point down using `ifdown` followed by `ifup` after 15sec,  repeating  every 60 

Failure is typically seen within 10 to 20 cycles.

NOTE:  Failure is only seen in project code.  A duplicate setup running example code only on a bare example board with unlimited retries continues to restart and reconnect appropriately.

### Debug Logs.

```plain

```

Xiehanxin, 2025-04-03:
hi @ShawnLord-Tovala could you describe your application steps? like connecting to the  wifi ,then disconnect immediatly

ShawnLord-Tovala, 2025-04-04:
The application starts the wifi in station mode and then calls connect to begin the connection. On the event `WIFI_EVENT_STA_DISCONNECTED`, esp_wifi_connect(); is called to restart the connection.  Currently for this test there is no backoff or other added logic to the retry mechanism.  The IP_EVENT_STA_GOT_IP, triggers a call to netif to trigger an sntp start.  There are other threads running on both cores but none of them are tied to wifi or networking.

As I posted above I am running on the released 5.4 branch.  I will try tomorrow with the top of main.

ShawnLord-Tovala, 2025-04-07:
@Xiehanxin 
I noticed in debugging today that the wifi will hang mid station scan after a call to: esp_wifi_connect().  What is happening in the wifi code after the wifi:start max timer??  Is it waiting for a callback from ESP_TIMER?

`D (21637180) wifi:Start wifi connect
D (21637180) wifi:connect status 3 -> 0
D (21637180) wifi:connect chan=0
D (21637190) wifi:first chan=13
D (21637190) wifi:connect status 0 -> 1
D (21637190) wifi:filter: set rx policy=3
D (21637190) wifi:clear scan ap list
D (21637200) wifi:start scan: type=0x50f, priority=2, cb=0x4205e170, arg=0x0, ss_state=0x1, time=162366308, index=0
V (21637200) wifi:inter_channel_timeout: arg=0x0, ss_state=0x1
D (21637220) wifi:perform scan: ss_state=0x9, chan<13,0>, dur<360,360>
V (21637220) wifi:change: chan<13,0>, dur<360,360>
V (21637230) wifi:enter start op, arg=0x3c102c54
V (21637230) wifi:scan operation: state=0x3, chan<13,0>, arg=0x3c102c54, status=0
V (21637240) wifi:passive scan, listen only
V (21637240) wifi:start max timer
[0;32mI (21637240) 🛜 TOVEN_WIFI: WiFi connect started[0m
D (21637600) wifi:scan end: arg=0x3c102c54, status=0, ss_state=0x3
D (21637600) wifi:perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
V (21637610) wifi:change: chan<1,0>, dur<0,120>
V (21637620) wifi:ht20 freq=2412, chan=1
V (21637620) wifi:enter start op, arg=0x3c102c54
V (21637620) wifi:scan operation: state=0x3, chan<1,0>, arg=0x3c102c54, status=0
V (21637630) wifi:scan specific ssid=LordyDev1
V (21637630) wifi:start max timer
D (21637750) wifi:scan end: arg=0x3c102c54, status=0, ss_state=0x3
D (21637750) wifi:perform scan: ss_state=0x9, chan<2,0>, dur<0,120>
V (21637760) wifi:change: chan<2,0>, dur<0,120>
V (21637770) wifi:ht20 freq=2417, chan=2
V (21637770) wifi:enter start op, arg=0x3c102c54
V (21637770) wifi:scan operation: state=0x3, chan<2,0>, arg=0x3c102c54, status=0
V (21637780) wifi:scan specific ssid=LordyDev1
V (21637780) wifi:start max timer
D (21637900) wifi:scan end: arg=0x3c102c54, status=0, ss_state=0x3
D (21637900) wifi:perform scan: ss_state=0x9, chan<3,0>, dur<0,120>
V (21637910) wifi:change: chan<3,0>, dur<0,120>
V (21637920) wifi:ht20 freq=2422, chan=3
V (21637920) wifi:enter start op, arg=0x3c102c54
V (21637920) wifi:scan operation: state=0x3, chan<3,0>, arg=0x3c102c54, status=0
V (21637930) wifi:scan specific ssid=LordyDev1
V (21637930) wifi:start max timer
D (21638050) wifi:scan end: arg=0x3c102c54, status=0, ss_state=0x3
D (21638050) wifi:perform scan: ss_state=0x9, chan<4,0>, dur<0,120>
V (21638060) wifi:change: chan<4,0>, dur<0,120>
V (21638070) wifi:ht20 freq=2427, chan=4
V (21638070) wifi:enter start op, arg=0x3c102c54
V (21638070) wifi:scan operation: state=0x3, chan<4,0>, arg=0x3c102c54, status=0
V (21638080) wifi:scan specific ssid=LordyDev1
V (21638080) wifi:start max timer  <<-----   !!!!!!!!!!!!!!!!!!!!!!!!!!WIFI IS STUCK HERE !!!!!!!!!!!!!!!!!!!!!!!!!!!
D (21676060) event: running post IP_EVENT:1 with handler 0x42012d2c and context 0x3fcd49d4 on loop 0x3fcc9454
[0;32mI (21676060) 🛜 TOVEN_WIFI: Got a WiFi event: IP_EVENT/1 - WifiState=5![0m
[0;32mI (21676070) 🛜 TOVEN_WIFI: Lost IP address[0m`

mhdong, 2025-04-10:
> triggers a call to netif to trigger an sntp start. There are other threads running on both cores but none of them are tied to wifi or networking.

hi @ShawnLord-Tovala 
D (1456485) wifi:Start wifi disconnect
When is esp_wifi_disconnect called? Could you please explain the main logic at the application layer? If possible, could you also provide a reproducible demo so that we can try to reproduce the issue locally?

ShawnLord-Tovala, 2025-04-24:
Solution:

After some investigating I found two issues in my code base:
1. In the wifi driver calling esp_wifi_start / disconnect from multiple threads / handlers.  I was able to get wifi in an unhappy state.
2. Possibly the larger issue was I was runnign the ESP_Timer in ISR mode from core 1, with wifi and my app_main running on core 0.  I moved my core 1 tasks to a different timer and left ESP Timer to run in task mode on core 0.  This seemed to make the wifi component rock solid.

It is still not understood why a deInit and reInit of the wifi driver does not restore it to a working condition.

closed 2025-04-24 by ShawnLord-Tovala