espressif/esp-idf #15706: esp-idf v5.4 wifi examples not working with xiao esp32c6

Preetam-Das, 2025-04-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I tried both `examples/wifi/getting_started/station` and `examples/wifi/getting_started/softAP` with seed studio's xiao esp32c6 board but it is not working. I attatched the logs. 

```
> idf.py --version
ESP-IDF v5.4
```

[debuglog.txt](https://github.com/user-attachments/files/19565429/debuglog.txt)

[log.txt](https://github.com/user-attachments/files/19565432/log.txt)

nachiketkukade, 2025-04-10:
@Preetam-Das , pls share the sdkconfig used for these builds. Also is there a ESP-IDF version that DOES work this board?

Preetam-Das, 2025-04-12:
Hi @nachiketkukade the official docs for the board uses v5.2.1 but the example is not for wifi so ig it wont work either https://wiki.seeedstudio.com/xiao_idf/#step-3-get-esp-idf. I haven't checked any other versions ... problem is with wifi examples only .. simple blink example works. 

The attatched logs are for the station example https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station

[debuglog.txt](https://github.com/user-attachments/files/19718967/debuglog.txt)

[sdkconfig.txt](https://github.com/user-attachments/files/19718968/sdkconfig.txt)

AkshatAgrawl, 2025-05-07:
Hi @preetamsad  
I noticed that `IDF_TOOLCHAIN=clang` is set in the `sdkconfig`. Just wanted to confirm if this was configured intentionally, as it could potentially be related to the issue.

AkshatAgrawl, 2025-05-07:
@preetamsad Yes. Switching to gcc should work.

Camacazi, 2025-09-10:
Thank you for the reply, I can confirm that disabling the compiler optimisations fixed the issue. 
I'm aware clang is not recommended for use currently but I'd love to have this fixed in the future.

closed 2025-09-22 by nachiketkukade