espressif/esp-idf #15707: ESP32C3 cannot disable UART log over USB port

florianhumblot, 2025-04-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.5-dev-2642-g50be9735b3

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

When disabling console output on the USB/JTAG port, I expect there to not be any output on that port.

### What is the actual behavior?

Logs from ESP-IDF still appear in my console.

### Steps to reproduce.

1. Build the usb_serial_jtag_echo example using the esp-idf docker image
2. Flash it on an ESP32C3
3. Type something in the USB-Serial port

### Debug Logs.

```plain
Logs from the UART0 port:

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
Valid secure boot key blocks: 0
secure boot verification succeeded
load:0x3fcd5990,len:0x2ef8
load:0x403cbf10,len:0xc10
--- 0x403cbf10: esp_bootloader_get_description at /COMPONENT_ESP_BOOTLOADER_FORMAT_DIR/esp_bootloader_desc.c:39 (discriminator 1)

load:0x403ce710,len:0x55a4
--- 0x403ce710: esp_flash_encryption_enabled at /COMPONENT_BOOTLOADER_DIR_support/src/flash_encrypt.c:89

entry 0x403cbf1a
I (74) boot: ESP-IDF v5.5-dev-2642-g50be9735b3 2nd stage bootloader
--- 0x403cbf1a: call_start_cpu0 at /COMPONENT_MAIN_DIR/bootloader_start.c:25

I (74) boot: chip revision: v0.4
I (74) boot: efuse block revision: v1.3
I (75) boot.esp32c3: SPI Speed      : 80MHz
I (79) boot.esp32c3: SPI Mode       : DIO
I (83) boot.esp32c3: SPI Flash Size : 2MB
I (87) boot: Enabling RNG early entropy source...
I (91) boot: Partition Table:
I (94) boot: ## Label            Usage          Type ST Offset   Length
I (100) boot:  0 nvs              WiFi data        01 02 0000d000 00006000
I (107) boot:  1 phy_init         RF data          01 01 00013000 00001000
I (113) boot:  2 factory          factory app      00 00 00020000 00100000
I (120) boot: End of partition table
I (123) esp_image: segment 0: paddr=00020020 vaddr=3c020020 size=07fc8h ( 32712) map
I (136) esp_image: segment 1: paddr=00027ff0 vaddr=3fc8ca00 size=01258h (  4696) load
I (144) esp_image: segment 2: paddr=00029250 vaddr=40380000 size=06dc8h ( 28104) load
I (151) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=13cdch ( 81116) map
I (166) esp_image: segment 4: paddr=00043d04 vaddr=40386dc8 size=05b30h ( 23344) load
I (171) esp_image: segment 5: paddr=0004983c vaddr=50000000 size=0001ch (    28) load
I (171) esp_image: segment 6: paddr=00049860 vaddr=00000000 size=06770h ( 26480)
I (181) esp_image: Verifying image signature...
I (182) secure_boot_v2: Verifying with RSA-PSS...
I (189) secure_boot_v2: Signature verified successfully!
I (195) boot: Loaded app from partition at offset 0x20000
I (196) secure_boot_v2: enabling secure boot v2...
I (200) secure_boot_v2: secure boot v2 is already enabled, continuing..
I (207) boot: Disabling RNG early entropy source...
I (223) cpu_start: Unicore app
I (238) cpu_start: Pro cpu start user code
I (238) cpu_start: cpu freq: 160000000 Hz
I (238) app_init: Application information:
I (238) app_init: Project name:     usb_serial_jtag_echo
I (243) app_init: App version:      v5.5-dev-2642-g50be9735b3
I (248) app_init: ELF file SHA256:  85bb3eaec...
I (253) app_init: ESP-IDF:          v5.5-dev-2642-g50be9735b3
I (258) efuse_init: Min chip rev:     v0.3
I (262) efuse_init: Max chip rev:     v1.99
I (266) efuse_init: Chip rev:         v0.4
I (270) heap_init: Initializing. RAM available for dynamic allocation:
I (276) heap_init: At 3FC8EC00 len 00031400 (197 KiB): RAM
I (281) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (287) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (293) heap_init: At 5000001C len 00001FCC (7 KiB): RTCRAM
I (299) spi_flash: detected chip: generic
I (302) spi_flash: flash io: dio
W (305) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (318) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (324) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (331) main_task: Started on CPU0
I (331) main_task: Calling app_main()
I (331) usb_serial_jtag echo: USB_SERIAL_JTAG init done
I (341) main_task: Returned from app_main()
I (4511) Recv str: : 0x3fc92700   74                                                |t|
I (4951) Recv str: : 0x3fc92700   65                                                |e|
I (6271) Recv str: : 0x3fc92700   73                                                |s|
I (6661) Recv str: : 0x3fc92700   74                                                |t|

Logs from the USB port:

tropy source...
tI (4511) Recv str: : 0x3fc92700   74                                                |t|
eI (4951) Recv str: : 0x3fc92700   65                                                |e|
sI (6271) Recv str: : 0x3fc92700   73                                                |s|
tI (6661) Recv str: : 0x3fc92700   74                                                |t|
```

### Diagnostic report archive.

[idf-diag-4774a809-d1b5-44d1-a21a-737a75603b69.zip](https://github.com/user-attachments/files/19567061/idf-diag-4774a809-d1b5-44d1-a21a-737a75603b69.zip)

florianhumblot, 2025-04-02:
Might somehow relate to #15343 
I'm extremely confused, this used to work just fine a few weeks ago apart from the extra log once at startup, but now, no matter what I do, I can't get the log to stop on that port.

florianhumblot, 2025-04-02:
It looks like the problem is this config: `CONFIG_VFS_SUPPORT_IO=n`
To save some code size I disabled Virtual File System IO functions. Re-enabling the IO functions makes the `ESP_CONSOLE_USB_SERIAL_JTAG_ENABLED` setting be respected again.
Maybe this is more of a documentation issue or maybe menuconfig should know about this dependency.

SoucheSouche, 2025-04-04:
Hi @florianhumblot, could you maybe be more clear as to what is your issue ?

I can see in the sdkconfig of your project that `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG` and `CONFIG_ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG` are not set which means that when using the console, the only port used for input output will be the default UART one (since it is the one enabled in your sdlkconfig and no secondary output is set).

The usb_serial_jtag_echo example is initializing and using the usb serial jtag driver directly, so the console component is not involved in the example. So regardless of the state of ESP_CONSOLE_USB_SERIAL_JTAG_ENABLED, you will see output on the usb serial jtag when running the example.

florianhumblot, 2025-04-04:
@SoucheSouche basically, if you disable VFS IO functions in menuconfig, the ESP32-C3 will always output its log over **both** the UART0 port AND the USB/JTAG serial port, regardless of the `ESP_CONSOLE_USB_SERIAL_JTAG` and `ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG` settings.

SoucheSouche, 2025-04-07:
@florianhumblot, thanks for the clarification. I was confused as to why your are using the usb_serial_jtag_echo example to reproduce an issue that concerns the console only it seems.

I will try to reproduce it locally. In the mean time could you share with me the pins you are using for the UART and USB Serial Jtag?

florianhumblot, 2025-04-07:
@SoucheSouche ah right, I was using that example because that functionality breaks for me when the log still outputs over USB :P

The pins I use for UART0 are RXD0 and TXD0 (pins 30 and 31 from ESP32-C3-MINI-1-H4) and the pins I use for USB are IO18 and IO19 (pins 26 and 27 from ESP32-C3-MINI-1-H4).

SoucheSouche, 2025-04-07:
Thanks, I'll have a look at it this week :)

florianhumblot, 2025-04-10:
@SoucheSouche 

[image]

On the left is the UART0 port, connected with a FTDI cable to my PC, over which I flashed the software.
On the right is tera term, connected to the USB port of the ESP32-C3.

SoucheSouche, 2025-04-10:
@florianhumblot,  Now I am really confused. You don't want anything printed on the USB serial output but the screenshot you sent are from the usb_serial_jtag_echo example which only purpose is to enable the USB serial driver and echo what you type on it.

florianhumblot, 2025-04-10:
@SoucheSouche I want to output my own things over the usb serial port but I don't want the esp's logs to be interspersed through that.
In my screenshots you can see that apart from echoing the input, the esp's log is also output on the usb serial port despite setting the secondary output console to none.

SoucheSouche, 2025-04-11:
@florianhumblot, I started looking into it more carefully and it is not easy to see what is going wrong. I noticed the issue does not show on some targets like esp32c6. I will continue investigating. I will report the progress next week.

SoucheSouche, 2025-05-05:
Hi @florianhumblot, It seems that I cannot reproduce the issue with usb_serial_jtag_echo on esp32c3 or other targets for that matter.

[image]

I tried several commits on several releases but couldn't manage to reproduce.

Just to make sure, you have the CONFIG_VFS_SUPPORT_IO enabled right ?