espressif/esp-idf #15708: Wi-Fi Aware (NAN): Service Specific Information (SSI) is not included in transmitted packets

prftgtwy, 2025-04-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1-156-gde92b624e6

### Espressif SoC revision.

ESP32-WROOM-32U

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-WROVER-KIT 3.3V

### Power Supply used.

USB

### What is the expected behavior?

The NAN Service Discovery frames transmitted by the ESP32 should include the Service Descriptor Attribute where:

- The "Service Info Present" bit in the Service Control field is set to True (1).
- The defined SSI with the custom payload like "AAAAAA" appears as the Service Info field within the Service Descriptor Attribute after the Service Control Type field

### What is the actual behavior?

The NAN Service Discovery frames transmitted by the ESP32 show the Service Descriptor Attribute with:

- The "Service Info Present" bit in the Service Control field set to False (0).
- No Service Info field is present in the attribute payload.

The captured packet looks identical to the one transmitted when no SSI is configured in `publish_cfg`. It appears the `ssi` and `ssi_len` members of the `wifi_nan_publish_cfg_t` struct are being ignored by the underlying NAN publish mechanism. An attempt to use the older `svc_info` field also did not result in the information being transmitted.

### Steps to reproduce.

1. **Obtain Example**: Check out the `examples/wifi/wifi_aware/nan_publisher` example from the specified ESP-IDF version.
2. Cherry-Pick commit `302d8e5fce87de9e8a59bd6890b21eda05967b41` - `feat(esp_wifi): Add extended SSI support for NAN Services
` into current ESP-IDF version. ([https://github.com/espressif/esp-idf/commit/302d8e5fce87de9e8a59bd6890b21eda05967b41](https://github.com/espressif/esp-idf/commit/302d8e5fce87de9e8a59bd6890b21eda05967b41))
3. **Modify Code**: Apply the following code modifications to `publisher_main.c`:

```
...
...

/* Publish a service */

uint8_t pub_id;
wifi_nan_publish_cfg_t publish_cfg = {
    .service_name = "TestService",
#if CONFIG_EXAMPLE_NAN_PUBLISH_UNSOLICITED
    .type = NAN_PUBLISH_UNSOLICITED,
#else
    .type = NAN_PUBLISH_SOLICITED,
[... 107 lines cut ...]
I (548) wifi:Init dynamic rx buffer num: 32
I (558) wifi_init: rx ba win: 6
I (558) wifi_init: accept mbox: 6
I (558) wifi_init: tcpip mbox: 32
I (568) wifi_init: udp mbox: 6
I (568) wifi_init: tcp mbox: 6
I (568) wifi_init: tcp tx win: 5760
I (578) wifi_init: tcp rx win: 5760
I (578) wifi_init: tcp mss: 1440
I (578) wifi_init: WiFi IRAM OP enabled
I (588) wifi_init: WiFi RX IRAM OP enabled
I (588) phy_init: phy_version 4860,6b7a6e5,Feb  6 2025,14:47:07
I (668) wifi:mode : NAN (0a:d1:f9:0d:b6:bc)
I (3668) NAN: Join Cluster with Id 50:6f:9a:01:ac:bd
I (3678) nan_app: NAN Discovery started.
I (3678) nan_app: Started Publishing TestService [Service ID - 5]
```

### More Information.

- Tested with both NAN_PUBLISH_UNSOLICITED and NAN_PUBLISH_SOLICITED types (CONFIG_EXAMPLE_NAN_PUBLISH_UNSOLICITED); the behavior is the same.
- Confirmed via logging (ESP_LOG_BUFFER_HEXDUMP) that memory for publish_cfg.ssi is successfully allocated and populated before calling esp_wifi_nan_publish_service.
- Crucially, the attached Wireshark screenshot shows the captured Service Descriptor Attribute where "Service Info Present" is incorrectly marked as False and the SSI is not present in the packet.

[image]

prftgtwy, 2025-04-04:
@nachiketkukade - did you experience something similar?

johndoesenior, 2025-04-04:
Joining the thread, had the same issue, thought I did something wrong.

nachiketkukade, 2025-04-08:
@prftgtwy , this extended SSI support for WiFi Aware is not merged in v5.4 yet (it'll be in v5.4.2). You cherry-picked the IDF commit, but there are changes in the wifi libraries as well which can't be cherry picked. Can you please confirm if your code works on master?

@johndoesenior , are you also testing this on IDF v5.4.1?

prftgtwy, 2025-04-10:
I cherry picked the relevant commit (https://github.com/espressif/esp32-wifi-lib/commit/779ab6d361654eaec11200001467b3fd55a8b28e) for the wifi libs as well and then the SSI worked as expected - thanks.
Closing this issue.

closed 2025-04-10 by prftgtwy

nachiketkukade, 2025-04-10:
@prftgtwy  Thanks for the update. Please note that once v5.4 starts to diverge cherry-picking wifi libs will lead to compilation failures. But soon the back port of this change will be available, so you won't have to cherry-pick