espressif/esp-idf #15709: Failed to resolve component 'linux' required by component 'src': component not registered.

yanmartins, 2025-04-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Build the program

### What is the actual behavior?

I got:

```
NOTICE: [1/1] idf (5.4.1)
CMake Error at C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/build.cmake:323 (message):
  Failed to resolve component 'linux' required by component 'src': component
  not registered.
Call Stack (most recent call first):
  C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/build.cmake:361 (__build_resolve_and_add_req)
  C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/build.cmake:665 (__build_expand_requirements)
  C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/project.cmake:717 (idf_build_process)
  CMakeLists.txt:22 (project)

-- Configuring incomplete, errors occurred!
```

And also:

```
NOTICE: Processing 1 dependencies:
NOTICE: [1/1] idf (5.4.1)
CMake Error at C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/build.cmake:323 (message):
  Failed to resolve component 'linux' required by component 'src': component
  not registered.
Call Stack (most recent call first):
  C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/build.cmake:361 (__build_resolve_and_add_req)
  C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/build.cmake:665 (__build_expand_requirements)
  C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/project.cmake:717 (idf_build_process)        
  CMakeLists.txt:22 (project)

-- Configuring incomplete, errors occurred!
HINT: The component 'linux' could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager, the component has been removed and refactored into some other component or the component may not 
be supported by the selected target.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.
```

### Steps to reproduce.

1. Try to build
...

### Debug Logs.

```plain

```

dobairoland, 2025-04-03:
@yanmartins Please share the project's and src's `CMakeList.txt`.

fhrbata, 2025-04-03:
Hello @yanmartins,  the `linux` component is one of the [common components](https://github.com/espressif/esp-idf/blob/master/tools/cmake/build.cmake#L272)  added automatically as requirement to every component in the `esp-idf` build if the target is set to `linux`.  This might be related to a [change](https://github.com/espressif/esp-idf/commit/fbe8bf89ee4431c72631bc227b) that fixed component resolution in the `esp-idf` build system. It looks like the `src` component explicitly requires the `linux` component, which should not be necessary and the build target is not set to `linux`. In that case the `linux` component is [not resolved](https://github.com/espressif/esp-idf/blob/master/components/linux/CMakeLists.txt#L3) and registered in the build system and this error is reported. It might work in `5.3.1` where this was silently ignored, but this was fixed in `5.4` with the above mentioned commit. This is also seen in the hint message

> ... the component may not be supported by the selected target.

Thank you

yanmartins, 2025-04-03:
I try to set a new target, but I got the same error. Is there any way I could workaround this issue?

```
Failed to set target esp32s3: non zero exit code 2

Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'C:\dynagateway-v2-fw\dyna_app\dynagateway-v2\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory C:\dynagateway-v2-fw\dyna_app\dynagateway-v2\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=c:\Users\yan.martins\5.4.1\.espressif\python_env\idf5.4_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 C:\dynagateway-v2-fw\dyna_app\dynagateway-v2"...
-- Existing sdkconfig 'C:/dynagateway-v2-fw/dyna_app/dynagateway-v2/sdkconfig' renamed to 'C:/dynagateway-v2-fw/dyna_app/dynagateway-v2/sdkconfig.old'.
-- Found Git: C:/Users/yan.martins/5.4.1/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/yan.martins/5.4.1/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/yan.martins/5.4.1/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/yan.martins/5.4.1/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
NOTICE: Processing 1 dependencies:

NOTICE: [1/1] idf (5.4.1)

CMake Error at C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/build.cmake:323 (message):
  Failed to resolve component 'linux' required by component 'src': component
  not registered.
Call Stack (most recent call first):
  C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/build.cmake:361 (__build_resolve_and_add_req)
  C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/build.cmake:665 (__build_expand_requirements)
  C:/Users/yan.martins/5.4.1/esp/v5.4.1/esp-idf/tools/cmake/project.cmake:717 (idf_build_process)
  CMakeLists.txt:21 (project)

-- Configuring incomplete, errors occurred!
[0;33mHINT: The component 'linux' could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager, the component has been removed and refactored into some other component or the component may not be supported by the selected target.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.[0m
cmake failed with exit code 1, output of the command is in the C:\dynagateway-v2-fw\dyna_app\dynagateway-v2\build\log\idf_py_stderr_output_19356 and C:\dynagateway-v2-fw\dyna_app\dynagateway-v2\build\log\idf_py_stdout_output_19356
```

fhrbata, 2025-04-03:
Hello @yanmartins ,

as pointed by @dobairoland , could you please share the `CMakeLists.txt` for the `src` component? Does it contain `linux` component as a requirement? 

Thank you

yanmartins, 2025-04-03:
Hello @fhrbata and @dobairoland

This is my  `CMakeLists.txt` for the src component?

````
# fake_name_app
set(FAKE_NAME_COMPONENT_SRCDIRS
    "../../../fake_name_hal/src_esp32"
	"../../../fake_name_drv/src"
	"../../../fake_name_api/src"
    "../../../fake_name_api/ble/src"
    "../../../fake_name_api/fake_namelogger/src"
	"../../../fake_name_srv/src"
    "../../../fake_name_srv/network/src"
    "../../../fake_name_srv/protocol/src"
    "../../../fake_name_srv/storage/src"
    "../../../fake_name_sandbox/src"
    "../comm/src"
	"../webserver/endpoints/src"
    "../webserver/src"
    "."
)

set(FAKE_NAME_COMPONENT_INCDIRS
	"../../../fake_name_hal/inc_esp32"
    "../../../fake_name_drv/inc"
	"../../../fake_name_api/inc"
    "../../../fake_name_api/ble/inc"
    "../../../fake_name_api/fake_namelogger/inc"
    "../../../fake_name_api/fake_namelogger/actions/inc"
	"../../../fake_name_srv/inc"
    "../../../fake_name_srv/network/inc"
    "../../../fake_name_srv/protocol/inc"
    "../../../fake_name_srv/storage/inc"
    "../../../fake_name_sandbox/inc"
    "../comm/inc"
    "../webserver/endpoints/inc"
    "../webserver/inc"
    "../inc"
)

set(IDF_UNUSED_COMPONENTS
    ".vscode"
)

# Add all esp-idf components to build process
file(GLOB component_dirs ${idf_path}/components/*)
foreach(component_dir ${component_dirs})
    # Get only directories and ignore other files
    if(IS_DIRECTORY ${component_dir})
        get_filename_component(component_dir ${component_dir} NAME)
        string(FIND ${component_dir} ${IDF_UNUSED_COMPONENTS} result)
        # Ignore some components defined above
        if(result EQUAL -1)
            # message(STATUS ${component_dir})
            list(APPEND FAKE_NAME_COMPONENT_REQUIRES ${component_dir})
        endif()
    endif()
endforeach()

# Add other components

list(APPEND FAKE_NAME_COMPONENT_REQUIRES led_strip Device-Shadow-for-AWS-IoT-embedded-sdk Jobs-for-AWS-IoT-embedded-sdk coreJSON coreMQTT)

set(FAKE_NAME_COMPONENT_EMBED_FILES
    "../webserver/front-end/index.html"
    "../webserver/front-end/favicon.ico.gz"
    "../webserver/front-end/assets/index.js.gz"
) 

idf_component_register(
    SRC_DIRS ${FAKE_NAME_COMPONENT_SRCDIRS}
    INCLUDE_DIRS ${FAKE_NAME_COMPONENT_INCDIRS}
    REQUIRES ${FAKE_NAME_COMPONENT_REQUIRES}
    EMBED_FILES ${FAKE_NAME_COMPONENT_EMBED_FILES}
)

idf_build_get_property(project_dir PROJECT_DIR)

create_esp_enc_img(${CMAKE_BINARY_DIR}/${CMAKE_PROJECT_NAME}.bin
    ${project_dir}/resources/security/pre-encryption_rsa_key.pem ${CMAKE_BINARY_DIR}/${CMAKE_PROJECT_NAME}_secure.bin app)
```

fhrbata, 2025-04-03:
Hello @yanmartins ,

you don't have to search for the `esp-idf` components manually. This should be handled correctly by the `esp-idf` build system. Note that by default, if you don't use `set(COMPONENTS ...)` or `idf_build_set_property(MINIMAL_BUILD ON)` in your project's `CMakeLists.txt`, all `esp-idf` components are added to the build. The build system documentation is [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html).

The problem is that with your approach you are adding the `linux` component explicitly to the build. This component should be used only with `linux` target`, set by `idf.py --preview set-target linux`, but you don't need this component for your `esp32s3` project. You can add it to your `IDF_UNUSED_COMPONENTS` list, but I would recommend not to manually search for `esp-idf` components. Is there any reason why it's needed for your use case? I believe you should be able to drop the whole component search in the `GLOB component_dirs ${idf_path}/components/*` and just use components in your list `list(APPEND FAKE_NAME_COMPONENT_REQUIRES esp_modem mdns esp_encrypted_img led_strip Device-Shadow-for-AWS-IoT-embedded-sdk Jobs-for-AWS-IoT-embedded-sdk coreJSON coreMQTT cbor)` in the `REQUIRES` parameter for `idf_component_register`. The build system should handle the rest. If some component will be missing you should see a hint message suggesting what component is missing and add it to your list.

Thank you

yanmartins, 2025-04-03:
Hello @fhrbata,

It's legacy code 😅.

I did what you suggested: I removed all the `GLOB component_dirs ${idf_path}/components/*` and added each component manually in the` REQUIRES`

Thank you

fhrbata, 2025-04-03:
Hello @yanmartins , great that you were able to figure it out and it's working for you.

closed 2025-04-03 by yanmartins