espressif/esp-idf #15710: Internal RC oscillator issues with 3.6V in ESP-IDF 5.3.2 (IDFGH-11064)

peter3099, 2025-04-03:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.2

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32

### Power Supply used.

Battery

### What is the expected behavior?

Related to #[Internal RC oscillator issues with 3.6V](https://github.com/espressif/esp-idf/issues/12240#top)

Issue still exists in IDF 5.3.2

### What is the actual behavior?

Related to #[Internal RC oscillator issues with 3.6V](https://github.com/espressif/esp-idf/issues/12240#top)
Issue still exists in IDF 5.3.2 

Drift on wake Up time on timer by hours instead of minutes, it works when GPIO is use for wake up and then wierd values appear in the uptime values and in the general clock

[image]

That value should be around 700 minutes

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

peter3099, 2025-04-09:
Hi @esp-wzh,
I wanted to share a follow-up based on additional testing. This issue still exists even when using a newer version of ESP-IDF — I’ve confirmed it across 420 devices in the field. We were seeing random "offline alerts" where devices failed to wake up from deep sleep and didn't send data as expected.

After investigating, I found that the issue disappears when modifying this line in rtc_sleep.c:

https://github.com/espressif/esp-idf/blob/e088c3766ba440e72268b458a68f27b6e7d63986/components/esp_hw_support/port/esp32/rtc_sleep.c#L111

Original:
out_config->dbg_atten_slp = RTC_CNTL_DBG_ATTEN_DEFAULT;
Changed to:
out_config->dbg_atten_slp = RTC_CNTL_DBG_ATTEN_NODROP;

We're using the RC_FAST clock (150kHz) as the sleep clock source, and it seems that the default voltage attenuation setting causes issues during wake-up. By setting it to NODROP, the devices wake up reliably again.

All we changed was this one line, and since reflashing with this tweak, we haven't seen the issue again on any of the devices.

I originally tried to reopen the previous issue since all the background information is still relevant, but I couldn’t. I hope it’s okay to continue the discussion here — this is likely useful for others facing similar random wake-up failures.

Let me know if you need logs or power profiler traces — happy to help however I can.

esp-wzh, 2025-04-09:
@peter3099 , thanks so much for your feedback, I still have the following questions
1. What does the value `uptimeInMinutes`  mean and how is it obtained in the code?
2. What is the temperature range of the devices with the problem? 
3. Do you mean that all 420 devices have this problem? What is the percentage of devices with the problem?

peter3099, 2025-04-09:
Thanks for the quick response.

1)
uptimeInMinutes is made of a converision of uptimeInMs, it is used to send the values over to the cloud. 

it comes from 
`uint32_t get_uptime_in_min()
{
    return uptimeInMs / 1000 / 60;
}`

and uptimeInMs is updated after every wake up

`void check_wakeup_reason_task()
{
    ESP_LOGI(TAG, "Check Wake UP reason");
    esp_sleep_wakeup_cause_t wakeup_reason;
    wakeup_reason = esp_sleep_get_wakeup_cause();

    struct timeval timeNow;
    struct timeval timeDiff;
    gettimeofday(&timeNow, NULL);
    timersub(&timeNow, &sleepTime, &timeDiff);
    uptimeInMs += (timeDiff.tv_sec * (uint64_t)1000) + (timeDiff.tv_usec / 1000);
    ESP_LOGW(TAG, "WK reason %d Now: %" PRIu64 "ms, Duration: %" PRIu64 "ms",
             wakeup_reason, (timeNow.tv_sec * (uint64_t)1000) + (timeNow.tv_usec / 1000), (timeDiff.tv_sec * (uint64_t)1000) + (timeDiff.tv_usec / 1000));`

It works perfectly fine if the issue doesn't happen, we have devices with uptimeInMinutes values of a year or so.

2)  15 to 25 degC

3) Most of the 420 don't have the issue but we can see that before the firmware upgrade around 3 of them will have spaces of hours on the sampling, now with the firmware change we don't see those gaps anymore. Actually we still see it in the devices that hasn't been updates, from the 420 devices, 39 couldn't be updated becuase of location access. Every now and then we get alerts that the devices has been offline for an hour or 2, if any other interrupt method wakes up the device other than the timer(motion detection or the button press) then the device wakes up but the whole uptime calculation is mess up as shown above

please let me know if you need anymore info

esp-wzh, 2025-04-09:
@peter3099 

Does your application use ULP or other RTC peripherals during deepsleep?

For a known problematic chip, can you reproduce the problem using [deepsleep example](https://github.com/espressif/esp-idf/blob/v5.3.2/examples/system/deep_sleep/main/deep_sleep_example_main.c#L149) (only timer wakeup enabled)?

esp-wzh, 2025-04-09:
What I want to know is what are the minimum conditions required to reproduce this problem, because we have some cases internally that cover common deepsleep usage scenarios, but this problem has never been reported.

You can test separately, gradually adding conditions, first run only the timer wakeup example, then enable ULP in deepsleep, etc.

BTW, you can use [`rtc_time_get` ](https://github.com/espressif/esp-idf/blob/v5.3.2/components/esp_hw_support/port/esp32/include/soc/rtc.h#L440) to get the raw counter value of RTC Timer.

peter3099, 2025-04-10:
I understand, since it behave in random ways it takes time to monitor things, but I'm concern now that the ULP could be one preventing the ESP32-WROVER-E from waking up, is there any way I can check for hints of this? 

This basically what I do to init the ULP 

```void init_ulp_program(void)
{
    esp_err_t err = ulp_load_binary(0, ulp_ultra_low_p_bin_start,
                                    (ulp_ultra_low_p_bin_end - ulp_ultra_low_p_bin_start) / sizeof(unsigned int));
    if (err != ESP_OK)
    {
        printf("Error loading ULP binary: %s\n", esp_err_to_name(err));
    }
    /* Initialize some variables used by ULP program.
     * Each 'ulp_xyz' variable corresponds to 'xyz' variable in the ULP program.
     * These variables are declared in an auto generated header file,
     * 'ulp_main.h', name of this file is defined in component.mk as ULP_APP_NAME.
     * These variables are located in RTC_SLOW_MEM and can be accessed both by the
     * ULP and the main CPUs.
     *
     * Note that the ULP reads only the lower 16 bits of these variables.
     */

    if (cycle_time_ulp == 0)
        cycle_time_ulp = 80;
    ulp_total_cyc = get_deep_sleep_time() / cycle_time_ulp;
    ulp_motion_min_time_cyc = mot_min_time * 1000 / cycle_time_ulp; // default mot_min_time 20   cycle_time_ulp 80

    ESP_LOGW(TAG, "ulp_total_cyc %lu ulp_motion_min_time_cyc: %lu cycle_time_ulp %d",
             ulp_total_cyc, ulp_motion_min_time_cyc, cycle_time_ulp);
    // initialize BTN2 for wake up
    gpio_num_t gpio_num = BTN2;
    int rtcio_btn2 = rtc_io_number_get(gpio_num);
    // assert(rtc_gpio_is_valid_gpio(gpio_num) && "GPIO used for pulse counting must be an RTC IO");
    /* Initialize selected GPIO as RTC IO, enable input, disable pullup and pulldown */
    rtc_gpio_init(gpio_num);
    rtc_gpio_set_direction(gpio_num, RTC_GPIO_MODE_INPUT_ONLY);
    rtc_gpio_pulldown_dis(gpio_num);
    rtc_gpio_pullup_dis(gpio_num);
    rtc_gpio_hold_en(gpio_num);

    ulp_btn2_debou_counter = 1;
    ulp_btn2_debou_max_count = 1;
    ulp_btn2_next_edge = 0;

    ulp_btn2_edge_cnt_to_wake_up = 1;
    ulp_io_btn2 = rtcio_btn2; /* map from GPIO# to RTC_IO# */

    // initialize PIR for counting motion
    gpio_num = PIR;
    int rtcio_pir = rtc_io_number_get(gpio_num);
    // assert(rtc_gpio_is_valid_gpio(gpio_num) && "GPIO used for pulse counting must be an RTC IO");
    /* Initialize selected GPIO as RTC IO, enable input, disable pullup and pulldown */
    rtc_gpio_init(gpio_num);
    rtc_gpio_set_direction(gpio_num, RTC_GPIO_MODE_INPUT_ONLY);
    rtc_gpio_pulldown_dis(gpio_num);
    rtc_gpio_pullup_dis(gpio_num);
    rtc_gpio_hold_en(gpio_num);
    // ulp_pir_next_edge = 1;
    ulp_pir_debou_counter = 0;
    ulp_pir_debou_max_count = 5;
    ulp_motion_detected_cyc = 0; // used to save the cycle number of the last motion
    ulp_pir_no_motion_factor_on = 0;

    // enable the motion sensor to wake up the sensor when last sample was a motion 0
    // if (get_no_motion_factor() > 1)   // old setup only to wake when fallback was enabled
    if (get_no_motion_counter() > 0)
    {
        ulp_pir_no_motion_factor_on = 1;
        // printf("ulp_pir_no_motion_factor_on: %lu noMotionCoutner %d \n", ulp_pir_no_motion_factor_on, get_no_motion_counter());
    }
    ulp_pir_wake_up = 0;
    ulp_io_pir = rtcio_pir; /* map from GPIO# to RTC_IO# */

    ulp_set_wakeup_period(0, cycle_time_ulp * 1000);
    start_ulp_program();
}

void start_ulp_program()
{
    /* Start the program */
    esp_err_t err = ulp_run(&ulp_entry - RTC_SLOW_MEM);
    ESP_ERROR_CHECK(err);
}
```

And then this is my sleep routine reduced

```
init_ulp_program();
if (get_mot_en() == false)
{
    ESP_LOGE(TAG, "MOT ROUTINE DISABLED");
    ESP_ERROR_CHECK(esp_sleep_enable_ext0_wakeup(BTN2, 0));
}
else
{
    ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());
}
print_lora_vars();

esp_deep_sleep_disable_rom_logging(); // suppress boot messages
if (deepSleepTime > 0)
{
    ESP_ERROR_CHECK(esp_sleep_enable_timer_wakeup(deepSleepTime * mS_TO_uS_FACTOR));
}
else
{
    ESP_LOGE(TAG, "MsTosleep Less than 0: %" PRIu64 "ms , going to 300000", deepSleepTime);
    ESP_ERROR_CHECK(esp_sleep_enable_timer_wakeup(300000 * mS_TO_uS_FACTOR));
}
gpio_reset_pin(GPIO_NUM_2);

gettimeofday(&sleepTime, NULL);
uptimeInMs += esp_timer_get_time() / 1000;
if (swtState == SWT_SLEEP)
{
    ESP_LOGE(TAG, "## sT:%" PRIu64 "ms sleepF:%" PRIu64 "s nM#: %d  nMF: %d C:%d uLuC: %d",
             uptimeInMs - sampleInitialUptime, deepSleepTime / 1000, get_no_motion_counter(),
             get_no_motion_factor(), get_cycle_count(), get_uplinks_not_conf());
}
else
{
    ESP_LOGI(TAG, "Awake Total Time:  %" PRIu64 "ms  Sleeping for %" PRIu64 "sec\r\n\r\n",
             uptimeInMs - sampleInitialUptime, deepSleepTime / 1000);
}
esp_deep_sleep_start();
vTaskDelete(NULL);
```

Could it be that the issue is on the assembler files?  Thanks in advance

I rememeber that we also try using the 8MHz crystal oscillator and the issue wasn't happening... but the consumption was higher....

esp-wzh, 2025-04-11:
> Is there any way I can check for hints of this?

You can let the ULP toggle an RTC IO to indicate the ULP working status, and verify that the Timer wakeup failures all occur during the ULP working period. If so, the failure is related to the ULP.

> Could it be that the issue is on the assembler files?

From the code you posted, , your ULP only uses GPIO, right? If so, I don't think it has anything to do with assembler code. If it uses adc, there may be problems, we need more investigation ...

You can change the value of `dbg_atten_slp` for workaround (maybe it is also the final solution),  we can be sure that this solution is safe, and the possible negative impact is just a little higher power consumption (you need to test and confirm that you can accept it).

Besides, you can try to verify the effect of increasing the value of `rtc_dbias_slp` on this problem(change to `RTC_CNTL_DBIAS_0V95` / `RTC_CNTL_DBIAS_1V00` or higher, without change `dbg_atten_slp`), which may solve the problem and get better power consumption.

peter3099, 2025-05-19:
Hi @esp-wzh sorry for the delay in answering this, I will be back in the lab in 2 weeks and will check it.

I'll keep you posted

Alvin1Zhang, 2025-11-26:
@peter3099 Thanks for reporting, would you please help share if any updates for the issue? Thanks.

peter3099, 2025-11-26:
Hi Alvin, the only way we found to solve the issue is to change the  rtc_sleep.c file in the IDF

Original:
out_config->dbg_atten_slp = RTC_CNTL_DBG_ATTEN_DEFAULT;
Changed to:
out_config->dbg_atten_slp = RTC_CNTL_DBG_ATTEN_NODROP;

We found it is stable and we don't see the a big increase in power consumption...