espressif/esp-idf #15711: v5.4.1-dirty

diplfranzhoepfinger, 2025-04-03:
Hi, 

i downloaded 
https://github.com/espressif/esp-idf/releases/tag/v5.4.1

[esp-idf-v5.4.1.zip](https://github.com/espressif/esp-idf/releases/download/v5.4.1/esp-idf-v5.4.1.zip)

--> after Project build i get a v5.4.1-dirty in the Binary .

dobairoland, 2025-04-03:
Hi @diplfranzhoepfinger.

There are some executable flag changes in the repository after extracting it. The file permissions were lost during the ZIP-ping process. You can observe this by running `git status` in the unzipped repository.

Here is the command to discard those changes:

```sh
git reset --hard HEAD
```

The repo won't be dirty after this.

dobairoland, 2025-04-21:
I think this can be closed for inactivity.

closed 2025-04-21 by dobairoland