espressif/esp-idf #15712: Smart config V2 not working on v5.4(.1)

lucaato, 2025-04-03:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 5V

### What is the expected behavior?

I open the app while connected to an ap (both 5ghz/2.4ghz).
The log reports `Found channel`.
I input the password in the esptouch application (smartconfig V2).
I press confirm.
The esp connects successfully to the provisioned ap.

### What is the actual behavior?

Used the app as described in the example `examples/wifi/smart_config`.
The log reports:
```
I (48367) smartconfig: Start to find channel...
I (48367) smartconfig_example: Scan done
I (48527) smartconfig: lock channel: 6
I (48527) smartconfig: TYPE: ESPTOUCH_V2
I (48527) smartconfig: offset:52 tods: 0
I (48527) smartconfig_example: Found channel
I (48697) smartconfig: total_seq: 5
```
But no connection happens as it seems the `SC_EVENT_GOT_SSID_PSWD` callback does not get called.

### Steps to reproduce.

1. Edit the `examples/wifi/smart_config` to use smartconfig v2 instead of v1. Replace `SC_TYPE_ESPTOUCH` with `SC_TYPE_ESPTOUCH_V2`.
```c
    ESP_ERROR_CHECK( esp_smartconfig_set_type(SC_TYPE_ESPTOUCH_V2) );
```
3. Follow the README on the example by using the app (i've tested it on an ap that has the same ssid for both the 2.4ghz and 5ghz bands)
4. By using smartconfig v2 the username and passwords do not get received in the callback. 

### Debug Logs.

```plain
I (31) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (31) boot: compile time Jan  1 1980 00:00:00
I (31) boot: Multicore bootloader
I (32) boot: chip revision: v3.0
I (35) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (43) boot.esp32: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1a4b4h (107700) map
I (127) esp_image: segment 1: paddr=0002a4dc vaddr=3ff80000 size=0001ch (    28) load
I (127) esp_image: segment 2: paddr=0002a500 vaddr=3ffb0000 size=03ed4h ( 16084) load
I (137) esp_image: segment 3: paddr=0002e3dc vaddr=40080000 size=01c3ch (  7228) load
I (141) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=8e14ch (581964) map
I (344) esp_image: segment 5: paddr=000be174 vaddr=40081c3c size=15b5ch ( 88924) load
I (390) boot: Loaded app from partition at offset 0x10000
I (390) boot: Disabling RNG early entropy source...
I (401) cpu_start: Multicore app
I (409) cpu_start: Pro cpu start user code
I (409) cpu_start: cpu freq: 160000000 Hz
I (409) app_init: Application information:
I (409) app_init: Project name:     smart_config
I (414) app_init: App version:      v5.5-dev-2844-g5c51472e82-dirty
I (420) app_init: Compile time:     Jan  1 1980 00:00:00
I (425) app_init: ELF file SHA256:  c3ea4f098...
I (429) app_init: ESP-IDF:          v5.4.1
I (433) efuse_init: Min chip rev:     v0.0
I (437) efuse_init: Max chip rev:     v3.99
I (441) efuse_init: Chip rev:         v3.0
I (445) heap_init: Initializing. RAM available for dynamic allocation:
I (451) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (456) heap_init: At 3FFB83A8 len 00027C58 (159 KiB): DRAM
I (461) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (467) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (472) heap_init: At 40097798 len 00008868 (34 KiB): IRAM
I (479) spi_flash: detected chip: generic
I (481) spi_flash: flash io: dio
W (484) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (497) main_task: Started on CPU0
I (507) main_task: Calling app_main()
I (527) wifi:wifi driver task: 3ffc01b8, prio:23, stack:6656, core=0
I (527) wifi:wifi firmware version: 79fa3f41ba
I (527) wifi:wifi certification version: v7.0
I (527) wifi:config NVS flash: enabled
I (527) wifi:config nano formatting: disabled
I (537) wifi:Init data frame dynamic rx buffer num: 32
I (537) wifi:Init static rx mgmt buffer num: 5
I (547) wifi:Init management short buffer num: 32
I (547) wifi:Init dynamic tx buffer num: 32
I (547) wifi:Init static rx buffer size: 1600
I (557) wifi:Init static rx buffer num: 10
I (557) wifi:Init dynamic rx buffer num: 32
I (567) wifi_init: rx ba win: 6
I (567) wifi_init: accept mbox: 6
I (567) wifi_init: tcpip mbox: 32
I (567) wifi_init: udp mbox: 6
I (577) wifi_init: tcp mbox: 6
I (577) wifi_init: tcp tx win: 5760
I (577) wifi_init: tcp rx win: 5760
I (587) wifi_init: tcp mss: 1440
I (587) wifi_init: WiFi IRAM OP enabled
I (587) wifi_init: WiFi RX IRAM OP enabled
I (597) phy_init: phy_version 4860,6b7a6e5,Feb  6 2025,14:47:07
I (667) wifi:mode : sta (c4:5b:be:31:c0:1c)
I (677) wifi:enable tsf
I (677) main_task: Returned from app_main()
I (727) smartconfig: SC version: V3.0.2
I (5557) wifi:ic_enable_sniffer
I (5567) smartconfig: Start to find channel...
I (5567) smartconfig_example: Scan done
I (13487) smartconfig: lock channel: 6
I (13487) smartconfig: TYPE: ESPTOUCH_V2
I (13487) smartconfig: offset:52 tods: 0
I (13487) smartconfig_example: Found channel
I (20527) smartconfig: total_seq: 5
I (43487) wifi:ic_disable_sniffer
I (43487) smartconfig: smartconfig restart
I (43537) smartconfig: SC version: V3.0.2
I (48367) wifi:ic_enable_sniffer
I (48367) smartconfig: Start to find channel...
I (48367) smartconfig_example: Scan done
I (48527) smartconfig: lock channel: 6
I (48527) smartconfig: TYPE: ESPTOUCH_V2
I (48527) smartconfig: offset:52 tods: 0
I (48527) smartconfig_example: Found channel
I (48697) smartconfig: total_seq: 5
I (78527) wifi:ic_disable_sniffer
I (78527) smartconfig: smartconfig restart
I (78577) smartconfig: SC version: V3.0.2
I (83407) wifi:ic_enable_sniffer
I (83417) smartconfig: Start to find channel...
I (83417) smartconfig_example: Scan done
I (232647) smartconfig: lock channel: 6
I (232657) smartconfig: TYPE: ESPTOUCH_V2
I (232657) smartconfig: offset:52 tods: 0
I (232657) smartconfig_example: Found channel
I (262657) wifi:ic_disable_sniffer
I (262657) smartconfig: smartconfig restart
I (262707) smartconfig: SC version: V3.0.2
I (267537) wifi:ic_enable_sniffer
I (267537) smartconfig: Start to find channel...
I (267537) smartconfig_example: Scan done
I (267817) smartconfig: lock channel: 6
I (267817) smartconfig: TYPE: ESPTOUCH_V2
I (267817) smartconfig: offset:52 tods: 0
I (267817) smartconfig_example: Found channel
I (297417) smartconfig: start scanning again
I (297817) wifi:ic_disable_sniffer
I (297827) smartconfig: smartconfig restart
I (297877) smartconfig: SC version: V3.0.2
I (302707) wifi:ic_enable_sniffer
I (302717) smartconfig: Start to find channel...
I (302717) smartconfig_example: Scan done

Done
```

### More Information.

The smart config V1 version seems to be working as expected.
The smart config V2 version seems to be broken in version following `v5.4`.
I've tested on `v5.3` and on that tag the smartconfig v2 works as expected.

zhangyanjiaoesp, 2025-04-07:
@lucaato 
1. When you use the v5.3 for testing, do you use the same phone and router?
2. Can you open the debug log to see more information ?
3. Can you capture the wifi packets?
4. Which phone and router do you use? And please provide the router configuration, I can try to reproduce the issue locally.

Kirinni, 2025-04-08:
I have the same problem with android14 phone.It can't work well.when I use android10 phone to send smart config packet,it can always success.After config WiFi with android10 phone,android14 phone can be success too.This problem probly caused by the high version android OS?

zhangyanjiaoesp, 2025-04-09:
@Kirinni When you were configuring the network using the Android 10 phone, was the Android 14 phone also connected to the same router? Some routers do not forward UDP packets when only one device is connected.

Kirinni, 2025-04-09:
@zhangyanjiaoesp 
I restarted the router by disconnecting the power supply, hoping that this might quickly clear the list of connected devices.

Under the WiFi of the same SSID, when only the android10 mobile phone is connected to this network, the android10 mobile phone can normally use SmartConfig; After restarting the router, when only the mobile phone of android14 is connected to the WiFi, SmartConfig cannot be used normally. Only when other devices (any device, I tried my laptop of windows 11 and the mobile phone of android10) are also connected to the WiFi of this SSID can the mobile phone of android14 use the SmartConfig normally.

zhangyanjiaoesp, 2025-04-10:
@Kirinni 
It seems the logic is as follows:
The packet format used when an Android 14 phone communicates with the router individually is different from that of an Android 10 phone. For example, Android 14 may use 802.11ac/ax, while Android 10 uses 802.11n. When both devices are connected to the router simultaneously, the router may downgrade the broadcast packet format for compatibility reasons. As a result, the UDP broadcast packets forwarded by the router adopt a format similar to that used when only the Android 10 phone is connected. This format can be recognized by the ESP device, which allows SmartConfig to succeed.

lucaato, 2025-04-10:
> [@lucaato](https://github.com/lucaato)
>
>     1. When you use the v5.3 for testing, do you use the same phone and router?
>
>     2. Can you open the debug log to see more information ?
>
>     3. Can you [...]

1. Yes I've used the same phone and router.
2. Currently I can't share the logs because they contain a network password used for other purposes other than testing. I will share them by repeating the test on another access point.
3. For the same reason described above I will do it when possible. Currenlty I can tell you that I've tested connecting the android and the esp board to an iPhone hotspot created ad-hoc so that most custom configurations can be ruled out. With this setup the provisioning did in fact work correctly. 
4. I use an S21 (Android 14) as the phone, when I'll use the custom ap I will provide you more info.