espressif/esp-idf #15713: . ./export.sh command line, Error while checking requirement 'esp-idf-kconfig<3.0.0,>=2.0.2'. Package was not found and is required by the application: ruamel.yaml.clib

WeDoX, 2025-04-03:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

checking requirement 'esp-idf-kconfig<3.0.0,>=2.0.2'. Package can found and is required by the application: ruamel.yaml.clib

### What is the actual behavior?

* Checking python version ... 3.9.6
* Checking python dependencies ... FAILED
Traceback (most recent call last):
  File "/Users/zl/esp/v5.4.1/esp-idf/tools/export_utils/activate_venv.py", line 195, in <module>
    main()
  File "/Users/zl/esp/v5.4.1/esp-idf/tools/export_utils/activate_venv.py", line 173, in main
    check_python_dependencies()
  File "/Users/zl/esp/v5.4.1/esp-idf/tools/export_utils/console_output.py", line 26, in wrapper
    rv = func(*args, **kwargs)
  File "/Users/zl/esp/v5.4.1/esp-idf/tools/export_utils/activate_venv.py", line 63, in check_python_dependencies
    run_cmd([sys.executable, conf.IDF_TOOLS_PY, 'check-python-dependencies'])
  File "/Users/zl/esp/v5.4.1/esp-idf/tools/export_utils/utils.py", line 46, in run_cmd
    raise RuntimeError(f'Command {cmd_str} failed with error code {p.returncode}\n{stdout}\n{stderr}')
RuntimeError: Command "/Users/zl/.espressif/python_env/idf5.4_py3.9_env/bin/python /Users/zl/esp/v5.4.1/esp-idf/tools/idf_tools.py check-python-dependencies" failed with error code 255
The following Python requirements are not satisfied:
Error while checking requirement 'esp-idf-kconfig<3.0.0,>=2.0.2'. Package was not found and is required by the application: ruamel.yaml.clib
To install the missing packages, please run "install.sh"

[image]

### Steps to reproduce.

1、cd /Users/zl/esp/v5.4.1/esp-idf
2、./install.sh
3、. ./export.sh

### Build or installation Logs.

```plain

```

fhrbata, 2025-04-03:
Hello @WeDoX , 

I believe you are experiencing the same problem as described and "fixed" in [fix(tools): handle packages with dots in their names during dependency checks](https://github.com/espressif/esp-idf/commit/3bad4348d0597597e4079878aa5de1871403e0b2). Unfortunately it's not present in `v5.4.1`.  Maybe you can try to cherry-pick it if it helps.

Thank you

WeDoX, 2025-04-04:
> Hello [@WeDoX](https://github.com/WeDoX) ,
>
> I believe you are experiencing the same problem as described and "fixed" in [fix(tools): handle packages with dots in their names during dependency [...]

Thank you. I have tried it and it seems to work normally now. There is only one WARNING:

WARNING: Failed to load shell autocompletion for bash version: 3.2.57(1)-release!

[image]

fhrbata, 2025-04-04:
Hello,

> WARNING: Failed to load shell autocompletion for bash version: 3.2.57(1)-release!

I believe this is expected, because the export script is using shell completion generated by `click`, which does not support bash versions lower than 4.4.

https://click.palletsprojects.com/en/stable/shell-completion/

> Click provides tab completion support for Bash (version 4.4 and up) ...