espressif/esp-idf #15716: TWAI Message Timeout handling ?

diplfranzhoepfinger, 2025-04-03:
Hi, 

did anybody make a Simple TWAI Timeout handling ? 

@wanckl @Dazza0  ?

i want to receive a 100ms Message, and get a Information if since more than 300 ms no new Message arrived ?

Thanks,
Franz

nopnop2002, 2025-04-05:
Generally, there is no receive timeout in CAN communication.

A timeout is determined when there is no response message to a transmitted message.

diplfranzhoepfinger, 2025-04-05:
> ```
> twai_message_t message;
> if (twai_receive(&message, pdMS_TO_TICKS(300)) == ESP_OK) {
>     printf("Message received\n");
> } else {
>     printf("Failed to receive message\n");
> }
> ```

no, i need it differently. 

say i get 20 Can Messages cyclically. 

0x190, 0x191, 0x192 and so on. 
from 10 Devices, 

and 1 Device stops delivering Messages, 
then the above approach not working...

diplfranzhoepfinger, 2025-04-05:
maybe it would work this way: 
a periodic Timer marks the Message as "unreceived" 
(can be a Free RTOS or a ESP-Timer, 
or even a Task with a Delay) 

if at the "second" loop the MEssage is still "unreceived" i see this is a timeout.

diplfranzhoepfinger, 2025-04-05:
hae another way in my Mind: 

when a Message is first received start a FreeRTOS Timer. 
when the Timer is running, and the Message is received again, call `xTimerReset` to extend the Timer

see https://www.freertos.org/Documentation/02-Kernel/02-Kernel-features/05-Software-timers/05-Resetting-a-timer

then, if the Timer is expired i know that this was a timeout. 

maybe a cool way to do ?

Dazza0, 2025-04-07:
@diplfranzhoepfinger I think what you're asking for is the concept of a "mailbox" in the CAN driver (which the current ESP32/IDF TWAI driver/hardware doesn't have, but is present in some CAN hardware/drivers from other vendors).

Mailboxes are basically different buffers to store received CAN frames based on ID/frame type/payload. Some CAN peripherals are able to do this in hardware, whereas other drivers do this in software. The current IDF driver can be thought of as having a single mailbox since all received frames are stored in the same RX queue. 

If the IDF driver adds a timestamp to `twai_message_t`, then implementing the mailbox logic from user code would also be feasible, where:

- You have a thread that constantly empties the RX queue via `twai_receive()`
- Checking the ID and timestamp of each received message, and updating a corresponding software timer for that ID based on the timestamp.

@wanckl  Can you think of any blockers to adding a mailbox and/or timestamp feature to the TWAI driver? From my POV:

- for timestamps, some of the newer targets should have timestamp in HW, and older targets can simply get the current time when the ISR gets invoked
- for mailboxes, there would need to be some API for users to allocate mailboxes and some RX ISR callback to sort received messages into mailboxes.

wanckl, 2025-04-07:
@Dazza0 Thank you Darian, timestamp feature will avilable as you say in going new driver.

The `1.0` new driver is now **without** timestamp and only support C5, should sync on github recently    
Next plan is to support legacy targets in new driver, then support the timestamp.   

Thanks again for your suggestion

diplfranzhoepfinger, 2025-04-18:
https://github.com/espressif/esp-idf/issues/5613