espressif/esp-idf #15718: error: implicit declaration of function 'gpio_wakeup_enable';

EternityFOR, 2025-04-04:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-1942-g1160a86ba0b

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

build successfully

### What is the actual behavior?

build failed

### Steps to reproduce.

1. Install component   espressif/esp-now: ^2.5.2
2. build

### Build or installation Logs.

```plain
/managed_components/espressif__esp-now/src/debug/src/commands/cmd_system.c:397:26: error: implicit declaration of function 'gpio_wakeup_enable'; did you mean 'rtc_gpio_wakeup_enable'? [-Wimplicit-function-declaration]
  397 |         ESP_ERROR_CHECK( gpio_wakeup_enable(io_num, level ? GPIO_INTR_HIGH_LEVEL : GPIO_INTR_LOW_LEVEL) );
```

### More Information.

If a head file miss in this .c file that I need to add it by myself or any setting I can modify to solve this? I just want to understand the root cause of this kind of problem when I directly add a new component and build.

mbratch, 2025-04-05:
According to the [esp-idf doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html#api-reference-normal-gpio), it looks like you need to include `driver/gpio.h` and you need to require the `esp_driver_gpio_driver` component in your `CMakeList.txt`.

EternityFOR, 2025-04-06:
> According to the [esp-idf doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html#api-reference-normal-gpio), it looks like you need to include `driver/gpio.h` and you need to require the `esp_driver_gpio_driver` component in your `CMakeList.txt`.

what I mean is that this file originally miss a header.(After I install this component by add  this line: espressif/esp-now: ^2.5.2  to the idf_component.yml file and build, then this file:managed_components/espressif__esp-now/src/debug/src/commands/cmd_system.c which automatically added in miss this header, it should be a problem?

suda-morris, 2025-04-07:
Could be a regression made in https://github.com/espressif/esp-idf/commit/3fde2017cd756c5f96a296c9f531bfdf2967a647 where we removed the "driver/gpio.h" from the "driver/rtc_io.h".