espressif/esp-idf #15720: 使用idf.py size指令之后打印的SRAM大小与手册上描述的不符

ZhkuWuMingyang, 2025-04-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

# 使用ESP-IDFv5.2.3，对工程使用idf.py size指令后打印的IRAM大小总共只有128K，DRAM也只有128K。手册上描述的SRAM大小一共有520K，其中SRAM0有192K，除去64K用于缓存，还剩下128K，这与idf.py size指令统计的相符合。但是手册上描述SRAM2有200K用于DRAM，实际上看DRAM只有128K。而手册上描述的可被指令总线和数据总线SRAM1，则没有显示出来？看起来的结果是，IRAM只计算了SRAM0中128K大小，而DRAM计算了SRAM2的大小？能告诉我使用idf.py size指令统计的IRAM和DRAM大小来自于SRAM的哪部分吗？

另一个问题是，如果IRAM的占用大于128K，编译就会报错IRAM太大。但不是还有**128K的SRAM1**可用用吗，这一点也很奇怪？

[esp32_technical_reference_manual_cn (1).pdf](https://github.com/user-attachments/files/19613954/esp32_technical_reference_manual_cn.1.pdf)

ZhkuWuMingyang, 2025-04-05:
这个是使用idf.py size后的内容
[image]

fhrbata, 2025-04-07:
Hello @ZhkuWuMingyang ,

[esp-idf-size](https://github.com/espressif/esp-idf-size) is using so called chip_info files describing the memory types for different memory regions. For esp32 the information used by esp-idf-size can be seen [here](https://github.com/espressif/esp-idf-size/blob/master/esp_idf_size/chip_info/esp32.yaml).

* IRAM  0x3ffae000 - 0x40000000 (335872B) SRAM2 SRAM1
* DRAM 0x40070000 - 0x400c0000 (327680B) SRAM0 SRAM1

The addresses and mapping can be seen in the TRM for esp32 in "Table 1-2. Embedded Memory Address Mapping". Note that SRAM1 can be accessed via instruction and also data bus. 

How the SRAM is used depends on the memory region definitions in the linker script and project's configuration. There is a template in the `esp_system` component for each target's memory definitions, which is expanded based on the configuration.

* IRAM - [iram0_0_seg (RX) : org = 0x40080000, len = 0x20000 + SRAM1_IRAM_LEN](https://github.com/espressif/esp-idf/blob/master/components/esp_system/ld/esp32/memory.ld.in#L55)
* DRAM - [dram0_0_seg (RW) : org = 0x3FFB0000 + CONFIG_BTDM_RESERVE_DRAM, len = DRAM0_0_SEG_LEN - CONFIG_BTDM_RESERVE_DRAM](https://github.com/espressif/esp-idf/blob/master/components/esp_system/ld/esp32/memory.ld.in#L81)

The expanded version of the linker script can be found in the project's build directory in `build/esp-idf/esp_system/ld/memory.ld`.

For example for the default `hello_world` example
* IRAM - `iram0_0_seg (RX) : org = 0x40080000, len = 0x20000 + 0x0` - 131072B of SRAM0
* DRAM - `dram0_0_seg (RW) : org = 0x3FFB0000 + 0,len = 0x2c200 - 0 ` - 180736B of SRAM2
0x3ffdc200

Note that the IRAM may be extended for portion of SRAM1 if  `CONFIG_ESP_SYSTEM_ESP32_SRAM1_REGION_AS_IRAM` is defined. Also the available DRAM size depends on `CONFIG_BTDM_RESERVE_DRAM`, which reserves `56156B` of the memory if Bluetooth is enabled. Leaving the available DRAM size `124580B`.

I guess your project is using Bluetooth, meaning `56156B` is reserved, leaving you with `124580B` DRAM available.
As stated above, the memory usage reported by esp-idf-size depends on the project's configuration and linker script. You can take a look at the referenced links above to take a look and have a better understanding. Also (GNU linker ld manual)[https://sourceware.org/binutils/docs/ld.html] may be helpful.

Thank you

fhrbata, 2025-04-07:
Just for reference we had similar discussion here https://github.com/espressif/esp-idf/issues/15354

fhrbata, 2025-04-14:
Hello, I'm closing this for inactivity. Please feel free to reopen if the information provided is not sufficient. Thank you

closed 2025-04-14 by fhrbata