espressif/esp-idf #15721: What's the 24bit audio data structure read through "i2s_channel_read"?

Ares-bit, 2025-04-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

BOARD: 
esp32-wroom-32

SDK-VERSION: 
v5.2

QUESTION:
I use ESP32 to read audio data of INMP441, which data is 24bit, but when I draw the results by SerialPlot, it's weird. So I'd like to know what the 24bit audio data structure is?

CODE:
`#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/i2s_std.h"
#include "driver/gpio.h"
#include "esp_log.h"

#define INMP_SD     GPIO_NUM_13
#define INMP_SCK    GPIO_NUM_14
#define INMP_WS     GPIO_NUM_27

#define DATA_LEN    1023

i2s_chan_handle_t rx_handle;

void i2s_init(void)
{
    i2s_chan_config_t chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_AUTO, I2S_ROLE_MASTER);
    chan_cfg.dma_frame_num = DATA_LEN;
    i2s_new_channel(&chan_cfg, NULL, &rx_handle);

    i2s_std_config_t std_cfg = {
        .clk_cfg = {
            .clk_src = I2C_CLK_SRC_DEFAULT,
            .mclk_multiple = I2S_MCLK_MULTIPLE_384,
            .sample_rate_hz = 44100,
        },
        .slot_cfg = I2S_STD_MSB_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_24BIT, I2S_SLOT_MODE_MONO),
        .gpio_cfg = {
            .mclk = I2S_GPIO_UNUSED,
            .dout = I2S_GPIO_UNUSED,
            .bclk = INMP_SCK,
            .ws = INMP_WS,
            .din = INMP_SD,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
            },
        },
    };

    i2s_channel_init_std_mode(rx_handle, &std_cfg);

    i2s_channel_enable(rx_handle);
}

void app_main(void){
    i2s_init();
    uint8_t* read_data_buff = (uint8_t*)malloc(sizeof(uint8_t) * 4 * DATA_LEN);
    size_t len = 0;
    for(;;){
        i2s_channel_read(rx_handle, read_data_buff, DATA_LEN * 4, &len, portMAX_DELAY);
        for (uint16_t i = 0; i < len; i += 4){
            int32_t real_data = (read_data_buff[i + 1]) | (read_data_buff[i + 2] << 8) | (read_data_buff[i + 3] << 16);// according to esp32 document, data aligned by 32bit, so I dropped buff[0], and reverse buffer[1][2][3]
            if (real_data & 0x00800000) {
                real_data |= 0xFF000000;
            }
            printf("%ld\r\n",real_data);
        }
        vTaskDelay(1);
    }
}`

BitsForPeople, 2025-04-05:
According to the INMP441's datasheet, it outputs 24-bit, 'left-aligned' data in big-endian order. (Delayed by 1 SCK clock.)
With the ESP32 being a little-endian architecture, your code needs to reverse the byte order read from the mic; but it doesn't do that.
Here's what you'd need to do:
```
// get the big-endian sample data into a (little-endian) 32-bit variable:
uint32_t raw_data = (read_data_buff[i] << 16) | (read_data_buff[i+1] << 8) | (read_data_buff[i + 2]<<0);
// raw_data now holds the sample value in its lower 24 bits
// sign-extend the 24-bit data to an int32 value:
int32_t real_data = (int32_t)(raw_data << 8) >> 8; 
```

Ares-bit, 2025-04-06:
> According to the INMP441's datasheet, it outputs 24-bit, 'left-aligned' data in big-endian order. (Delayed by 1 SCK clock.) With the ESP32 being a little-endian architecture, your code needs to reverse the byte order read from the mic; but it doesn't do that. Here's what you'd need to do:
> 
> ```
> int32_t real_data = (int8_t)read_data_buff[i]; // sign-extend the MSB to 32 bits
> real_data = (real_data << 8) | read_data_buff[i+1];
> real_data = (real_data << 8) | read_data_buff[i+2];
> ```

It doesn't work, real data is always 0, but I try another way seems work, I couldn't explain:
`            int32_t real_data = (read_data_buff[i + 1]) | (read_data_buff[i + 2] << 16) | (read_data_buff[i + 3] << 8);`

BitsForPeople, 2025-04-06:
If the data is already in little-endian format in the buffer, this should be a lot faster:
```
const int32_t* samples = (int32_t*)read_data_buf; // access the buffer as signed 32-bit values
for (uint32_t i = 0; i < (len/sizeof(int32_t)); i += 1) {
  int32_t real_data = samples[i] >> 8; // adjust to 24 bits, discarding the lowest 8 bits.
  ...
}
```

(Your last code mixes up read_data_buff[i + 2] and read_data_buff[i + 3] (MSB).)

Ares-bit, 2025-04-16:
> If the data is already in little-endian format in the buffer, this should be a lot faster:
> 
> ```
> const int32_t* const samples = (const int32_t*)read_data_buf; // access the buffer as signed 32-bit values
> for (uint32_t i = 0; i < (len/sizeof(int32_t)); i += 1) {
>   int32_t real_data = samples[i] >> 8; // adjust to 24 bits, discarding the lowest 8 bits.
>   ...
> }
> ```
> 
> (Your last code mixes up read_data_buff[i + 2] and read_data_buff[i + 3] (MSB).)

Thanks bro, it works. I use this method to get the real data and double it to amplify my voice, it works well. I'm no longer paying attention to SerialPlot display.

L-KAYA, 2025-04-24:
Thanks for the contribution! @BitsForPeople 

Close this issue since it looks solved, please feel free to reopen it if there are still something related to discuss. @Ares-bit

closed 2025-04-24 by L-KAYA