espressif/esp-idf #15725: Duplicate Samples

nopnop2002, 2025-04-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This and this one seem to have the same purpose.

https://github.com/espressif/esp-idf/tree/master/examples/storage/fatfs/getting_started

https://github.com/espressif/esp-idf/tree/master/examples/storage/wear_levelling

RathiSonika, 2026-08-18:
Thanks for the report — the overlap is easy to notice because both use esp_vfs_fat_spiflash_mount_rw_wl.
They are intentional separate examples, not duplicates:

* examples/storage/fatfs/getting_started — minimal FATFS entry point (mount → write → read → unmount).
* examples/storage/wear_levelling — same mount path, plus wear-levelling focused lifecycle: format with esp_vfs_fat_spiflash_format_rw_wl, verify files are gone, write again.

We are keeping both. To make that clearer for new users (and tooling), we will improve the example READMEs with:

1. an explicit Purpose / What this demonstrates section
2. a Related examples table (when to use which)
3. a short note that LittleFS and NVS handle wear differently and do not use this wear_levelling example path
4. corrections so the wear_levelling README matches the actual format/verify flow in the code

So this is by design; the fix is documentation clarity, not removing either example.

nopnop2002, 2026-08-18:
@RathiSonika 

Please close this issue once the document revisions are complete.