espressif/esp-idf #15726: IDF installation fails on MacOS

BbIKTOP, 2025-04-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect to have ESP-IDF installed

### What is the actual behavior?

Hi there!

I had a perfectly working IDF installation for years but dropped my ESP projects for a while because of a lot of tasks on my primary work. Now I tried to reinstall the EDF and failed. Please help.
What I did:
Removed ~/esp/* and ~/.espressif completely.
Downloaded the IDF by ``git clone --recursive https://github.com/espressif/esp-idf.git`` to the ~/esp/v5.4.1/esp-idf directory
Checked out with ``git checkout v5.4.1``
Executed ``export IDF_PATH=~/esp/v5.4.1/esp-idf`` and added it to the .profile
restarted terminal session again
cd to ``~/esp/v5.4.1/esp-idf``
Ran ``./install.sh all``

At this point, when I run ``. ./export.sh``, I am constantly getting an error:

```
viktor@work:~/esp/v5.4.1/esp-idf $ . ./export.sh
Checking "python3" ...
Python 3.9.6
"python3" has been detected
Activating ESP-IDF 5.4
Setting IDF_PATH to '/Users/viktor/esp/v5.4.1/esp-idf'.
* Checking python version ... 3.9.6
* Checking python dependencies ... FAILED
error:  Command "/Users/viktor/.espressif/python_env/idf5.4_py3.9_env/bin/python /Users/viktor/esp/v5.4.1/esp-idf/tools/idf_tools.py check-python-dependencies" failed with error code 255
The following Python requirements are not satisfied:
Error while checking requirement 'idf-component-manager~=2.1'. Package was not found and is required by the application: ruamel.yaml.clib
To install the missing packages, please run "install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /Users/viktor/.espressif/python_env/idf5.4_py3.9_env
    Python interpreter used: /Users/viktor/.espressif/python_env/idf5.4_py3.9_env/bin/python
Constraint file: /Users/viktor/.espressif/espidf.constraints.v5.4.txt
Requirement files:
 - /Users/viktor/esp/v5.4.1/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /Users/viktor/.espressif/python_env/idf5.4_py3.9_env/bin/python

ERROR: Activation script failed
Command '['/Users/viktor/.espressif/python_env/idf5.4_py3.9_env/bin/python', '/Users/viktor/esp/v5.4.1/esp-idf/tools/export_utils/activate_venv.py', '--export', '--shell', 'bash']' returned non-zero exit status 1.
To view detailed debug information, set ESP_IDF_EXPORT_DEBUG=1 and run the export script again.
```

Running it multiple times, I'm getting different unsatisfied requirements, like ``Error while checking requirement 'esp-idf-size>=1.4.0'. Package was not found and is required by the application: ruamel.yaml.clib`` or ``Error while checking requirement 'esp-idf-kconfig<3.0.0,>=2.0.2'. Package was not found and is required by the application: ruamel.yaml.clib``, which looks completely random.

[... 87 lines cut ...]
Requirement already satisfied: certifi>=2017.4.17 in /Users/viktor/.espressif/python_env/idf5.4_py3.9_env/lib/python3.9/site-packages (from requests->idf-component-manager->-r /Users/viktor/esp/v5.4.1/esp-idf/tools/requirements/requirements.core.txt (line 16)) (2025.1.31)
Requirement already satisfied: ruamel.yaml.clib>=0.2.7 in /Users/viktor/.espressif/python_env/idf5.4_py3.9_env/lib/python3.9/site-packages (from ruamel.yaml->idf-component-manager->-r /Users/viktor/esp/v5.4.1/esp-idf/tools/requirements/requirements.core.txt (line 16)) (0.2.12)
All done! You can now run:

  . ./export.sh

viktor@work:~/esp/v5.4.1/esp-idf $
```

Please advise - am I doing something wrong? Is it a bug? Tried with ``git checkout master`` which gives me the same v5.4.1 branch and the same error.

Thank you.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

closed 2025-04-07 by BbIKTOP

fhrbata, 2025-04-07:
Hello @BbIKTOP ,

I guess you were able to resolve this problem. I will just add that this seems like duplicate of https://github.com/espressif/esp-idf/issues/15713

Thank you

BbIKTOP, 2025-04-07:
@fhrbata 

Hi Frantisek,
Yes, thank you. But now I think it might be worth adding this to the documentation. The reason is that no Python version works under macOS by default, except those installed externally. For example, I had a perfectly working 3.9 installed with Xcode, but the problem disappeared as soon as I installed 3.13 via Homebrew and switched to it.

fhrbata, 2025-04-07:
Hello @BbIKTOP ,

thank you for getting back with the additional information!

I tried to explain what's happening in the commit message for the initial (fix)[https://github.com/espressif/esp-idf/commit/3bad4348d0597597e4079878aa5de1871403e0b2]

> From packaging point of view `ruamel.yaml.clib-0.2.12.dist-info` and
`ruamel_yaml_clib-0.2.12.dist-info` are the same packages, but this is not
reflected in `importlib.metadata` prior python 3.10[9], which does not perform
name normalization prior the distribution search. This causes the `version`
from `importlib.metadata` to fail on python prior the 3.10 version if the
package with dots in distribution name was generated with normalized paths with
newer `setuptools`. Note that the distribution name normalization was
backported to some later 3.9 python version.

So this seems to be related to python version and `importlib.metadata`. Meaning I would expect that with python `3.13` you should not see this error, because the `importlib.metadata` should normalize the package name since python `3.10`. We removed the `ruamel.yaml.clib` packages with `_` in the distribution name from our pypi and the upstream ones seem fine. Along with the fix mentioned, this problem should be hopefully mitigated. Do you see the same error if you install the upstream package within the esp-idf environment activated?

`pip install --no-cache-dir ruamel.yaml.clib`

The `--no-cache-dir` is used just to make sure that the upstream `ruamel.yaml.clib` package is really used instead of some cached version.

With python 3.9 you will probably not be able to activate the esp-idf environment with the `. ./export.sh` script because of the error, but you can try to activate it manually with

`. /Users/viktor/.espressif/python_env/idf5.4_py3.9_env/bin/activate`

and install the upstream version with

```
pip uninstall ruamel.yaml.clib
pip install --no-cache-dir ruamel.yaml.clib
```

Thank you very much!

BbIKTOP, 2025-04-08:
Hi Frantisek,
Looks like the upstream is working fine. Sorry, I have a lot of urgent tasks at work, so I can't investigate further - at least not until the weekend. But I’ll test it then if necessary. Thank you!

fhrbata, 2025-04-08:
> Hi Frantisek,
Looks like the upstream is working fine. Sorry, I have a lot of urgent tasks at work, so I can't investigate further - at least not until the weekend. But I’ll test it then if necessary. Thank you!

Hello @BbIKTOP , I of course understand. Thank you very much for confirmation! It really helps to know that the upstream package is working for you. 

Again thank you very much