espressif/esp-idf #15727: I'm able to use the usb host example to get info but what about data?

ulao, 2025-04-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Are there any exmaples that shows  how to poll a USB interup device? I'd like to get soem data from a pipe and also maybe recive and send control trasfers.

tore-espressif, 2025-04-07:
@ulao When interacting with a USB device, you will typically need a class driver, that is on top of USB Host Library. 
There are many classes for various device, eg. UVC for video, CDC for data, or HID for peripherals.

You can have a look at some host class examples here https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host

ulao, 2025-04-07:
@tore-espressif 
thank you this is what I was looking for.  It has been a long time since I tried to use the exmaple code and I'm once again confused where to find these files. 

I'm able to see the exmaple but when I try to include them in my project they are not found. I did a search for the files in the entire repo and I also can not find tehse files. 

#include "usb/hid_host.h"
#include "usb/hid_usage_keyboard.h"
#include "usb/hid_usage_mouse.h"

This is my current makelist

idf_component_register(SRCS "usb2llapi.c" "usb_host.c" "LLAPI.c" "class_driver.c"
                    INCLUDE_DIRS "."
                    PRIV_REQUIRES usb esp_driver_gpio esp_timer
                    )

How di I get these included?