espressif/esp-idf #15728: PTP can be support ESP32S3

skylin008, 2025-04-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Thank you shared this impressive project. I had notice the ptp can be support esp32p4, does it can be support other esp32 chip, for example esp32s3, esp32c6, esp32c3, so on. Thanks!

suda-morris, 2025-04-07:
ESP-IDF currently only support the PTP at the MAC layer. And only ESP32-P4 has the EMAC controller which support IEEE1588v2.

skylin008, 2025-04-13:
Thanks @suda-morris. How about PTP master accurately  be run in esp32p4 eval board? What are the requirements about ETH PHY? IP101GRI be can support PTP master?

Dominaezzz, 2025-04-13:
Duplicate of https://github.com/espressif/esp-idf/issues/1223 ?

skylin008, 2025-04-14:
When I run ptp example , any error message shows, attach logs .

[ptp.log](https://github.com/user-attachments/files/19728936/ptp.log)

kostaond, 2025-04-14:
Any statistical method to measure PTP precision hasn't been performed yet but when I was testing the the PTP, it's mostly in 60ns (there are time to time peaks to ~100ns). This was measured by GPIO toggling from ISR. 40 MHz CLK was used as as PTP CLK reference. 80 MHz reference could be used too. However, I haven't tried yet. This option is currently not available for user space. 

Regarding PHY, it's PHY independent since time stamping is performed in internal MAC.

> When I run ptp example , any error message shows, attach logs .

What is your setup? `errno` 5 indicates I/O error. Is your link UP? Try to enable debug messages.

skylin008, 2025-04-14:
Thanks @kostaond, It running default setup with esp32p4 eval board. what is ptp default ip setting?

kostaond, 2025-04-14:
Do you have two boards connected together? PTP messages are exchanged at IEEE 802.3 Ethernet (L2) layer only, i.e. no IP.

skylin008, 2025-04-15:
Thanks every one! I want to used esp32p4 EVAL BOARD for  ptp master, and lidar , camera  connect to switch, attach figure.

[image]

kostaond, 2025-04-15:
OK, that's certainly possible. Have you been able to resolve the I/O error in your setup? I tried locally and the PTP master sends sync message as expected. I tested on master branch.

skylin008, 2025-04-16:
Thanks @kostaond, I had only one ESP32P4 EVAL board. If I can test PTP master to another slave device(for instance ubuntu host machine), the machine run linux ptp4l and phc2sys function.

skylin008, 2025-04-16:
And I want to add GPS module, it can provide the UTC time, then through the UART port , parse the GPS sentence to PTPT master

kostaond, 2025-04-16:
> Thanks [@kostaond](https://github.com/kostaond), I had only one ESP32P4 EVAL board. If I can test PTP master to another slave device(for instance ubuntu host machine), the machine run linux ptp4l and phc2sys function.

If the link us UP, it should work. Only don't expect precise sync since it is really not common to PCs have hardware time stamp support.

skylin008, 2025-04-29:
@kostaond Thanks. I had tested the PTP master via ubuntu machine,it has a hardware timestamp. ubuntu machine is slaveonly mode, it runs follow comad: sudo ptp4l -i enp0s31f6 -m -H -s . esp32p4 eval board show as logs only: 

I (1312327) ptpd: Sent sync + follow-up, seq 1295

I (1313337) ptpd: Sent announce, seq 130

I (1313337) ptpd: Sent sync + follow-up, seq 1296

I (1314347) ptpd: Sent sync + follow-up, seq 1297

I (1315357) ptpd: Sent sync + follow-up, seq 1298

I (1316367) ptpd: Sent sync + follow-up, seq 1299

I (1317377) ptpd: Sent sync + follow-up, seq 1300

is it right?

skylin008, 2025-06-15:
Is there IEEE1588 PTP master function test example?

kostaond, 2025-06-16:
https://github.com/espressif/esp-idf/tree/master/examples/ethernet/ptp

kostaond, 2025-07-02:
@skylin008, to be honest, I still don't see any real problem to be reported by this GH issue... If you want to just discuss or share your PTP related observations, please use our forum at www.esp32.com for that purpouse. Thank you for you understanding.