espressif/esp-idf #15729: Possible race condition in esp_log_system_timestamp()

gonzzor, 2025-04-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-2844-g5c51472e82

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Correct usage of locking in `esp_log_system_timestamp()`.

### What is the actual behavior?

Notice that the lock protects the buffer when writing to it and then return the same static buffer that could be written to by the next caller of `esp_log_system_timestamp()` before it gets used by the first.

https://github.com/espressif/esp-idf/blob/5c51472e82a58098dda8d40a1c4f250c374fc900/components/log/src/log_timestamp_common.c#L52-L62

### Steps to reproduce.

I haven't reproduced the issue, only read the code.

### Debug Logs.

```plain

```

gonzzor, 2025-04-07:
I must have missed that issue.

For a single core system it makes sense. For a multi core system I still think there is a race here.

Say two task, each on it's own core.

1. Task A on core 0 takes the lock.
2. Task B on core 1 is blocked by the lock.
3. Task A releases lock.
4. Task B takes lock and begin updating buffer.
5. At the same time task A uses the buffer.

In the scenario above task B is updating the buffer while task A is using it. If timed correctly there can be some strange timestamps.

Thanks for the quick response.

igrr, 2025-04-07:
Good point, thank you for outlining this scenario. Indeed it seems like we can get incorrect timestamps this way (e.g. 01:59 when time changes from 00:59 to 01:00).

I'm not sure how whether the issue can be completely fixed given the existing API of `char* esp_log_system_timestamp(void)` without incurring an unreasonable cost.

The good news is that this function is no longer used if the new implementation (CONFIG_LOG_VERSION_2) is enabled.

A few options come to mind:

1. `buffer` can be made thread-local (using attribute `__thread`). This will use 18 bytes of RAM (20 due to alignment) times the number of tasks in the system. Maybe not be unreasonable given that `CONFIG_LOG_TIMESTAMP_SOURCE_SYSTEM` is not enabled by default.
2. We could reduce the window for the race condition (but not eliminate it) by `snprintf`'ing into a temporary buffer, then doing a `memcpy` into the shared buffer.
3. Keep as is and consider this a "known issue" in LogV1 if CONFIG_LOG_TIMESTAMP_SOURCE_SYSTEM is used...

cc @KonstantinKondrashov