espressif/esp-idf #15731: Clang and C++23 (IDFGH-15015)

falvia, 2025-04-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Hello everyone,

I'm attempting to use static analysis in my project and I am compiling with Clang, following the ESP-IDF documentation: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/tools/idf-clang-tidy.html

While most of my code builds successfully, I'm encountering some issues with C++23 features. For example, std::expected and std::unexpected fail to compile with the following error:

`error: no template named 'unexpected' in namespace 'std'
   47 | inline std::unexpected<Error>
`

Could you advise on how to resolve this?

Thanks in advance for your help.

### Actual behavior (suspected bug)

I don't know if it is a bug or it is not yet supported.

### Error logs or terminal output

```shell
41:13: error: no template named 'unexpected' in namespace 'std'
   41 | inline std::unexpected<Error>
      |        ~~~~~^
44:14: error: no member named 'unexpected' in namespace 'std'
   44 |         return std::unexpected(Error {{ErrorInfo {msg, location}}});
      |                ~~~~~^
47:13: error: no template named 'unexpected' in namespace 'std'
   47 | inline std::unexpected<Error>
      |        ~~~~~^
52:14: error: no member named 'unexpected' in namespace 'std'
   52 |         return std::unexpected(std::move(error));
      |                ~~~~~^
 error: no template named 'expected' in namespace 'std'
   11 |         std::expected<void, ***::util::Error> init();
      |         ~~~~~^
error: no template named 'expected' in namespace 'std'
    8 | std::expected<void, ***::util::Error> Console::init()
      | ~~~~~^
6 errors generated.
```

### Steps to reproduce the behavior

1. following the documentation (set IDF_TOOLCHAIN=clang, install esp-clang, run export.sh)
2.  use std::expected in the code
3. idf.py clang-check (this already fails)
4. idf.py build

### Project release version

v5.4.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu

### Shell

Bash

dobairoland, 2025-04-07:
@falvia I've moved your report into this repository where I think it will be addressed.

falvia, 2025-04-07:
@mfialaf @dobairoland many thanks!

closed 2025-04-18 by espressif-bot