espressif/esp-idf #15732: RISC-V ULP wake-up timer and gpio

JVKran, 2025-04-07:
### Describe the solution you'd like.

Instead of being able to select only ```ULP_RISCV_WAKEUP_SOURCE_GPIO``` or only ```ULP_RISCV_WAKEUP_SOURCE_TIMER```, I'd like to use both of them at the same time to wakeup periodically and when an interrupt occurs.

`c
 ulp_riscv_cfg_t ulp_cfg = {
    .wakeup_source = ULP_RISCV_WAKEUP_SOURCE_GPIO,
};
`

### Additional context.

This is useful for battery monitoring purposes. On interrupt, we decide if we want to wake-up or not and enter shipping mode (very low power ~10uA) if the battery is nearly empty. We also want to do so periodically, like every hour.

SoucheSouche, 2025-04-09:
Hello @JVKran, unfortunately configuring several wakeup sources is not supported by hardware.

You can however take a look at [this example](https://github.com/espressif/esp-idf/tree/master/examples/system/ulp/ulp_riscv/interrupts) where you can see how to use GPIO interrupts to wake the main processor.

JVKran, 2025-04-11:
Hi @SoucheSouche, that's a bummer, but it is what it is. Thanks for your help.

closed 2025-04-11 by JVKran

X-Ryl669, 2025-04-15:
You can partially do that using the LP core. In the LP core, program both interrupt (lp_timer and gpio_intr) and then wait for interrupt. 

I'm doing similar thing. It'll consume ~240µA instead of 10µA, so it still might not fit your need, but it's way better than the mA of the LP core consumption.

JVKran, 2025-04-17:
Interesting, I'll have to do some documentation reading on the LP core. Thanks for pointing that out @X-Ryl669!

X-Ryl669, 2025-04-19:
In fact, I got it to work even in deep sleep. See attached archive.
@SoucheSouche answer is wrong.

The example uses 2 wakeup sources (GPIO + LP_TIMER). The LP core program is dumb, but easy enough to follow:
1. It starts, 
2. It burns CPU for 2s, 
3. It programs its next wakeup (3s later) 
4. It halts (entering deep sleep)

Whenever a event happens here (in deep sleep), it's woken up and redo the same sequence.
You'll get this output from the power profiler:

[image]

Where you see "burning" time (~500µA), wait/deep sleep time (~40µA) and short wait time when I pressed the button (resuming to burning time immediately)

[ULPDualWakeup.tar.gz](https://github.com/user-attachments/files/19820740/ULPDualWakeup.tar.gz)

sudeep-mohanty, 2025-04-23:
Indeed @X-Ryl669 there could be multiple wakeup sources for the LP Core (on targets like esp32c6). But unfortunately, on targets like esp32s2 and esp32s3 which support the previous iteration of the ULP coprocessor (known as ULP-RISCV coprocessor), we do not have a way to program multiple wakeup sources inherently. However, it is still possible to have GPIO interrupts on the ULP-RISCV coprocessor to achieve something similar.

X-Ryl669, 2025-04-23:
Yes, I've seen that in the HAL of this repository. I wonder if it makes sense to provide an abstraction layer for all targets on the ULP itself. Typically, when one program more than a single wake up source, the code could instead prevent the ULP from halting and go into WFI mode instead. So it works, but it consumes more.

Also, it doesn't make sense to have WFI on C5/C6x/P4 in most case. Since you can put the LP core to deep sleep completely and have it wake up upon interrupt anyway. The consumption in WFI for us is like ~300µA while it's ~30µA for deep sleep (and ~560µA if busy waiting the LP core).