espressif/esp-idf #15733: Simultaneous use of wifi_prov and nimble

ele-peterh, 2025-04-07:
Hi,
I am currently using wifi provisioning similar to the provided example in esp-idf/examples. Wifi provisioning is done through BLE. Now I would like to add additional functionality via BLE. I am using only BLE/nimble, no classic Bluetooth as that consumes too much RAM for our use case. 

Is it possible to add additional BLE characteristics which would be available regardless if the wifi provisioning is running?

In documentation I found only custom BLE endpoints via wifi_prov, which are available only when the wifi provisioning is running.

hrushikesh430, 2025-08-13:
Hello @ele-peterh,
From what I understand, you want to add characteristics while provisioning.
If that’s the case, you can use the custom-data endpoint to send the custom data to the device. This can be done both during and after provisioning.

Just ensure that BLE remains on after provisioning so you can send the custom data to the custom-data endpoint. This can be achieved by enabling WIFI_PROV_KEEP_BLE_ON_AFTER_PROV.

If I’ve misunderstood your issue, please feel free to correct me.

ele-peterh, 2025-08-13:
Yes your understanding is correct. It has been quite a long time since I posted this and we have since resolved our issue by implementing a custom provisioning and BLE characteristics. 

If I remember correctly, with the approach you mentioned, the problem was that you had to start provisioning regardless if the wifi is provisioned or not, so that the BLE stack would startup. Also if you use custom-data endpoint, you are constrained to using protocomm, correct? There is no possibility to register BLE characteristic (or multiple characteristics) and use raw data from it, right?

Anyway, thank you for the reply.

hrushikesh430, 2025-08-20:
> I see. I think the reason we strayed away from using protocomm endpoints was also that you had to implement a protocomm client in mobile app for which we couldn't find much documentation.

You can refer [esp_prov.py](https://github.com/espressif/esp-idf/tree/master/tools/esp_prov) script, tool which helps for provisioning and other features like sending `custom-data` and `reprovisioning`.

ele-peterh, 2025-08-20:
Great, thanks! I think we can close this thread now.