espressif/esp-idf #15734: I2C Transaction failed after update to ESP 5.4.1 and Arduino 2.3.5

MarksWizard, 2025-04-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I auto updated to ESP 5.4.1 and Arduino 2.3.5 today, and uploaded code now causes errors where there were none previously.  System is an ESP32-S3 N8R2, with PSRAM turned on.

I have a 0.96 OLED display (SSD1306) on Wire, and 2 different I2C devices (SHT30 temp sensor and vl53l4 TOF sensor) on Wire1.  

2:55:19.265 -> E (4123) i2c.master: I2C software timeout
12:55:19.265 -> E (4123) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
12:55:19.265 -> E (4124) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
12:55:19.331 -> E (4181) i2c.master: I2C software timeout
12:55:19.331 -> E (4181) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed

mythbuster5, 2025-04-08:
Increase timeout threshold and show your logic wave if possible which can provide some clue.

mythbuster5, 2025-04-08:
@MarksWizard  Could you please try it on master? A bug related to `i2c_master_multi_buffer_transmit` has just been solved.

MarksWizard, 2025-04-08:
Do you know if Arduino code will compile under ESP-IDF?  This is a fairly complex program with several libraries and 3000+ lines of my own code.  I am not sure how easily this will port to the ESP-IDF, though I have definitely considered making the switch.  I also have never used the tool chain for my own code, only to compile and upload other people’s projects.  Again, I had never seen this error before auto updating Arduino IDE and ESP yesterday.  I checked the Arduino changelog and there is no mention of I2C in the latest update.

1) I updated from whatever the last release was, it looks 5.2.5?
2) I am using several software libraries to control/communicate with the various sensors on the device, so I am not sure where the troubled call is coming from.  It literally floods my serial output with those messages.  I will try to poke through the libraries today and see where the command `i2c_master_multi_buffer_transmit` is coming from.
3) I don’t have an oscilloscope unfortunately.

In the mean time I am going to try to comment out the I2C sensors individually to see if I can find which one is causing the issue.

MarksWizard, 2025-04-08:
Arduino uses this link for the “Boards Manager” to include ESP32 programming on the Arduino IDE.  Not sure if it provides any clues?

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

me-no-dev, 2025-04-08:
@MarksWizard while it is not possible to install Arduino based on ESP-IDF master through IDE's board manager, you can manually install it in your sketches folder by following these steps: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation (further down there are guides for Linux and Mac if that is what you are running). Just make sure that when you are pulling the Arduino repo, you are pulling the `idf-master` branch. This will get the latest development sources and libs based on the latest IDF master.

ve3sjk, 2025-04-08:
I have the exact same problem as well with a single bme680 unit connected to esp32. Previous no issues as unit is long running, but i started to get these error after the same updates. i thought it was the wiring as the unit would run for some time before the error showed itself again. A reboot gets it running again but it eventually comes back.

MarksWizard, 2025-04-08:
@me-no-dev 

I believe I have correctly installed master branch to /Users/MarksWizard/Documents/Arduino

 > Verify archive...  Extracting archive...
Renaming mkspiffs-0.2.3-arduino-esp32-osx to mkspiffs ...
Elapsed time 00:00.01 Files extracted successfully.

Platform Tools Installed

Restarted Arduino IDE and reuploaded code, but the issue remains.  I am going to restart my system now, I will report back if anything changes.

`12:30:55.002 -> E (225576) i2c.master: I2C software timeout
12:30:55.002 -> E (225576) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
12:30:55.002 -> E (225577) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
12:30:55.034 -> E (225634) i2c.master: I2C software timeout
12:30:55.034 -> E (225634) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
12:30:55.068 -> E (225634) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
12:30:55.100 -> E (225692) i2c.master: I2C software timeout
12:30:55.100 -> E (225692) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
12:30:55.100 -> E (225693) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
12:30:55.166 -> E (225750) i2c.master: I2C software timeout
12:30:55.166 -> E (225750) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
12:30:55.166 -> E (225750) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
12:30:55.231 -> E (225808) i2c.master: I2C software timeout
12:30:55.231 -> E (225808) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
12:30:55.231 -> E (225809) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
`

ve3sjk, 2025-04-08:
@me-no-dev 

That does not seem to work.

In my case i have multiple version of arduino on my system so the IDE 2.0 resides under "documents/arduino 2.0 Esp" which i installed as per the instructions in your post. 

I can see the hardware folder under the directory now.

But when i compile the ide uses this folder to compile with

 "C:\Users\sysop\AppData\Local\Arduino15\packages\esp32\hardware\esp32

So it does not use the one in the sketchbook folder so what i did was copied the files for the one created with the instructions into the proper directory under AppData and recompiled and upload my sketch.

As soon as a boot the unit with the button on the esp i get the following errrors as my sketch runs.

`
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4888
load:0x40078000,len:16516
load:0x40080400,len:4
load:0x40080404,len:3476
entry 0x400805b4

E (17) i2c.master: I2C hardware timeout detected
E (17) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pul                                                                                                                                                             l-ups are correctly set up
I2C device found at address 0x77
done

BSEC library version 1.4.8.0
E (4059) i2c.master: I2C hardware NACK detected
E (4059) i2c.master: I2C transaction unexpected nack detected
E (4060) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (4067) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (4674) i2c.master: I2C hardware NACK detected
E (4674) i2c.master: I2C transaction unexpected nack detected
E (4674) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (4680) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (4787) i2c.master: I2C hardware NACK detected
E (4787) i2c.master: I2C transaction unexpected nack detected
E (4787) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (4792) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (4900) i2c.master: I2C hardware NACK detected
E (4900) i2c.master: I2C transaction unexpected nack detected
E (4900) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (4905) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (5013) i2c.master: I2C hardware NACK detected
E (5013) i2c.master: I2C transaction unexpected nack detected
E (5013) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (5018) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (5126) i2c.master: I2C hardware NACK detected
E (5126) i2c.master: I2C transaction unexpected nack detected
E (5126) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (5131) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (5239) i2c.master: I2C hardware NACK detected
E (5239) i2c.master: I2C transaction unexpected nack detected
E (5239) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (5244) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (5352) i2c.master: I2C hardware NACK detected
E (5352) i2c.master: I2C transaction unexpected nack detected
E (5352) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (5357) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (5465) i2c.master: I2C hardware NACK detected
E (5465) i2c.master: I2C transaction unexpected nack detected
E (5465) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
[... 129 lines cut ...]
                                                    Setting up time
  Got the time from NTP
  Setting Timezone to EST5EDT,M3.2.0,M11.1.0

Tuesday, April 08 2025 12:33:34 zone EDT -0400
Time variables
12

Connect to MQTT broker...connected

topic home/nb/node03/dev99 value:NODE 3 WAKEUP: ESP32_03_FAN
topic home/nb/node03/dev00 value:0
topic home/nb/node03/dev01 value:0
topic home/nb/node03/dev02 value:-74
topic home/nb/node03/dev03 value:ESP32_03_FAN
topic home/nb/node03/dev10 value:192.168.1.103
topic home/nb/node03/dev16 value:OFF
topic home/nb/node03/dev17 value:OFF
topic home/nb/node03/dev18 value:OFF
topic home/nb/node03/dev19 value:OFF
topic home/nb/node03/dev52 value:{"dev":"52","N4_TH":[34.30,44.06,343,67,34,44]}
topic home/nb/node03/dev49 value:{"dev":"49","N1_T8":[41.90,44.06,46.04,41.00,40.10,45.50,41.90,40.82]}
E (103229) i2c.master: I2C hardware NACK detected

`

ve3sjk, 2025-04-09:
I beleive this error is pointing a to wiring problem and not the latest release, i have serveral units all with i2c devices, i have recompiled and uploaded their firmware using the stock updates of arduino without any changes and the trouble does not follow to the other units.

MarksWizard, 2025-04-09:
@ve3sjk although it’s not entirely impossible, I did not move my device at all during the upload.  Several people are reporting this issue, all arising from the update.

[https://github.com/espressif/arduino-esp32/issues/11228](url)

me-no-dev, 2025-04-09:
@MarksWizard when you select a board, you will see two cores in the boards menu, one will say `ESP32 Arduino (in Sketchbook)` and that is the one you need to select board from. The one that says `esp32` is the one installed from Board Manager

ve3sjk, 2025-04-09:
@MarksWizard That is true in my case i did re locate the sensor. What i found intersting is that i have two identical boards, both freenov breakouts and freenov esp32. I select esp32 dev module under boards. I do see that link under boards and can confirm i am compiling under the new git stuff that i installed as above.

So the board with the error has 1 i2c device on it bme680 and is talking to 3 modbus devices and I am getting the same error as you.

The other board same hardware has 6 i2c devices and is talking to 2 ultrasonic sensors with serial and 6 modbus devices and it has no errors at all, despite the fact that it also has the same device as the first board, being a bme680 sensor as one of its sensors.

I just flashed the first one again now that i confirmed i was using the new boards will let it run and see how long it takes to throw the errors.

mythbuster5, 2025-04-09:
@ve3sjk 
```
E (17) i2c.master: I2C hardware timeout detected
E (17) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pul l-ups are correctly set up
I2C device found at address 0x77
```

This has already been fixed on idf master. Becuase probe doesn't support stretch before.

MarksWizard, 2025-04-09:
Hi @me-no-dev @mythbuster5 

I have uploaded my code using the ESP-IDF master that me-no-dev helped me set up using ESP32 Arduino (in Sketchbook).  Compiling and uploading code to a bare (no sensors) ESP32 S3 dev board still gives me the same errors. :(

Today I will work on trying to reproduce the issue with a smaller sketch.  I’m not sure yet, but I think the issue may have something to do with the OLED display.  For that I am using the this library.

https://github.com/vlast3k/Arduino-libraries/blob/master/SSD1306/SSD1306Wire.h

MarksWizard, 2025-04-09:
Ok since I thought it was my OLED library, I decided to switch it up.  This much simpler example code from  a different library (Adafruit), but is still producing the error.  

ssd1306_128x64_i2c example from https://github.com/adafruit/Adafruit_SSD1306

MarksWizard, 2025-04-09:
@mythbuster5 

After a bit of a struggle, I was able to compile and upload the TextBoxDemo from https://github.com/nopnop2002/esp-idf-ssd1306/tree/master directly through ESP-IDF (No Arduino).  I still saw the errors.

Then I made sure that my ESP-IDF was on the master branch, reinstalled everything, and reuploaded the TextBoxDemo code..  I am still seeing the same error.

idf.py --version                       
ESP-IDF v5.5-dev-1655-gc5865270b5-dirty

ve3sjk, 2025-04-10:
> [@ve3sjk](https://github.com/ve3sjk)
> 
> ```
> E (17) i2c.master: I2C hardware timeout detected
> E (17) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pul l-ups are correctly set up
> I2C device found at address 0x77
> ```
> 
> This has already been fixed on idf master. Becuase probe doesn't support stretch before.
> 
> And lots of following nack, I guess 0x77 is not correct.

Device 0x77 is a bme680 sensor, i moved it back to directly wired to breakout board from a quad about 4 foot long.

I have not seen the error since doing that so far been running for about 5 hours.

ve3sjk, 2025-04-10:
> [@ve3sjk](https://github.com/ve3sjk)
>
> ```
> E (17) i2c.master: I2C hardware timeout detected
> E (17) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pul l-ups are correctly set up [...]

Device 0x77 is a bme680 sensor, i moved it back to directly wired to breakout board from a quad about 4 foot long.

I have not seen the error since doing that so far been running for about 5 hours.

mythbuster5, 2025-04-10:
@MarksWizard I believe you can reproduce the issue but... I'm trying to reproduce the issue via TextBoxDemo. But every thing works well with S3-DevKit.

[image]

[image]

We also simply provide a ssd1306 example in esp-idf: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/i2c_oled

Thanks a lot for your effort, but more clue is very helpful. What about connect a logic analyzer and see data on line?

guy9ak, 2025-04-12:
> [@MarksWizard](https://github.com/MarksWizard) when you select a board, you will see two cores in the boards menu, one will say `ESP32 Arduino (in Sketchbook)` and that is the one you need to select board from. The one that says `esp32` is the one installed from Board Manager

@me-no-dev  If I select the one that says "esp32", the "LoRa_rx_tx.ino" sketch compiles and upload, but then I run into this issue (i2c.master...).  I'm using a Heltec LoRa 32 v3 and Arduino IDE v2.3.6
If I select the one that says "ESP32 Arduino (in Sketchbook)", it does not compile and gives me the following error:  "The system cannot find the path specified.  exit status 1".  Not very helful as it does not tell me which path is specified.
Thanks,

Guy

me-no-dev, 2025-04-14:
@guy9ak you probably did not execute `get.py` or `get.exe` as given by the instructions, so install did not pull the tools and libs required for Arduino

guy9ak, 2025-04-14:
> [@guy9ak](https://github.com/guy9ak) you probably did not execute `get.py` or `get.exe` as given by the instructions, so install did not pull the tools and libs required for Arduino

@me-no-dev 
Thanks for your reply!  Sorry, but I am not an advanced user.  Could you tell me where I could find the instructions that you are referencing?  Thanks.

me-no-dev, 2025-04-14:
@guy9ak [my comment](https://github.com/espressif/esp-idf/issues/15734#issuecomment-2786511690) in this same thread

MarksWizard, 2025-04-14:
Hi there, I built and uploaded the I2C_oled example from the ESP32 peripherals folder, but it doesn’t work either...  I had SDA on GPIO 1 and SCL on GPIO per your post.

Output:
`11:51:34.406 -> ���ESP-ROM:esp32s3-20210327

11:51:34.406 -> Build:Mar 27 2021

11:51:34.406 -> rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)

11:51:34.406 -> Saved PC:0x403759c8

11:51:34.406 -> SPIWP:0xee

11:51:34.406 -> mode:DIO, clock div:1

11:51:34.406 -> load:0x3fce2820,len:0x15d0

11:51:34.406 -> load:0x403c8700,len:0xd20

11:51:34.406 -> load:0x403cb700,len:0x2f60

11:51:34.406 -> entry 0x403c8924

11:51:34.439 -> I (29) boot: ESP-IDF v5.5-dev-1655-gc5865270b5-dirty 2nd stage bootloader

11:51:34.439 -> I (29) boot: compile time Apr 14 2025 11:06:09

11:51:34.439 -> I (29) boot: Multicore bootloader

11:51:34.439 -> I (32) boot: chip revision: v0.2

11:51:34.439 -> I (34) boot: efuse block revision: v1.3

11:51:34.439 -> I (38) boot.esp32s3: Boot SPI Speed : 80MHz

11:51:34.439 -> I (42) boot.esp32s3: SPI Mode       : DIO

11:51:34.471 -> I (46) boot.esp32s3: SPI Flash Size : 2MB

11:51:34.471 -> I (49) boot: Enabling RNG early entropy source...

11:51:34.471 -> I (54) boot: Partition Table:

11:51:34.471 -> I (56) boot: ## Label            Usage          Type ST Offset   Length

11:51:34.471 -> I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000

11:51:34.471 -> I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000

11:51:34.471 -> I (76) boot:  2 factory          factory app      00 00 00010000 00100000

11:51:34.504 -> I (82) boot: End of partition table

11:51:34.504 -> I (86) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=0efdch ( 61404) map

11:51:34.504 -> I (104) esp_image: segment 1: paddr=0001f004 vaddr=3fc92a00 size=01014h (  4116) load

11:51:34.504 -> I (105) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=49a54h (301652) map

11:51:34.570 -> I (161) esp_image: segment 3: paddr=00069a7c vaddr=3fc93a14 size=018f4h (  6388) load

11:51:34.570 -> I (163) esp_image: segment 4: paddr=0006b378 vaddr=40374000 size=0e90ch ( 59660) load

11:51:34.570 -> I (178) esp_image: segment 5: paddr=00079c8c vaddr=600fe000 size=0001ch (    28) load

11:51:34.570 -> I (184) boot: Loaded app from partition at offset 0x10000

11:51:34.601 -> I (184) boot: Disabling RNG early entropy source...

11:51:34.601 -> I (196) cpu_start: Multicore app
[... 150 lines cut ...]
11:39:46.442 -> Build:Mar 27 2021

11:39:46.442 -> rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)

11:39:46.442 -> SPIWP:0xee

11:39:46.442 -> mode:DIO, clock div:1

11:39:46.442 -> load:0x3fce2820,len:0x118c

11:39:46.442 -> load:0x403c8700,len:0x4

11:39:46.442 -> load:0x403c8704,len:0xc20

11:39:46.442 -> load:0x403cb700,len:0x30e0

11:39:46.442 -> entry 0x403c88b8

11:39:46.836 -> Initializing display

11:39:47.825 -> E (1068) i2c.master: I2C transaction unexpected nack detected

11:39:47.825 -> E (1068) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed`

[image]

MarksWizard, 2025-04-14:
@ve3sjk yeah definitely let me know what your testing turns up.  I am not sure where you see I have ESP32 selected, in my photo it clearly shows ESP32S3 in the port selection drop down and the lower right corner of the Arduino window?

In the example where I uploaded Arduino code, I selected pins 12 and 13 for the SCL/SDA and identify them in the software.  Prior to the update this code would have ran fine.  Just incase I was wrong though... I switched it to 8 for SDA, 9 for SCL and changed Wire.begin(SDA,SCL) to the default Wire.begin().  **It worked!!!**, switching back to Wire.begin(SDA,SCL) and the issue returns.  So it is definitely something up with trying to software select the I2C @mythbuster5 @me-no-dev

MarksWizard, 2025-04-14:
@mythbuster5 
Well I thought I had finally tracked down the bug to Wire.begin(SDA,SCL), but after a bit more testing that isn’t the issue.

I am now convinced that it is a wiring issue like @ve3sjk originally uncovered.  I don’t know what changed from my earlier testing, or this simple wiring, but I am now able to get the above Arduino code to run without issue on pins 12/13 for SDA/SCL.  If I disconnect either of those pins the issue returns.  So it’s great that it’s working again.. and I appreciate the message which informs us that there might be a wiring issue.  However, is there a way to keep it from completely spamming the serial console, like some sort of rate limiting timer on the output message?

HanksSaloon, 2025-04-16:
@mythbuster5 et al
In another thread the same problem has been discussed. @MarksWizard asked me to copy my comment from there, but I think a link is sufficient: https://github.com/espressif/arduino-esp32/issues/11228#issuecomment-2809209692

Some additional details: Arduino IDE is 2.3.6. I updated yesterday (from 2.3.4, jumped over 2.3.5), and at the same time updated the ESP32 board library from 3.1.3 to 3.2.0. The problems appeared immediately. but a downgrade to ESP32 ver 3.1.3 fixed it.
No other changes to any library, nor to the sketch itself.

MarksWizard, 2025-04-17:
@mythbuster5 if you want to trigger the issue for yourself upload the I2C_olef example and wire the screen normally like you did in your previous post.  Everything should run fine.

Now disconnect any wire from the screen to make it unreachable, and reboot the ESP.  You should see it enter the boot/crash loop I posted above.

mythbuster5, 2025-04-17:
https://github.com/user-attachments/assets/8fe444fd-74d3-4e25-9e57-6f969b12b6ab

> disconnect any wire from the screen to make it unreachable, and reboot the ESP.

Do you connect wire back? If no, the log you have posted is correct I have to say because you connect nothing.

Like this

[image]

If yes, I have recorded the vedio to show everything works here. Am I mistaken sth? @MarksWizard

MarksWizard, 2025-04-17:
@mythbuster5 No I don’t connect the wire back, that’s the point.  On my device, without the connected I2C device, the serial output is flooded with the transaction failed messages.  Check the image that I posted to see how fast the messages are coming in.  Other users are reporting the same flooding.  This was not the case previous to the recent update.

This causes a huge issue for me, because I can no longer do basic testing on code without having all of the I2C devices being present.  The error message floods the serial without some sort of rate limit or flag to stop it. Previously, if a sensor wasn’t present, it would just report a bad value which was typically constant depending on device.  As an example, if I am testing part of web server handling code, it doesn’t matter if I have a sensor connected and reading properly.

For my board using the I2C_oled example, the I2C device not being present causes the board to boot/crash.  That wasn’t the previous behavior, and shouldn’t be the default.

MarksWizard, 2025-04-17:
@mythbuster5 i forgot to say that I agree your board is working properly and the way I would like mine to work, receiving a single block of messages.  However, something is causing the flooding behavior on some boards.

MarksWizard, 2025-04-17:
@mythbuster5 

https://github.com/user-attachments/assets/fb08f335-eca5-41fb-bed9-976e53316af2

mythbuster5, 2025-04-18:
@MarksWizard Ok, I see. That is because when you reboot, the device is not connected to esp chip. So there should be a NACK error right? And in this example, (also others), we use `ESP_ERROR_CHECK(func...)` 

[image]

And inside the `ESP_ERROR_CHECK`, you will see

[image]

So abort here is a correct behavior. If you don't want to see abort, simply define the `NDEBUG` and you will see abort gone. Like this. (This is not the i2c behavior, it's common behavior in idf)

[image]

So, perhaps you are not facing critical issue. Right?

MarksWizard, 2025-04-18:
@mythbuster5 thank you for continuing to take the time to look into this.  I see now that you are correct with the abort being normal, although my output and your output in the test videos seem different.  Yours does not appear to enter the boot/crash loop that mine does.

The video below shows how the issue is manifesting for me.  The device I am uploading code to is complex with 2 motors, 2 flow sensors, 2 water level sensors, relays, air/temp sensor, water temp sensor, time of flight sensor.  If I want to test a small bit of code at my desk or at home, I use just a bare dev board for testing.   Under ESP-IDF 5.3, my code runs fine.  Even though screen and sensors are not connected, software catches that on the read and informs us.  I can still debug any code that doesn’t specifically require the I2C device.

https://github.com/user-attachments/assets/308afd18-ebdb-4262-9e60-385d2d62f657

However, under ESP-IDF 5.4, these error messages now spam the serial. If the I2C fails or disconnects after initialization, it will cause the board to freak out and eventually leads to boot/crash.  So all I really am asking for is a way simple way to turn these NACK messages off, such as VIEW_NACK_ERR = false.  Then my issue will literally disappear.  However, without this option I will need to check for the presence of the I2C before every read/write, which will still cause the error messages just fewer of them.  Furthermore, this check is not currently included in any publicly available libraries or examples.

https://github.com/user-attachments/assets/9c16b2b3-42b0-4070-99f0-016c0db2b428

chucklink, 2025-04-20:
My environment updated today.  I see exact same problem.  Boards are PCB with no wires and very clean design using ESP32- C6.  Been using this set of boards for months with any problems and all the sudden:

E (360388) i2c.master: I2C transaction unexpected nack detected
E (360388) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (360392) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed 

I am going to try to downgrade my libraries to see what fixes the problem.

chucklink, 2025-04-20:
Confirming downgrade fixes the problem.   Downgrading ESP32 3.2.0 to ESP32 3.1.3 fixes the problem.  No more messages like in previous comment.  Good luck and thanks in advance for keeping things going.

mythbuster5, 2025-04-21:
@chucklink May I know the sensor(s) you use. I want to reproduce this on my side.

MarksWizard, 2025-04-23:
@mythbuster5 do you have any response to my previous post? (https://github.com/espressif/esp-idf/issues/15734#issuecomment-2815474884)

me-no-dev, 2025-04-23:
@HanksSaloon here are the steps:
- Close Arduino IDE
- Download https://github.com/espressif/esp32-arduino-lib-builder/actions/runs/14618886394/artifacts/2994265315
- Execute the following script:
```bash
rm -rf ~/Arduino/hardware/espressif/esp32 && \
mkdir -p ~/Arduino/hardware/espressif && \
cd ~/Arduino/hardware/espressif && \
git clone https://github.com/espressif/arduino-esp32.git -b idf-master esp32 && \
cd esp32/tools && \
python3 get.py && \
rm -rf esp32-arduino-libs
```
- Unarchive the downloaded file and place the extracted folder `esp32-arduino-libs` into `~/Arduino/hardware/espressif/esp32/tools`
- Start Arduino IDE
- Open your sketch
- From the `Tools` menu, select `Board` -> `ESP32 Arduino (in Sketchbook)` -> Your desired board
  [image]
- Compile and upload the sketch

@MarksWizard you can add the following line to the beginning of `setup()` to turn off I2C logs
```cpp
esp_log_level_set("i2c.master", ESP_LOG_NONE);
```

@mythbuster5 the patch did not produce good results for me. Testing with SSD1306, it resulted in artifacts on the screen and some logs in the console

me-no-dev, 2025-04-23:
> Please let me know if there is some way I can tip you, send you a gift card, anything to show how appreciative I truly am.

Thanks, but that is not needed :)

mythbuster5, 2025-04-24:
@HanksSaloon  Thank you for your reply and affirmation，but I still want to find the root cause then. Now what we can almost confirm that shorten the time between last data and stop works. But follow the i2c protocol, this time doesn't matter at all. As a synchronous protocol what slave should do is just follow the master scl clock. So my question is about do you know which device really nack in this condition? i want to go through its doc and figure out why.

HanksSaloon, 2025-04-24:
@mythbuster5 If the only change you’ve done so far is shortened the time between last data and stop, then I fully agree: the root cause is yet to be found and understood.

That also makes me think, that there might be something wrong in the sketch and/or device drivers (i.e. libraries).
What I do not understand fully, yet, is that when I run a very simple I2C-scanning sketch, (which scans both I2C channels in this particular hardware setup), I find three addresses that I cannot explain
(The comments in parentheses are my additions for clarification):

...
Scanning I2C channel 1
I2C device found at address 0x1A 		(47L16 external RAM, Control register)
I2C device found at address 0x21 		(PCF8574 => connects a 4x20 LCD display)
I2C device found at address 0x48 		(ADS7830 8ch AD-converter)
I2C device found at address 0x50 		(24CSM01 external EEPROM)
I2C device found at address 0x51 		(PCF8563T Real Time Clock)
I2C device found at address 0x52 		(47L16 external RAM, SRAM array)
I2C device found at address 0x58 		(unknown responder)
I2C device found at address 0x59 		(unknown responder)
I2C device found at address 0x65 		(ACS37800 Power Monitor)
I2C device found at address 0x7C 		(unknown responder)
done

...
Scanning I2C channel 2
I2C device found at address 0x20 		(MCP23008)
done

The mystery thing are the three unknown responses. I have looked through the data sheets, but so far I have not found who they (might) come from.
However, my sketch does not use any of these unknown addresses (and not the address 0x1A, either). Therefore these addresses should never appear on the channels, unless they are not hidden in some of the libraries. (Searching through the libraries for these addresses does not produce any hits).
Most libraries are quite simple and easy, the only exception is the library for ACS37800, which is a very versatile and complex circuit. Looking for strange behavior or maybe hidden features, I would perhaps start there.

On the other hand, I might have a ticking issue with the EEPROM address (0x50), and am considering shifting to another address (0x56). Requires some hw tweaking...

I have tried to find that ”failing NACK”, but so far no success. There’s thousands and thousands of traces from the analyzer to walk through. It is a little bit tricky, though, as now it works. I need to find a way to go back to the setup that did not work, without loosing this setup. I’ll see what can be done and I’ll keep you up informed on any progress.

In all this thinking I do find it disturbing, though, that with exactly the same HW and SW setup, and the only difference being ESP32-library version 3.1.3 vs version 3.2.0, the other worked, but the latter not, and I still can’t explain why.

HanksSaloon, 2025-04-24:
@mythbuster5 A quick update. As mentioned in the post above, I changed the address of the EEPROM to 0x56. Now the scan looks as follows:
...
Scanning I2C channel 1
I2C device found at address 0x1A  ! (47L16 external RAM, Control register)
I2C device found at address 0x21  ! (PCF8574 => LCD display)
I2C device found at address 0x48  ! (ADS7830 8ch AD-converter)
I2C device found at address 0x51  ! (PCF8563T Real Time Clock)
I2C device found at address 0x52  ! (47L16 external RAM, SRAM)
I2C device found at address 0x56  ! (24CSM01 external EEPROM)
I2C device found at address 0x57  ! (24CSM01 external EEPROM with A16 set)
I2C device found at address 0x5E  ! (=> EEPROM)
I2C device found at address 0x5F  ! (=> EEPROM)
I2C device found at address 0x65  ! (ACS37800 Power Monitor)
I2C device found at address 0x7C  ! (unknown)
done

...
Scanning I2C channel 2
I2C device found at address 0x20  ! (MCP23008)
done

The new scan also revealed, that two of the mystery addresses belongs to the EEPROM. I have to look into that.
The third ghost address is still a mystery, have to look through the datasheets more carefully at some point.

I also managed to re-create the "issue". 
First I changed the hardware address of the EEPROM and the started up the sketch, and then a lot of errors were thrown on the serial monitor.
As soon as I had updated the sketch, too, they disappeared and the sketch work as expected.

Conclusion: The environment in my project is somewhat complex with many different devices and both known and unknown addresses. Perhaps this caused, or at least was part of the problem in the first place(?)
Even so, I still cannot explain why version 3.1.3 worked fine, but version 3.2.0 caused problems, when there were absolutely no other changes.

ve3sjk, 2025-04-25:
This is probally not related to the I2C error codes but i do find i am seeing errors on device that have been working for years. The latest unknow error is this one.

`11:59:10.743 -> ets Jul 29 2019 12:21:46
11:59:10.743 -> 
11:59:10.743 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11:59:10.743 -> configsip: 0, SPIWP:0xee
11:59:10.743 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:59:10.743 -> mode:DIO, clock div:1
11:59:10.743 -> load:0x3fff0030,len:4888
11:59:10.788 -> load:0x40078000,len:16516
11:59:10.788 -> load:0x40080400,len:4
11:59:10.788 -> load:0x40080404,len:3476
11:59:10.788 -> entry 0x400805b4
11:59:11.104 -> 
11:59:11.104 -> MBUS BASE INIT COMPLETE
11:59:11.104 -> E (21) rmt: rmt_new_tx_channel(269): not able to power down in light sleep
11:59:11.148 -> CHAN:	4
11:59:11.148 -> done...`

I have no idea what the rmt error is have never seen it before.

Here is the code being executed in the MBUS BASE INIT

`  #ifdef MBUS_BASE  
    SerialPort.begin(9600, SERIAL_8N1, 25, 26);
    node.begin(1, SerialPort);
    Serial.println("MBUS BASE INIT COMPLETE");
  #endif`

electronicsguy, 2025-05-08:
I'm getting the same error on ESP_IDF 5.4.0, esp32s3 board, SSD1306

`I (266) main_task: Started on CPU0
I (276) main_task: Calling app_main()
I (276) example: Initialize I2C bus
I (276) example: Install panel IO
I (276) example: Install SSD1306 panel driver
E (286) i2c.master: I2C hardware NACK detected
E (286) i2c.master: I2C transaction unexpected nack detected
E (296) i2c.master: s_i2c_synchronous_transaction(918): I2C transaction failed
E (296) i2c.master: i2c_master_multi_buffer_transmit(1180): I2C transaction failed
E (306) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(193): i2c transaction failed
E (316) lcd_panel.ssd1306: panel_ssd1306_init(151): io tx param SSD1306_CMD_SET_MULTIPLEX failed
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x42009e4e`

me-no-dev, 2025-06-10:
@mythbuster5 we are having more and more issues with I2C. Especially when there are more than one devices on the network. Let's have a talk and find a way to resolve those. The proposed patch fixes it only for some devices

mythbuster5, 2025-06-24:
No clear result. Some of idf users adjust pull-up registers can solve similar problems. We still keep an eye on it.

hitmangy, 2025-07-12:
any updates to resolve this issue !

thadius83, 2025-08-15:
> For me, I needed to set my pins
> 
> `Wire.setPins(3, 4); `
> 
> before
> 
> `Wire.begin()`
> 
> and then it worked ...

@Casey10110 that’s not using the new i2c master implementation.

ndastur, 2025-12-12:
Came across this issue using a SEEED XAIO C6 device and the 0.96 OLED display.
```  Wire.setPins(22, 23); ``` before ```display.begin(...)``` fixes the error. Maybe this helps