espressif/esp-idf #15737: Execute a docker command in the same shell

adrianpanicek, 2025-04-08:
### Is your feature request related to a problem?

When I'm trying to use Linaro LAVA testing environment it hooks into docker's shell. But this shell only runs entrypoint and then executes command on a separate shell via "exec". 

Because the current implementation of LAVA can't follow shell change it will stop after entrypoint and assumes everything on the docker finished executing.

### Describe the solution you'd like.

I'd like to add environment variable to docker image to set whether we want to execute docker command on separate or on the same shell as entrypoint

fhrbata, 2025-04-09:
Hello @adrianpanicek ,

isn't it possible to overwrite the entrypoint with the `--entrypoint` option and start whatever command you need in a way you want?

```
docker run --entrypoint /bin/bash --rm -it --platform linux/x86_64 espressif/idf:latest -c '. $IDF_PATH/export.sh && echo "whatever command"'
````

Maybe this can help with your use case.

Thank you

adrianpanicek, 2025-04-09:
Hi @fhrbata 

thank you for your recommendation. This was the first thing I considered when debugging this error. However I consider entrypoint to be an "internal API" and overloading it, might mean that in case of any change of entrypoint, I would need to modify my command accordingly.

That's what I wanted to avoid for example your suggested solution doesn't include "IDF_GIT_SAFE_DIR" logic, which might make behavior of the container inconsistent. 

In this case this might be easily debuggable problem, but I'd like to avoid having to track changes to entrypoint.

Thank you and have a nice day

fhrbata, 2025-04-15:
Hello @adrianpanicek ,

my apologies for the delayed reply. 

> That's what I wanted to avoid for example your suggested solution doesn't include "IDF_GIT_SAFE_DIR" logic, which might make behavior of the container inconsistent.

Yes, I understand. Using `--entrypoint` seems more like a workaround.

I have no experience with LAVA so I'm probably missing something. I was wondering what do you mean by this

> When I'm trying to use Linaro LAVA testing environment it hooks into docker's shell. 

I tried to set the LAVA up with [docker-compose](https://gitlab.com/lava/pkg/docker-compose) and following [docker as a DUT setup](https://lava.readthedocs.io/en/latest/admin/basic-tutorials/device-setup/docker/). After few hiccups I managed to get it working. And it seems to be working for me with the default esp-idf docker image.

Here is the job definition I used. It's based on the [simple test job](https://lava.readthedocs.io/en/latest/admin/basic-tutorials/device-setup/docker/#submit-a-job) referenced in the docker as a DUT setup](https://lava.readthedocs.io/en/latest/admin/basic-tutorials/device-setup/docker/). I only adjusted the [timeouts](https://docs.lavasoftware.org/lava/actions-timeout.html). First I thought that there might be a problem with [boot prompt](https://docs.lavasoftware.org/lava/actions-boot.html?utm_source=chatgpt.com#boot-prompts), but it doesn't seem to be the case. The default `root@lava:/` based on the hostname passed to docker seem to be working fine.
```
device_type: docker
job_name: simple docker test

timeouts:
  job:
    minutes: 60
visibility: public
priority: medium

actions:
- deploy:
    to: docker
    image: espressif/idf:latest
    os: debian

- boot:
    method: docker
    command: bash
    prompts: ['root@lava:/']
- test:
    timeout:
      minutes: 30
    definitions:
    - from: inline
      name: env-dut-inline
      path: inline/env-dut.yaml
      repository:
        metadata:
          format: Lava-Test Test Definition 1.0
          name: smoke-tests-basic
        run:
          steps:
          - cd /opt/esp/idf/examples/get-started/hello_world/
          - idf.py build
```

Here is the output from my testing example. It just builds the `hello_world` example

[image]

I'm not sure if I'm doing something differently or if you are experiencing different problem, but I'm not sure that it's related to the `exec` call in the `entrypoint.sh`.

Thank you

dobairoland, 2025-04-28:
I'm closing this because it looks this is not an issue.

closed 2025-04-28 by dobairoland