espressif/esp-idf #15739: ESP32-MINI-1 module I2S not work.

CMCCAdmin, 2025-04-08:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm having trouble running I2S on ESP32-MINI-1 and can't receive data from the I2S RX channel and I'm having the same fault with both PCBs I'm soldering. I tried changing the ESP32 I2S role to Master or Slave but it didn't work. Works well on a ESP32S3-MINI-1 module with the same code. In order to rule out that the I2S is not working due to the poor contact of the module, I re-soldered the module and tested it, controlled the high and low level of the GPIO output corresponding to the I2S, and the test results showed that the pin contact was normal.

IDF Version: ESP-IDF v5.4.1-208-g17a2461297
Chip Version: ESP32-U4WDH (revision v3.1)

`#define I2S_WS_PIN      GPIO_NUM_27
#define I2S_BCLK_PIN    GPIO_NUM_26
#define I2S_SDATA_PIN   GPIO_NUM_14

void i2s_init()
{
    i2s_chan_config_t chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_0, I2S_ROLE_SLAVE);
    chan_cfg.dma_frame_num = 320;
    chan_cfg.intr_priority = 5;
    /* dma_buffer_size = dma_frame_num * slot_num * slot_bit_width / 8 */
    ESP_ERROR_CHECK(i2s_new_channel(&chan_cfg, NULL, &i2s_rx_handle));
    i2s_std_config_t std_cfg = {
        .clk_cfg =
        {
            .clk_src = I2S_CLK_SRC_DEFAULT,
            .sample_rate_hz = 32000U,
            .mclk_multiple = I2S_MCLK_MULTIPLE_256,
        },
        .slot_cfg = I2S_STD_MSB_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO),
        .gpio_cfg = 
        {
            .mclk = I2S_GPIO_UNUSED,
            .bclk = I2S_BCLK_PIN,
            .ws = I2S_WS_PIN,
            .dout = I2S_GPIO_UNUSED,
            .din = I2S_SDATA_PIN,
            .invert_flags =
            {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
            },
        },
    };
    /* 初始化通道 */
    ESP_ERROR_CHECK(i2s_channel_init_std_mode(i2s_rx_handle, &std_cfg));
    i2s_event_callbacks_t const i2s_event_cb =
    {
        .on_recv = i2s_on_recv,
        .on_recv_q_ovf = NULL,
        .on_sent = NULL,
        .on_send_q_ovf = NULL,
    };
    ESP_ERROR_CHECK(i2s_channel_register_event_callback(i2s_rx_handle, &i2s_event_cb, NULL));
    ESP_ERROR_CHECK(i2s_channel_enable(i2s_rx_handle));
}
`

`D (645) i2s_common: rx channel is registered on I2S0 successfully
D (647) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 1280
D (658) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 19 [mclk] 8192000 Hz [bdiv] 8 [bclk] 1024000 Hz
D (668) i2s_std: The rx channel on I2S0 has been initialized to STD mode successfully
D (675) i2s_common: i2s rx channel enabled`

suda-morris, 2025-04-09:
`chan_cfg.intr_priority = 5;` => `chan_cfg.intr_priority = 0;`

Also please refer to the I2S STD Slave test case in the esp-idf:  https://github.com/espressif/esp-idf/blob/a25e7ab59ed197817d4a78e139220b2707481f67/components/esp_driver_i2s/test_apps/i2s/main/test_i2s.c#L636

CMCCAdmin, 2025-04-09:
w

> `chan_cfg.intr_priority = 5;`=>`chan_cfg.intr_priority = 0;`
> 
> 另请参阅 esp-idf 中的 I2S STD Slave 测试用例：
> 
> [esp-idf/组件/esp_driver_i2s/test_apps/i2s/main/test_i2s.c](https://github.com/espressif/esp-idf/blob/a25e7ab59ed197817d4a78e139220b2707481f67/components/esp_driver_i2s/test_apps/i2s/main/test_i2s.c#L636)
> 
> 636 行 [a25e7ab](/espressif/esp-idf/commit/a25e7ab59ed197817d4a78e139220b2707481f67)
> 
>  测试用例( “I2S_master_write_slave_read_test” , “[i2s]” )

Thanks, the problem is solved.

closed 2025-04-09 by CMCCAdmin