espressif/esp-idf #15744: WiFi task consumes a lot of internal RAM( nearly 90KB)

aygh4266, 2025-04-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello everyone,

WiFi task consumes a lot of internal RAM( nearly 90KB), so other tasks can't be created correctly.

I tried to allocate all static and global variable to bss. Other Objects are allocated dynamically with `heap_caps_malloc`. However wifi still consumes a lot of RAM and I still have 7 MB PSRAM free.
I tried to find the implementation of esp_wifi_start and change the `xTaskCreatePinnedToCore` to `xTaskCreatePinnedToCoreWithCaps` manually but I didn't find the implementaiton of the API.

Any other advices will be appreciated.

hansw123, 2025-04-11:
@aygh4266   sorry for late reply 
if chip has PSRAM , the tx and some other buffer will be set static defautly,  for example  , a static tx buffer will take 1.6k  size. so can you provide your sdkconfig?   In default case . the static tx buffer = 32  = 32*1.6 k,  the tx cache buffer aslo is static  32*1.6  , static rx buffer 16*1.6k and others .
so your case is ok

hansw123, 2025-04-11:
@aygh4266 
1. CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP  is y ,this config will Try to allocate memories of WiFi and LWIP in SPIRAM firstly, so ,the buffers  will not take internal ram size if psram is enough.
2. if you enale wpa3 ,wifi stack size is 6k,    disable  wap3 is 3k
3. CONFIG_ESP32_WIFI_IRAM_OPT and CONFIG_ESP32_WIFI_RX_IRAM_OPT  will put some important funcs in ram , this wil also take some ram size,  but disable this configs will low wifi performance
4. can you describe how you get the size of wifi task consmes  internal ram?

aygh4266, 2025-04-11:
> 4. can you describe how you get the size of wifi task consmes  internal ram?

Thank you very much. I am logging the free heap before starting wifi and after it 

```
ESP_LOGI(TAG, "Free internal heap after WLAN task: %d", heap_caps_get_free_size(MALLOC_CAP_INTERNAL));
ESP_LOGI(TAG, "Free PSRAM heap after WLAN task: %d", heap_caps_get_free_size(MALLOC_CAP_SPIRAM));
```

hansw123, 2025-04-15:
@aygh4266  you  can just test 

add internal free heap log
esp_wifi_init()
esp_wifi_start()

delay()
add internal free heap log

esp_Wifi_stop()
esp_wifi_deinit()

delay()
add internal free heap log

these test can see the true iram wifi task take. it can not be 90k

closed 2025-06-30 by Alvin1Zhang