espressif/esp-idf #15747: Unable to Build Local Docs for ESP-IDF v5.5-dev with sphinx-build

geng-Max, 2025-04-09:
I found requirements.txt in the ESP-IDF repository and installed it with pip3 install -r requirements.txt. However, I don't know what to do next. I tried running sphinx-build -b html . _build/html in the docs/zh_CN directory, but it didn't build successfully. Here's the error I got:
`Extension error (esp_docs.idf_extensions.build_system):
Handler <bound method IdfBuilder.generate_idf_info of <esp_docs.idf_extensions.build_system.IdfBuilder object at 0x7fb201d4d9a0>> for event 'config-inited' threw an exception (exception: expected str, bytes or os.PathLike object, not NoneType)`

- ESP-IDF Version: v5.5-dev-2916-g1c468f6825 (commit 1c468f6825)

- OS: WSL (Ubuntu)

- Sphinx Version: 7.1.2

What should I do to build the Chinese documentation locally?

geng-Max, 2025-04-09:
I found the build-docs --help in the docs/ directory's README, but when I try to run the command build_docs.py [-h] [--language {en,zh_CN}] [--target {esp8266,esp32,...}], I get the error build_docs.py: command not found.

geng-Max, 2025-04-09:
I tried running the following command in the ~/esp-idf directory to find the build_docs.py file:
`find ~/esp-idf -name "build_docs.py"`
But I got no results. Is the file missing, or has it been removed in the latest version of ESP-IDF?

Alvin1Zhang, 2025-04-10:
Thanks for reporting.

igrr, 2025-04-10:
> I noticed that the official documentation downloads provided only go up to v4.1

By "downloads", do you mean zip files with HTML documentation, or PDF files?

geng-Max, 2025-04-10:
> > I noticed that the official documentation downloads provided only go up to v4.1
> 
> By "downloads", do you mean zip files with HTML documentation, or PDF files?

PDF file. What I want is the HTML zip file, but the HTML zip file version is older."

KonstantinKondrashov, 2025-04-10:
Hi @geng-Max!
[How to Build Documentation Locally](https://docs.espressif.com/projects/esp-docs/en/latest/building-documentation/building-documentation-locally.html#building-documentation-locally).

1. Install `Doxygen
**sudo apt-get install doxygen**

2. The `--enable-docs`` will install the ``esp-docs`` Python package. This package is a wrapper around `Sphinx <https://www.sphinx-doc.org/>`_ and is required to build ESP-IDF documentation.
 **./install.sh --enable-docs && . ./export.sh**

3. You could clean the esp-idf repo (may help to avoid some errors):
git clean -d -f -f -X && git clean -ffd.

4. Build the zh_CN doc for esp32
**cd ~/esp/esp-idf/docs**
**build-docs -t esp32 -l zh_CN build**

geng-Max, 2025-04-10:
@KonstantinKondrashov 
Thanks for the reply! I've successfully implemented the local documentation!

closed 2025-04-11 by KonstantinKondrashov