espressif/esp-idf #15748: HTTP client fails after "Connection timed out before data was ready"

gustavowd, 2025-04-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using ESP32-C3 with rust in standard mode, with esp-idf 5.2.3 providing the HTTP client functions. On several occasions I received timeout errors. For example, "Connection timed out before data was ready".

The error itself is not a problem, but after it occurs, all other requests to the http client fail. I solved the problem by adding a call to close the connection (**esp_http_client_close(client)**) in these cases and it worked. If you agree, I can make a pull request with the fix.

`int64_t esp_http_client_fetch_headers(esp_http_client_handle_t client)
{
    if (client->state < HTTP_STATE_REQ_COMPLETE_HEADER) {
        return ESP_FAIL;
    }

    client->state = HTTP_STATE_REQ_COMPLETE_DATA;
    esp_http_buffer_t *buffer = client->response->buffer;
    client->response->status_code = -1;

    while (client->state < HTTP_STATE_RES_COMPLETE_HEADER) {
        errno = 0;
        buffer->len = esp_transport_read(client->transport, buffer->data, client->buffer_size_rx, client->timeout_ms);
        if (buffer->len <= 0) {
            if (buffer->len == ERR_TCP_TRANSPORT_CONNECTION_TIMEOUT) {
                ESP_LOGW(TAG, "Connection timed out before data was ready");
                esp_http_client_close(client);
                return -ESP_ERR_HTTP_EAGAIN;
            }
            return ESP_FAIL;
        }`

`int esp_http_client_read(esp_http_client_handle_t client, char *buffer, int len)
{
    esp_http_buffer_t *res_buffer = client->response->buffer;

    int rlen = ESP_FAIL, ridx = 0;
    if (res_buffer->raw_len) {
        int remain_len = client->response->buffer->raw_len;
        if (remain_len > len) {
            remain_len = len;
        }
        memcpy(buffer, res_buffer->raw_data, remain_len);
        res_buffer->raw_len -= remain_len;
        res_buffer->raw_data += remain_len;
        ridx = remain_len;
        if (res_buffer->raw_len == 0) {
            esp_http_client_cached_buf_cleanup(res_buffer);
        }
    }
    int need_read = len - ridx;
    bool is_data_remain = true;
    while (need_read > 0 && is_data_remain) {
        if (client->response->is_chunked) {
            is_data_remain = !client->is_chunk_complete;
        } else {
            is_data_remain = client->response->data_process < client->response->content_length;
        }
        ESP_LOGD(TAG, "is_data_remain=%d, is_chunked=%d, content_length=%"PRId64, is_data_remain, client->response->is_chunked, client->response->content_length);
        if (!is_data_remain) {
            break;
        }
        int byte_to_read = need_read;
        if (byte_to_read > client->buffer_size_rx) {
            byte_to_read = client->buffer_size_rx;
        }
        errno = 0;
        rlen = esp_transport_read(client->transport, res_buffer->data, byte_to_read, client->timeout_ms);
        ESP_LOGD(TAG, "need_read=%d, byte_to_read=%d, rlen=%d, ridx=%d", need_read, byte_to_read, rlen, ridx);

        if (rlen <= 0) {
            if (errno != 0) {
                esp_log_level_t sev = ESP_LOG_WARN;
                /* Check for cleanly closed connection */
                if (rlen == ERR_TCP_TRANSPORT_CONNECTION_CLOSED_BY_FIN && client->response->is_chunked) {
                    /* Explicit call to parser for invoking `message_complete` callback */
                    http_parser_execute(client->parser, client->parser_settings, res_buffer->data, 0);
                    /* ...and lowering the message severity, as closed connection from server side is expected in chunked transport */
                    sev = ESP_LOG_DEBUG;
                }
                ESP_LOG_LEVEL(sev, TAG, "esp_transport_read returned:%d and errno:%d ", rlen, errno);
            }

            if (rlen == ERR_TCP_TRANSPORT_CONNECTION_TIMEOUT) {
                ESP_LOGD(TAG, "Connection timed out before data was ready");
                esp_http_client_close(client);
                /* Returning the number of bytes read upto the point where connection timed out */
                if (ridx) {
                    return ridx;
                }`

nileshkale123, 2025-04-10:
Hello @gustavowd,

A timeout occurs when the network delay exceeds the client's configured timeout. In such cases, the data read should ideally resume from where it left off before the timeout.

You can try increasing the timeout duration as a first step. If the issue still persists, please share the detailed reproduction steps or a patch, so we can investigate it further.

gustavowd, 2025-04-10:
Hello @nileshkale123.
I understand that a timeout occurs when the network delay exceeds the timeout configured by the client. However, no matter what value I set for timeout, at some point this problem occurs. Since I'm using the HTTP client through the Rust wrapper, I basically only have the GET, POST, PUT, HEAD, DELETE, etc. methods. If I try to make a new request after a timeout occurs, in some cases the request works, but it returns the content of the previous request. And in some cases the connection is closed with an unrecoverable error. The only way I managed to solve the problem was by closing the connection and returning the error. This way the HTTP client reconnects to the server on the next request.

Reproducing the problem is a bit difficult, because I'm accessing a REST API on our University's private network, so an external client wouldn't be able to test it. But basically it's a query every 15 seconds for 3 REST API routes in sequence, and the returned content is XML.

In fact, a REST API client ideally should connect and disconnect with each request, but I understand that when we use TLS on an ESP32 the connection can take a while (in my case around 1.5 seconds) and so the client maintains the connection. I believe that perhaps the server closes the connection after a while, since there may be bursts of connections and the server closes the oldest connections. Maybe this is the timeout problem.

Therefore, closing the connection when an error occurs seems like a viable option to me. Perhaps there could be a configuration that defines the behavior in case of timeout for the client, such as a flag to request the connection to be closed or not in case of timeout. How does this look to you?

Best regards.

AxelLin, 2025-04-15:
> Hello [@nileshkale123](https://github.com/nileshkale123). I understand that a timeout occurs when the network delay exceeds the timeout configured by the client. However, no matter what value I set for timeout, at some point this problem occurs. 

+1, I do observe this sometimes. Increasing timeout doe not help.

nileshkale123, 2025-04-18:
Hello @gustavowd ,

Have you enabled the keep-alive header in the config? As per your observation, the server might release the client connection if there is no communication for a specific time. Please try enabling the keep-alive header and let me know if the issue persists.

You can provide struct as per your requirenment in config for esp_http_client_t. [here](https://github.com/espressif/esp-idf/blob/master/components/esp_http_client/esp_http_client.c#L140)

gustavowd, 2025-04-22:
Hello @nileshkale123 ,

Unfortunately, the Rust http client does not allow you to modify this configuration. I generated a print of the default configuration imposed by the Rust http client.

esp_http_client_config_t { url: 0x3c154284, host: 0x0, port: 0, username: 0x0, password: 0x0, auth_type: 0, path: 0x0, query: 0x0, cert_pem: 0x0, cert_len: 0, client_cert_pem: 0x0, client_cert_len: 0, client_key_pem: 0x0, client_key_len: 0, client_key_password: 0x0, client_key_password_len: 0, tls_version: 0, user_agent: 0x0, method: 0, timeout_ms: 0, disable_auto_redirect: false, max_redirection_count: 0, max_authorization_retries: 0, event_handler: Some(0x42038170), transport_type: 0, buffer_size: 0, buffer_size_tx: 0, user_data: 0x3fcb790c, is_async: false, use_global_ca_store: false, skip_cert_common_name_check: false, common_name: 0x0, crt_bundle_attach: None, keep_alive_enable: false, keep_alive_idle: 0, keep_alive_interval: 0, keep_alive_count: 0, if_name: 0x0, ds_data: 0x0 }

As you can see, keep alive is disabled. Do you think setting the keep_alive_enable variable to true would solve the problem? I can try to do this in a fork.

nileshkale123, 2025-04-23:
Hello @gustavowd , 

Logically, it must solve the problem. You can try once with the right configuration, and please let me know the results. We will debug further if the issue isn't solved.

gustavowd, 2025-04-23:
Hello @nileshkale123 

Unfortunately, nothing has changed. The timeout problem continues to occur and, in some cases, after the timeout fails, the http client stops working. Closing the connection during the timeout, as I suggested, is still the only alternative that solves the problem.

AxelLin, 2025-04-24:
> Hello [@nileshkale123](https://github.com/nileshkale123)
>
> Unfortunately, nothing has changed. The timeout problem continues to occur and, in some cases, after the timeout fails, the http client [...]

Hi @gustavowd 

If you can easily reproduce the issue, please capture the wireshark log.

gustavowd, 2025-04-24:
Hi @AxelLin ,

The problem with using the Wireshark is that the communication is occurring between an esp32-c3 and a cloud REST API server, so the communication traffic is not passing through my PC. I can try running a local instance of the REST API on my machine to check this.

However, I don't see any major problems with closing the connection in case of timeout. In fact, the problem always occurs at one point, when the esp_http_client_fetch_headers() function timed out, which could be seen as a critical error that would require closing the connection.

gustavowd, 2025-04-24:
The first retransmissions and TCP Dup ACK coincide with the header timeout.

[image]

gustavowd, 2025-04-24:
And here:

[image]

I have this error:
E (352219) transport_base: poll_write select error 104, errno = Connection reset by peer, fd = 54
W (352220) transport_base: Poll timeout or error, errno=Success, fd=54, timeout_ms=1500
E (352227) HTTP_CLIENT: Error write request

which is not a problem, cause in the next request everything is working. The problem only occurs when the connection timed out.

nileshkale123, 2025-05-06:
Hello @gustavowd,

Closing connection in API esp_http_client_read() when error ERR_TCP_TRANSPORT_CONNECTION_TIMEOUT occur will not be acceptable solution as it might arise new issues in OTA procedure. I will recommend you to close the connection on your side instead.

gustavowd, 2025-05-06:
Hi @nileshkale123 ,

I understand the motivation of not closing the connection in the esp_http_client_read(). However, most of the problem occurs in the esp_http_client_fetch_headers(). I believe that in this function there would be no problem closing the connection. What do you think?

When you talk about closing the connection on my side, would that mean closing the connection when the client returns an error in my application code?

AxelLin, 2025-05-06:
@nileshkale123 

Can you reproduce the symptom?
Is it expected behavior that retransmissions and TCP Dup ACK will cause fetching header timeout.
(It looks not normal to me.)

AxelLin, 2025-05-19:
@nileshkale123 
I hit this issue when test on v5.5 branch:
In esp_http_client_fetch_headers:
W (65299636) HTTP_CLIENT: Connection timed out before data was ready!

It's not easy to reproduce this issue in my test, only occassionally happened.
So it's difficult to get more debug information.

Do you have any suspect about this symptom?
Any idea to clarify the root cause of this issue?

AdityaHPatwardhan, 2025-05-20:
Hi @gustavowd,
Even in esp_http_client_fetch_headers(), connection cannot be closed just after connection timeout by the API itself.. Because connection timeout doesn't necessarily imply that the connection needs to be closed. Hence, we return `-ESP_ERR_HTTP_EAGAIN` as the error code. After that it is the applications responsibility to either close the connection or try again.

I understand your problem that there is no way for you to call esp_http_client_close() as I assume that your RUST Wrapper handles manages the connection lifecycle internally and doesnt expose any public interface for you to do so.

Instead of updating esp_http_client, Should the Rust Wrapper be updated to close the connection in case of timeout?

@AxelLin,
Regarding the issue,

> I believe that perhaps the server closes the connection after a while, since there may be bursts of connections and the server closes the oldest connections. Maybe this is the timeout problem.

I would assume the root cause of the issue in this case to be same. Alternatively, It could be that there is a network failure and no data was received in the configured timeout.

AxelLin, 2025-05-21:
> I would assume the root cause of the issue in this case to be same. Alternatively, It could be that there is a network failure and no data was received in the configured timeout.

@AdityaHPatwardhan 

Did you check the wireshark log?
Why retransmissions and TCP Dup ACK will cause fetching header timeout?
https://github.com/espressif/esp-idf/issues/15748#issuecomment-2828651821

AdityaHPatwardhan, 2025-05-26:
@AxelLin, it is hard to diagnose the problem without being able to reproduce it.
My guess is that due to poor network, TCP retransmission is taking place. The HTTP layer is still waiting for the headers so it then gets timed out.
While that is happening, probably the server has closed the connection without informing the client so once again requesting on the same connection is met with bad data. 
Please let me know if you have any other ideas

AxelLin, 2025-05-26:
@AdityaHPatwardhan 

Don't guess, you can ask for the raw wireshark log for analysis (Looks like @gustavowd can reproduce it).

AdityaHPatwardhan, 2025-06-11:
@gustavowd As mentioned before, we cannot accept the solution proposed here.
If you are able to reproduce it and can offer more data, please do so. We can investigate it further
Thanks

AdityaHPatwardhan, 2025-06-17:
I will go ahead and close this issue, as the change is not acceptable and the relevant solution is proposed.
Please feel free to re-open if the issue persists.
Thanks