espressif/esp-idf #15749: Switch between Synchronous and Asynchronous I2C modes in master mode don't work!

DushkaF, 2025-04-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### Espressif SoC revision.

ESP32 (wroom32)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 DevKit

### Power Supply used.

USB

### What is the expected behavior?

When using a function `i2c_slave_register_event_callbacks ` with an argument `NULL` , asynchronous mode must be disabled.

In docs https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2c.html
> User can deregister a previously registered callback by calling this function and setting the callback member in the cbs structure to NULL.

### What is the actual behavior?

I need to first configure the sensors in synchronous mode, since there are several other sensors configured through the _path-through_ of one main sensor (and in asynchronous mode it is impossible to service several sensors!), and then switch to asynchronous mode to read the only main sensor FIFO with other sensors data.

My I2C settings:
```
i2c_master_bus_config_t i2c_master_config;
  i2c_master_config.clk_source = I2C_CLK_SRC_DEFAULT; 
  i2c_master_config.i2c_port = I2C_NUM_0;           
  i2c_master_config.scl_io_num = 22;                
  i2c_master_config.sda_io_num = 21;                
  i2c_master_config.flags.enable_internal_pullup = 1; 
  i2c_master_config.glitch_ignore_cnt = 7;           
  i2c_master_config.intr_priority = 0;                
  i2c_master_config.trans_queue_depth = 32; // for usage in Synchronous and Asynchronous modes

  i2c_master_bus_handle_t bus_handle;
  ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_master_config, &bus_handle)); 
  ESP_LOGI(logTAG, "I2C bus is configured");
```

Device configuration:
```
i2c_device_config_t i2c_device_config;
  i2c_device_config.dev_addr_length = I2C_ADDR_BIT_LEN_7;
  i2c_device_config.device_address = 0x68;  
  i2c_device_config.scl_speed_hz = 100000;   
  i2c_device_config.scl_wait_us = 0;               
  i2c_device_config.flags.disable_ack_check = 0;  

  i2c_master_dev_handle_t dev_handle;
  ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &i2c_device_config, &dev_handle));
```

This is what I use to disable the callback function and go to the synchronous

`i2c_master_register_event_callbacks(dev_handle, NULL, NULL);`

Then I saw that `ESP_ERROR_CHECK(i2c_master_receive(dev_handle, &bufData[0], 7, -1));` is a non-blocking function and the second device is not working! So I don't see synchronous mode!

Obviously, when I switch to asynchronous mode later, assigning a callback function, everything works. But I don't have the sensors configured. This is no good.

### Steps to reproduce.

1. Set up the I2C bus as I do. **Don't forget to set `trans queue depth` to a non-zero value!**
1. Don't set up callback function for I2C device.
1. Try to use bus in synchronous mode.

### Debug Logs.

```plain

```

### More Information.

How can I reconfigure I2C between synchronous and asynchronous mode without reconfiguring the bus, without removing devices?

mythbuster5, 2025-04-10:
FYI, the sync and async is not depends on callback parameter, it depends on `trans_queue_depth`. Because sync transaction doesn't use queue. And we don't support swtich asnyc and sync transaction mode via `i2c_master_register_event_callbacks`

Don't know why you want to use anyc transaction, if you want to use in an isr, and then want to use it in task. Remove the device and add it again without queue set.

DushkaF, 2025-04-10:
@mythbuster5 

I know that I can't fly on one bus, one device synchronous, and the second asynchronous. The point is in sequential action. First, the synchronous mode, and I configure the path through mode through the hub device, consider it so. In fact, this is a smart sensor that can add data from other sensors to its FIFO. While it is in this mode, I configure the child sensors, which are at different addresses. Then I switch to the hub reading mode, where I get a long packet that I want to receive in asynchronous mode. And I need to switch the bus to this mode. The idea is this