espressif/esp-idf #15750: ESP32-S3 - Possible race condition in xEventGroupSetBits with Wi-Fi FTM

ianholland999, 2025-04-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-2916-g1c468f6825

### Espressif SoC revision.

ESP32-S3 ( revision v0.2 )

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Waveshare ESP32-S3-Touch-LCD-1.28

### Power Supply used.

USB

### What is the expected behavior?

I have a relatively simple project using a ESP32-S3 as a FTM initiator, that does trilateration using FTM measurements from 3 APs - the APs are all ESP32-C3s.  The project uses xTaskNotifyGive/Take to coordinate between a task that runs an AP scan, and a task that iterates through the FTM-enabled APs and runs a FTM session.  

Building the initiator code for the ESP32-C3, it works perfectly, runs indefinitely, and doesn't crash.  However, on the ESP32-S3, there is a frequent crash, usually when calling into FreeRTOS EventGroup related functions - either setting bits or waiting for bits.  It crashes in spinlock_acquire when entering the critical section for the event group.

It seems like a multiprocessing issue, but it also happens when enabling CONFIG_FREERTOS_UNICORE on the S3.  I don't really have a great way to reason about what FreeRTOS does with the "internal" wifi tasks such as the Wi-Fi FTM and AP scan functions, so I don't really know how they interact with my code.

### What is the actual behavior?

Overall architecture of my code is:

Scan task: takes task notification, runs start_scan; waits for an event group bit to be set by the event handler indicating scan results are in.  once scan ready, copies scan data out, gives task notification to FTM task.

FTM task: takes notification; iterates through AP results and runs an FTM session with each one, again waiting on the event handler to set "FTM_REPORT_BIT or FTM_FAILURE_BIT" in the event group.  it usually panics on this xEventGroupSync / xEventGroupWaitBits call ( i tried both ).  after iterating through all APs, it does some trilateration math and determines location based on known locations of the APs, then gives the task notification back to the scan task. 

Tasks have different priorities and neither are pinned to a core.

### Steps to reproduce.

Have two tasks with different priorities that use TaskNotifications to run one-at-a-time.  one task does an ap scan, waits for the event handler to announce it's done scanning, then copies the ap records to a static global buffer before giving the task notification to the other task.  

the other task iterates through the ap records, and for each one with the "FTM responder" bit set, it initiates a FTM session and waits on the event worker to set either "FTM_SUCCESS" bit or "FTM_FAILED" bit.  the task uses xEventGroupWait to wait on this bit status.  after iterating like this through all the known FTM responder APs, this task does some other irrelevant work before notifying the first task to begin another scan.

### Debug Logs.

```plain

```

### Diagnostic report archive.

[idf-diag-f4aaf278-8db2-40a4-a2bc-02b512a436ce.zip](https://github.com/user-attachments/files/19674589/idf-diag-f4aaf278-8db2-40a4-a2bc-02b512a436ce.zip)

nachiketkukade, 2025-04-15:
@ianholland999 , after `xEventGroupWaitBits` and checking which bit released the wait, are you clearing the same bit with `xEventGroupClearBits` ? Not doing this may cause some failures if you try to reuse the bit. If this doesn't solve it, please share a console log and example code for further debugging. Without it I can only guess so much.

ianholland999, 2025-04-15:
for sure, i will come back with a code sample and debug log.  @nachiketkukade - yes, i wait on a FTM_REPORT_BIT or FTM_FAILURE_BIT in xEventGroupSync, and then do an & check for each one and xEventGroupClearBits each bit if it's set.  i was also thinking it might be an issue with lifetime of the FTM report, but i disabled use_get_report_api and am simply using the data in the FTM event struct, doing no manual allocation for the report, and it still occurs.

a strange clue is that it doesn't occur when there's only one FTM responder to iterate.  so it seems like some sort of resource contention or race, but i'm simply iterating the APs one after another.  i will follow up with sample code and a debug log.

davidandreoletti, 2025-05-07:
> Not doing this may cause some failures if you try to reuse the bit

Would you mind expanding on "if you try to reuse the bit" please ?

nachiketkukade, 2025-05-09:
@davidandreoletti , no trouble at all. Yes, I mean to to wait on the same bit again as OP had mentioned in his description

> the task uses xEventGroupWait to wait on this bit status. after iterating like this through all the known FTM responder APs

If it's not cleared, issue could arise when trying to access some data structure that wasn't valid.
If Wait is only used repeatedly for a single event, we can set `xClearOnExit` parameter of `xEventGroupWaitBits` to need to internally clear it when the wait ends

davidandreoletti, 2025-05-09:
@nachiketkukade > issue could arise when trying to access some data structure that wasn't valid

```
do 
| nothing
while (xEventGroupWaitBits(group, xWaitForBits, pdFALSE, pDTRUE, 1) != xWaitsForBits)
// Condition satisfied 
```

The pseudo code pattern above does not clear the bits on exit but requires all wait bits to be set to exit (or when timeout). I am using this pattern without issueo no ESP32-S3 (Xiao Seed with latest RTOS available for this platform's stable release). Is this usage pattern also possibly impacted somehow by the issue you raised (invalid data structure) ?