espressif/esp-idf #15751: ulp_lp_core_wakeup_main_processor Only wakes up the main processor once

jath03, 2025-04-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-2844-g5c51472e82

### Espressif SoC revision.

ESP32-C6FH4 (QFN32) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Seeed studio XIAO ESP32C6

### Power Supply used.

USB

### What is the expected behavior?

I expected that `ulp_lp_core_wakeup_main_processor` would always be able to wake the main processor from deep sleep, not just the first time.  

### What is the actual behavior?

Instead the HP core is woken once, but after going into deep sleep again with the wakeup sources configured exactly the same way, it doesn't get woken up again.  

### Steps to reproduce.

I've attached my demo code.  It is adapted from the new example I wrote for #15717 to be a minimal demonstration of the issue I'm running into.
I'm using my fork of the ESP-IDF, https://github.com/jath03/esp-idf/tree/lp-timer-int but none of the changes made are related to sleep/wake.

### Debug Logs.

```plain
HP Core awake
Not woken up by LP core.  Cause = 0
Initializing...
LP core loaded with firmware and running successfully
Configuring ULP wakeup
Entering deep sleep
--- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
--- Waiting for the device to reconnect............
--- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
--- Waiting for the device to reconnect..................................................................

Not positive why printf statements aren't displayed once woken up, but they do work in the full project I'm working on.
```

### Diagnostic report archive.

[idf-diag-4dcb944e-5c3a-46d6-bcb7-aec9fd604b32.zip](https://github.com/user-attachments/files/19675775/idf-diag-4dcb944e-5c3a-46d6-bcb7-aec9fd604b32.zip)

### More Information.

Background: #15717 

I have confirmed that the interrupt on the LP core is repeatedly running.

jath03, 2025-04-10:
My only thought is that `LP_CORE_ISR_ATTR` might possibly be messing with the wakeup trigger register in some way, but I can't find the sources for that in the ESP-IDF, and I'm not sure where else it might be.

KonstantinKondrashov, 2025-04-10:
@jath03
> Instead the HP core is woken once, but after going into deep sleep again with the wakeup sources configured exactly the same way, it doesn't get woken up again.

How do you judge that it is not woken? By these logs? These logs from the esp-idf-monitor, they say that the USB device is disconnected for some reason. Why is that? Maybe the power supply cuts of it, I know some power banks does it. Could you confirm that it is not the case?

```
--- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
--- Waiting for the device to reconnect............
--- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
--- Waiting for the device to reconnect..................................................................
```

jath03, 2025-04-10:
The device is directly plugged into my laptop.  There is no power loss.  Calling the wakeup function in a long running loop in the main function of the LP code works fine, but the same thing in the interrupt handler fails.

KonstantinKondrashov, 2025-04-11:
@jath03 I tried your example with the mentioned PR. I confirm that for some reason there is no farther wakeup for the HP core (only first time it works). I had a look but I do not see anything obvious that can be wrong here. Colleges that work on the ULP support on holidays now. Probably, they will have a look later. cc @sudeep-mohanty

sudeep-mohanty, 2025-04-17:
HI @jath03,
Apologies for the delayed response. I can confirm that the observation you have is consistent with my observation. Upon further investigation, the problem seems to be [here](https://github.com/espressif/esp-idf/blob/e9bdd395995f60878e01cece10525987bacc8a5f/components/bootloader_support/src/bootloader_clock_init.c#L103). The HP CPU bootloader disables the LP Timer interrupt during startup. Hence, the HP CPU wakes up reliably once but during the bootup process, LP timer interrupts are disabled and the LP CPU no longer receives the interrupt.  While we work on a fix for this, a quick workaround for you would be to disable the part of code which disables the LP Timer interrupt from the bootloader. Thanks.

closed 2025-04-23 by espressif-bot