espressif/esp-idf #15753: Reboot with http request

FROeHlyEisvogel, 2025-04-10:
### Is your feature request related to a problem?

When my raspberry pi restarts, ebusd and openhab looses the connection to the adapter.
The adapter is connected via USB to the raspberry pi.
Neither restarting the openhab thing nor restarting ebusd helps at this point.
But when I restart the adapter from webinterface the problem is fixed.

### Describe the solution you'd like.

So it would be nice for me to have something like a http request to trigger the adapter reboot from a rule in openhab on system reboot.

hrushikesh430, 2025-05-16:
Hi @FROeHlyEisvogel, If I understood correctly, it that you want that http request having handler with calls the esp_restart() api?

hrushikesh430, 2025-05-29:
Hi @FROeHlyEisvogel, closing this issue due to no response. Please feel free to reopen the issue is you any have queries.

closed 2025-05-29 by hrushikesh430