espressif/esp-idf #15754: After installation on Windows version is "dirty"

SMinin, 2025-04-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

C:\Espressif\frameworks\esp-idf-v5.4.1>idf.py --version
ESP-IDF v5.4.1

### What is the actual behavior?

C:\Espressif\frameworks\esp-idf-v5.4.1>idf.py --version
ESP-IDF v5.4.1-dirty

### Steps to reproduce.

1. After installing ESP IDF from the offline installer: https://dl.espressif.com/dl/esp-idf/?idf=4.4
2. download ESP-IDF v5.4.1 - Offline Installer
3. install
4. run shortcut from desktop ESP-IDF 5.4 CMD
5. C:\Espressif\frameworks\esp-idf-v5.4.1>idf.py --version
6. result: ESP-IDF v5.4.1-dirty

### Build or installation Logs.

```plain

```

KonstantinKondrashov, 2025-04-10:
@SMinin 
What does `git status` show? I do not know if the git tracks the esp-idf files for the Windows installation.

SMinin, 2025-04-10:
a lot of changes, although nothing changed

C:\Espressif\frameworks\esp-idf-v5.4.1>"C:\Program Files\Git\bin\git.exe" status
Not currently on any branch.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
        modified:   components/app_update/otatool.py
        modified:   components/bootloader/subproject/components/micro-ecc/micro-ecc (modified content)
        modified:   components/bt/controller/lib_esp32 (modified content)
        modified:   components/bt/controller/lib_esp32c2/esp32c2-bt-lib (modified content)
        modified:   components/bt/controller/lib_esp32c3_family (modified content)
        modified:   components/bt/controller/lib_esp32c5/esp32c5-bt-lib (modified content)
        modified:   components/bt/controller/lib_esp32c6/esp32c6-bt-lib (modified content)
        modified:   components/bt/controller/lib_esp32h2/esp32h2-bt-lib (modified content)
        modified:   components/bt/esp_ble_mesh/lib/lib (modified content)
        modified:   components/bt/host/nimble/nimble (modified content)
        modified:   components/cmock/CMock (modified content)
        modified:   components/efuse/efuse_table_gen.py
        modified:   components/efuse/test_efuse_host/efuse_tests.py
        modified:   components/esp_coex/lib (modified content)
        modified:   components/esp_coex/test_md5/test_md5.sh
        modified:   components/esp_phy/lib (modified content)
        modified:   components/esp_wifi/lib (modified content)
        modified:   components/esp_wifi/test_md5/test_md5.sh
        modified:   components/espcoredump/espcoredump.py
.....

C:\Espressif\frameworks\esp-idf-v5.4.1>"C:\Program Files\Git\bin\git.exe" pull origin
Already up to date.

C:\Espressif\frameworks\esp-idf-v5.4.1>"C:\Program Files\Git\bin\git.exe" version
git version 2.47.1.windows.2

KonstantinKondrashov, 2025-04-10:
Seems to me the installation is not completed. Have you tried re-install?
The colleges from the tool team will have a look at this issue. Thanks for the report.

mfialaf, 2025-04-11:
Hi @SMinin,

The offline installer packages the repository using the `--shallow-submodules` flag, to reduce the size of the `.exe` file. This optimization can sometimes cause the version to be marked as "dirty" over time, even if no changes have been made to the code.

To resolve this, you can run `git submodule update --init --recursive --checkout`. This command will properly initialize and update all submodules, which should remove the "dirty" flag from the version.

Let me know if that helped.

SMinin, 2025-04-12:
Thanks for the clarification. The same problem if the toolchain is installed on Linux via a plugin in Eclipse. Executable scripts such as idf.py without the execution flag have to be installed manually via chmod +x , also the "dirty" version. Updating the toolchain via git solves the problem. But the initial installation must still be clean and working.

mfialaf, 2025-04-14:
I'm glad we were able to clarify the issue.
If you encounter any problems related to the Eclipse plugin, I recommend reaching out or opening an issue in the dedicated repository: https://github.com/espressif/idf-eclipse-plugin

Since the issue with the `dirty` flag has been resolved, I’ll go ahead and close this ticket.

closed 2025-04-14 by radimkarnis