espressif/esp-idf #15756: Secure boot and flash encrytion issue in release mode

lekshith, 2025-04-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,
I'm trying to enable secure boot v2 and flash encrytion in "Release Mode" for ESP32S3 chipset.
Here is the sdkcnfig for security features that I've setup:

   

>  #
>     # Security features
>     #
>     CONFIG_SECURE_SIGNED_ON_BOOT=y
>     CONFIG_SECURE_SIGNED_ON_UPDATE=y
>     CONFIG_SECURE_SIGNED_APPS=y
>     CONFIG_SECURE_BOOT_V2_RSA_SUPPORTED=y
>     CONFIG_SECURE_BOOT_V2_PREFERRED=y
>     CONFIG_SECURE_BOOT_V2_RSA_ENABLED=y
>     CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
>     CONFIG_SECURE_BOOT=y
>     CONFIG_SECURE_BOOT_V2_ENABLED=y
>     CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
>     CONFIG_SECURE_BOOT_SIGNING_KEY="PRIVATE_SIGNING_KEY.pem"
>     # CONFIG_SECURE_BOOT_ENABLE_AGGRESSIVE_KEY_REVOKE is not set
>     # CONFIG_SECURE_BOOT_V2_ALLOW_EFUSE_RD_DIS is not set
>     # CONFIG_SECURE_BOOT_FLASH_BOOTLOADER_DEFAULT is not set
>     # CONFIG_SECURE_BOOT_INSECURE is not set
>     CONFIG_SECURE_FLASH_ENC_ENABLED=y
>     CONFIG_SECURE_FLASH_ENCRYPTION_AES128=y
>     # CONFIG_SECURE_FLASH_ENCRYPTION_AES256 is not set
>     # CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT is not set
>     CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE=y
>     CONFIG_SECURE_FLASH_HAS_WRITE_PROTECTION_CACHE=y
>     CONFIG_SECURE_FLASH_ENCRYPT_ONLY_IMAGE_LEN_IN_APP_PART=y
>     CONFIG_SECURE_FLASH_CHECK_ENC_EN_IN_APP=y
>     CONFIG_SECURE_ROM_DL_MODE_ENABLED=y
>     # CONFIG_SECURE_DISABLE_ROM_DL_MODE is not set
>     CONFIG_SECURE_ENABLE_SECURE_ROM_DL_MODE=y
>     # CONFIG_SECURE_INSECURE_ALLOW_DL_MODE is not set
>     # end of Security features 

Following that here is the custom partition table I use:

Code: [Select all](https://esp32.com/viewtopic.php?t=45178#)

```
Code: Select all

# Name,	    Type,	SubType,	Offset,	    Size,	Flags
nvs,	    data,	nvs,	    0x9000,	    0xf000,	
phy_init,	data,	phy,	    0x18000,	0x1000,	
otadata,	data,	ota,    	0x19000,	0x2000,	
factory,	app,	factory,	0x20000,	0x250000,	
ota_0,	    app,	ota_0,	    0x270000,	0x250000,	
www,	    data,	spiffs,	    0x4C0000,	0x450000,	
www2,	    data,	spiffs,	    0x910000,	0x450000,	
```

with the partition table offset set to: CONFIG_PARTITION_TABLE_OFFSET=0x8000

Once everything is enabled and the buld successful, I use to following command to flash:

\

    

> esptool.py -p COM11 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x20000 CN1127.bin 0x8000 partition_table/partition-table.bin 0x19000 ota_data_initial.bin 0x4c0000 [www.bin](http://www.bin/)

After flashing when I use monitor to check first booting process I end up with invalid header logs:

```
Code: [Select all](https://esp32.com/viewtopic.php?t=45178#)

invalid heESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048836
invalid header: 0xdf322330
invalid header: 0xdf322330
invalid header: 0xdf322330
invalid header: 0xdf322330
invalid header: 0xdf322330
invalid header: 0xdf322330
invalid header: 0xdf322330
```

All these steps were integrated under development mode, however, in release mode this is always the case.
Any suggestions are welcome, thank you!

AdityaHPatwardhan, 2025-04-10:
Hi @lekshith,
I had following questions

1) Is this the very first log that you observe when you have flashed the chip for the first time?

2) Can you confirm that no eFuses were written on your chip ?  by executing `idf.py efuse-summary` and sharing it here.

3) If no eFuses were already written then that would mean that the bootloader is not correct.
By the way, when secure boot V2 and Flash encryption are enabled in the bootloader. The bootloader doesnt fit in `0x8000`, have you disabled logs from the bootloader or maybe enabled some other option to make it fit?

lekshith, 2025-04-10:
Hi @AdityaHPatwardhan,
Thank you for your reply. 

1. Yes, this is the very first log that I observe, prior to burning efuses. This is right after enabling security features, build it and flashed to a new chip. After that I use idf.py monitor command.
2. No, I haven't reached to the part of burning efuses.
3. You're right about bootloader not fitting in 0x8000. Yes, I have disabled logs and set optimisation level for performance. 

But during this testing process I tried with the increased bootoader offset to 0x10000 and custom partition as follows:
# Name,    Type, SubType, Offset,   Size,    Flags
nvs,      data, nvs,     0x11000,  0xf000,    # 60 KB
phy_init, data, phy,     0x20000,  0x1000,    # 4 KB
otadata,  data, ota,     0x21000,  0x2000,    # 8 KB
factory,  app,  factory, 0x30000,  0x250000,  # 2.3 MB
ota_0,    app,  ota_0,   0x280000, 0x250000,  # 2.3 MB
www,      data, spiffs,  0x4d0000, 0x450000,  # 4.3 MB
www2,     data, spiffs,  0x920000, 0x450000,  # 4.3 MB

KonstantinKondrashov, 2025-04-10:
`invalid header: 0xdf322330`. This log is from the ROM Bootloader. You wrote the bootloader image at 0x0 offset. It is the correct address for the esp32-s3. The bootloader image starts with a header structure (flash_hdr). The log shows that is completely different from that expected. Did you flash the bootloader image encrypted already? If not then I think the flash encryption is enabled, please check `SPI_BOOT_CRYPT_CNT`.

printf("0x%02x%02x%02x%02x\n", fhdr.pad, fhdr.read_mode, fhdr.blocks, fhdr.magic).
```
struct flash_hdr {
    uint8_t magic; // == 0xE9
    uint8_t blocks;
    uint8_t read_mode;
    uint8_t pad;
    uint32_t entry_addr;
}
```

> --flash_mode dio --flash_freq 80m --flash_size 16MB

If these settings are different from the project then the esptool will change the images but the appended signatures will not be recalculated because. I think it should be safer not to use these flags when the secure boot is used.

KonstantinKondrashov, 2025-04-10:
if you disabled all logs for the bootloader to optimize its size then nothing will be printed. So we will not be able to say what is wrong there.

lekshith, 2025-04-14:
Hi @KonstantinKondrashov 
I have verified the bootloader images from that I had flashed earlier into the device and it seems to the same. For clarification, I'll show you an another device where I enabled the secure boot and flash encryption with logs enabled and increased bootloader offset.
This device successfully enabled secure boot in "development mode" but had similar issue when I chnaged it to release mode.
Yes, this device is flash encrption enabled and I have also burned the efuses. After that I tried to flash the encrpted bootloader and app but resulted in invalid header error.
here is the efuse summary:

> PS C:\CompiledProject\MDC-V2> idf.py efuse-summary -p COM12
> Executing action: efuse-summary
> Running espefuse in directory C:\CompiledProject\MDC-V2\build
> Executing "c:\Espressif\tools\python_env\idf5.4_py3.11_env\Scripts\python.exe -mespefuse summary -p COM12 --chip esp32s3"...
> espefuse.py v4.8.1
> Connecting...
> 
> === Run "summary" command ===
> EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
> ----------------------------------------------------------------------------------------
> Calibration fuses:
> K_RTC_LDO (BLOCK1)                                 BLOCK1 K_RTC_LDO                                   = -20 R/W (0b1000101)
> K_DIG_LDO (BLOCK1)                                 BLOCK1 K_DIG_LDO                                   = -40 R/W (0b1001010)
> V_RTC_DBIAS20 (BLOCK1)                             BLOCK1 voltage of rtc dbias20                      = -28 R/W (0x87)
> V_DIG_DBIAS20 (BLOCK1)                             BLOCK1 voltage of digital dbias20                  = -64 R/W (0x90)
> DIG_DBIAS_HVT (BLOCK1)                             BLOCK1 digital dbias when hvt                      = -44 R/W (0b11011)
> ADC2_CAL_VOL_ATTEN3 (BLOCK1)                       ADC2 calibration voltage at atten3                 = -16 R/W (0b100100)
> TEMP_CALIB (BLOCK2)                                Temperature calibration data                       = -8.4 R/W (0b101010100)
> OCODE (BLOCK2)                                     ADC OCode                                          = 86 R/W (0x56)
> ADC1_INIT_CODE_ATTEN0 (BLOCK2)                     ADC1 init code at atten0                           = 484 R/W (0x79)
> ADC1_INIT_CODE_ATTEN1 (BLOCK2)                     ADC1 init code at atten1                           = 116 R/W (0b011101)
> ADC1_INIT_CODE_ATTEN2 (BLOCK2)                     ADC1 init code at atten2                           = 116 R/W (0b011101)
> ADC1_INIT_CODE_ATTEN3 (BLOCK2)                     ADC1 init code at atten3                           = 104 R/W (0b011010)
> ADC2_INIT_CODE_ATTEN0 (BLOCK2)                     ADC2 init code at atten0                           = 504 R/W (0x7e)
> ADC2_INIT_CODE_ATTEN1 (BLOCK2)                     ADC2 init code at atten1                           = -20 R/W (0b100101)
> ADC2_INIT_CODE_ATTEN2 (BLOCK2)                     ADC2 init code at atten2                           = 48 R/W (0b001100)
> ADC2_INIT_CODE_ATTEN3 (BLOCK2)                     ADC2 init code at atten3                           = 100 R/W (0b011001)
> ADC1_CAL_VOL_ATTEN0 (BLOCK2)                       ADC1 calibration voltage at atten0                 = 316 R/W (0x4f)
> ADC1_CAL_VOL_ATTEN1 (BLOCK2)                       ADC1 calibration voltage at atten1                 = 380 R/W (0x5f)
> ADC1_CAL_VOL_ATTEN2 (BLOCK2)                       ADC1 calibration voltage at atten2                 = 280 R/W (0x46)
> ADC1_CAL_VOL_ATTEN3 (BLOCK2)                       ADC1 calibration voltage at atten3                 = 340 R/W (0x55)
> ADC2_CAL_VOL_ATTEN0 (BLOCK2)                       ADC2 calibration voltage at atten0                 = -84 R/W (0x95)
> ADC2_CAL_VOL_ATTEN1 (BLOCK2)                       ADC2 calibration voltage at atten1                 = -68 R/W (0b1010001)
> ADC2_CAL_VOL_ATTEN2 (BLOCK2)                       ADC2 calibration voltage at atten2                 = -32 R/W (0b1001000)
> 
> Config fuses:
> WR_DIS (BLOCK0)                                    Disable programming of individual eFuses           = 25166596 R/W (0x01800304)
> RD_DIS (BLOCK0)                                    Disable reading from BlOCK4-10                     = 3 R/W (0b0000011)
> DIS_ICACHE (BLOCK0)                                Set this bit to disable Icache                     = False R/- (0b0)
> DIS_DCACHE (BLOCK0)                                Set this bit to disable Dcache                     = False R/- (0b0)
> DIS_TWAI (BLOCK0)                                  Set this bit to disable CAN function               = False R/- (0b0)
> DIS_APP_CPU (BLOCK0)                               Disable app cpu                                    = False R/- (0b0)
> DIS_DIRECT_BOOT (BLOCK0)                           Disable direct boot mode                           = True R/W (0b1)
> UART_PRINT_CONTROL (BLOCK0)                        Set the default UART boot message output mode      = Enable R/W (0b00)
> PIN_POWER_SELECTION (BLOCK0)                       Set default power supply for GPIO33-GPIO37; set wh = VDD_SPI R/W (0b1)
>                                                    en SPI flash is initialized
> PSRAM_CAP (BLOCK1)                                 PSRAM capacity                                     = 8M R/W (0b01)
> PSRAM_TEMP (BLOCK1)                                PSRAM temperature                                  = 85C R/W (0b10)
[... cut ...]
>                                                     eFuse to configure VDD_SPI
> 
> Wdt fuses:
> WDT_DELAY_SEL (BLOCK0)                             RTC watchdog timeout threshold; in unit of slow cl = 40000 R/W (0b00)
>                                                    ock cycle
> 
> Flash voltage (VDD_SPI) set to 3.3V by efuse.

`--flash_mode dio --flash_freq 80m --flash_size 16MB`
Yes, these are the settings set in the menuconfig. I can try to remove these options when flashing the next tinme.

lekshith, 2025-04-14:
Hi @KonstantinKondrashov 
Did you mean to say that flash encryption must not be enabled until we finish the secure boot process?
Also, to clarify my setup, I use espressif extension on visual studio code. I use a custom board with ESP32-s3 which is programmed using micro-USB only. The micro-USB gets disbaled once the secure boot is enabled (we have OTA procedure in place for future updates). Hence, we are testing out this process on ESP32-S3 sample dev-kits. Can you please clarify the correct the procedure to enable this for production release?

AdityaHPatwardhan, 2025-04-14:
@lekshith,

I see, 

> 2. No, I haven't reached to the part of burning efuses.

is not correct then,
The eFuses have already been burned and the flash encryption is enabled for the device. as we can see in your output of `idf.py efuse-summary`

The issue in your case is as follows,

Your `bootloader` has already enabled flash encryption for the chip. Hence the chip now expects that the content on the flash is present in encrypted format. When you flash the bootloader again, you are flashing it in plaintext format. That is why when the ROM is trying to read the bootloader it is decrypting the plaintext content and getting an `invalid header` error. Since you have enabled flash encryption. 
> DIS_DOWNLOAD_MANUAL_ENCRYPT (BLOCK0) Set this bit to disable flash encryption when in d = True R/- (0b1)
download boot modes

If you dont have the flash encryption key externally available, then I am afraid the chip cant be used anymore.

I would recommend to at least set bootloader log verbosity to ERROR so that you are aware of any errors occurred in the boot up instead of disabling the logs completely.

lekshith, 2025-04-14:
Hi @AdityaHPatwardhan,

> is not correct then,
> The eFuses have already been burned and the flash encryption is enabled for the device. as we can see in your output of idf.py efuse-summary

Like I've said in the comment, this is a different device where I have followed through the whole procedure of enabling secure boot and flash encryption. This ultimately destroyed the board cause the bootloader isn't flashed properly and since this is custom board we only have an USB programmer connected, and now that conenctor is disabled. For this reason, I'm using a regular dev-kit and followed the procedure that I've mentioned in the description of this post. And I confirm you that the board that I've mentioned to you in the beginning of this post is not efuse burned and have just been through what I've mentioned.

Regardless, I would really appreciate if you guide me the correct procedure that needs to followed for "Release mode". Also, these boards will be programmed using the ESP flash download tool in production. I'm not able find any clear resources that can confirm this process!

AdityaHPatwardhan, 2025-04-14:
Hi @lekshith,

Can you try out this example https://github.com/espressif/esp-idf/tree/master/examples/security/security_features_app,
You may first try it on qemu interface ( the instructions are provided in the same README).

Alternatively,
There is a [Security Features Enablement Workflow](https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://docs.espressif.com/projects/esp-idf/en/stable/esp32/security/security-features-enablement-workflows.html&ved=2ahUKEwivi7TwrNeMAxUlslYBHemUNtwQFnoECBQQAw&usg=AOvVaw1oCMQIVotU4N5Peav8d1lJ) which you can refer for enabling security features using external tools.

lekshith, 2025-05-06:
Hi @AdityaHPatwardhan @KonstantinKondrashov,

I have been unsuccessful on following the methods suggeested to enable secure boot or flash encrytion. I have tried both the workflow you've provided, once on the device and again using the qemu simulators, and the end result is the same as before:
invalid heESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048836
invalid header: 0xdf322330
invalid header: 0xdf322330
invalid header: 0xdf322330
invalid header: 0xdf322330
invalid header: 0xdf322330
invalid header: 0xdf322330

For the qemu simulator, I cannot check the starting of the logs, since the " invalid header" message runs out of monitor page.
**This workflow works perfectly fine for regular ESP32 chips and not for ESP32S3**. I have come across similar issues in ESP32 Forum that have been unanswered regarding ESP32S3 secure boot as below:
[](https://esp32.com/viewtopic.php?t=39891) and [](https://esp32.com/viewtopic.php?t=40861)
[](https://esp32.com/viewtopic.php?t=45178). I'm yet to come across one successful example.

Please advice on this matter. Thank you.

AdityaHPatwardhan, 2025-05-06:
Hi @lekshith, this invalid header is usually due to nothing being present at the address where bootloader is flashed.

Can you confirm if you flashed the bootloader at 0x0 offset ?
When Secure Boot is enabled, the bootloader is not flashed automatically and you get a command to flash it manually. If you miss this step and only call idf.py flash then the bootloader is not flashed and we get this invalid header error.

Could you try enabling this in your menuconfig

`idf.py -> menuconfig -> Security Features -> Flash bootloader along with other artifacts when using the default flash command`

This would flash the bootloader through the default flash command itself.
I will take a note of this and fix required documentation, Thanks

lekshith, 2025-05-06:
Hi @AdityaHPatwardhan, thanks for your resposne! 
> Can you confirm if you flashed the bootloader at 0x0 offset ?

Yes, I can since I'm using manual command to flash.
Turns oout it works as expected on secure boot alone, it is when I enable flash encrytion we have an a issue. I finally got it working with just enabling secure boot mode, here is the logs:
```

I (504) esp_image: Verifying image signature...
I (505) secure_boot_v2: Secure boot V2 is not enabled yet and eFuse digest keys are not set
I (507) secure_boot_v2: Verifying with RSA-PSS...
I (514) secure_boot_v2: Signature verified successfully!
I (527) boot: Loaded app from partition at offset 0x30000
I (527) secure_boot_v2: enabling secure boot v2...
I (527) efuse: Batch mode of writing fields is enabled
I (530) esp_image: segment 0: paddr=00000020 vaddr=3fce2980 size=02e44h ( 11844)
I (540) esp_image: segment 1: paddr=00002e6c vaddr=403c8700 size=00004h (     4) 
I (545) esp_image: segment 2: paddr=00002e78 vaddr=403c8704 size=00d20h (  3360)
I (553) esp_image: segment 3: paddr=00003ba0 vaddr=403cb700 size=04e90h ( 20112)
I (563) esp_image: Verifying image signature...
I (564) secure_boot_v2: Secure boot V2 is not enabled yet and eFuse digest keys are not set
I (572) secure_boot_v2: Verifying with RSA-PSS...
I (579) secure_boot_v2: Signature verified successfully!
I (581) secure_boot_v2: Secure boot digests absent, generating..
I (594) secure_boot_v2: Digests successfully calculated, 1 valid signatures (image offset 0x0)
I (595) secure_boot_v2: 1 signature block(s) found appended to the bootloader.
I (602) secure_boot_v2: Burning public key hash to eFuse
I (607) efuse: Writing EFUSE_BLK_KEY0 with purpose 9
I (836) secure_boot_v2: Digests successfully calculated, 1 valid signatures (image offset 0x30000)
I (836) secure_boot_v2: 1 signature block(s) found appended to the app.
I (840) secure_boot_v2: Application key(0) matches with bootloader key(0).
I (846) secure_boot_v2: Revoking empty key digest slot (1)...
I (852) secure_boot_v2: Revoking empty key digest slot (2)...
I (857) secure_boot_v2: blowing secure boot efuse...
I (862) secure_boot: Enabling Security download mode...
I (867) secure_boot: Disable hardware & software JTAG...
I (872) secure_boot: Prevent read disabling of additional efuses...
I (879) efuse: BURN BLOCK4
I (884) efuse: BURN BLOCK4 - OK (write block == read block)
I (886) efuse: BURN BLOCK0
I (890) efuse: BURN BLOCK0 - OK (all write block bits are set)
I (894) efuse: Batch mode. Prepared fields are committed
I (899) secure_boot_v2: Secure boot permanently enabled
I (904) boot: Disabling RNG early entropy source......
```

I didn't burn any fuses manually, it was all done by itself. Which gives me another issue of using the comms port. I was under the impression the USB-OTF stack is automatically disable, but the UART comms port will still be active and accept signed images. Is that not the case?
I'm testing this out in a devkit - ESP32S3 - N16R8: now both comms and USB port are both disabled.
```
esptool.py v4.8.1
Serial port COM26
Connecting...
Chip is ESP32-S3 in Secure Download Mode
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
WARNING: In case of failure, please set a specific --flash_size.
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00030000 to 0x00230fff...
Flash will be erased from 0x00010000 to 0x00010fff...
Flash will be erased from 0x00021000 to 0x00022fff...
Flash will be erased from 0x004d0000 to 0x0091ffff...
Erasing flash...
WARNING: Security features enabled, so not changing any flash settings.
Took 0.23s to erase flash block
Wrote 21504 bytes at 0x00000000 in 0.4 seconds (396.0 kbit/s)...
Erasing flash...
WARNING: Security features enabled, so not changing any flash settings.

A fatal error occurred: Failed to enter Flash download mode (result was 01060000: Operation or feature not supported)
```
Our main board has a method to accept OTA updates, hopefully that should work. **However, is there a way to write a signed firmware on this dev**kit?
Also, the same process with just enabling the flash encryption still fails as before.

lekshith, 2025-05-06:
Hi @AdityaHPatwardhan,

Here is the answer (https://github.com/espressif/esp-idf/issues/14172),
you have worked on it, lol. The key is the order of flashing the binaries. Everything is working as expected. 
Please request this to be added in the official documentation and specify the order of the flashing done after enabling security features!
Thanks for your time, I'll be closing the ticket now!

lekshith, 2025-05-07:
Hi @AdityaHPatwardhan,

Looks like solution from this #14172 works for secure boot perfectly, but I think the flash encryption should work in similar way, I'll dig more and get it right. 
The order of flashing the binaries is really the key aspect, you need to request this to be added in the documentation. Thank you!

closed 2025-05-07 by lekshith

dmitrij9992905, 2025-06-29:
Hello everyone!

I have tha same issue even with self-generated Flash Encryption key with Secure boot.
FE - Flash Encryption
SB - Secure Boot
OB - onboard-generated flash encryption key
H - host-generated key
P - preencrypted images

AES128XTS:
FE OB + SB -> invalid header
FE H + SB -> invalid header
FE H P + SB -> invalid header

AES256XTS:
FE OB + SB -> invalid header
FE H + SB -> tbc
FE H P + SB -> tbc

SB -> OK

In the ESP32 case in my experience, FE H + SB -> OK, but didn't work on ESP32S3.
Looks like a maskROM bug