espressif/esp-idf #15758: 5.4.1 Build failed when set-target linux

liyafe1997, 2025-04-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1-4c2820d3

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The exactly same codebase works on previous version (v5.3.1)

### What is the actual behavior?

```
d -o esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_startup.c.o -c /home/liyafe1997/esp/v5.4.1/esp-idf/components/efuse/src/esp_efuse_startup.c
In file included from /home/liyafe1997/esp/v5.4.1/esp-idf/components/efuse/src/esp_efuse_startup.c:18:
/home/liyafe1997/esp/v5.4.1/esp-idf/components/esp_system/include/esp_private/startup_internal.h:30:44: error: ‘SOC_CPU_CORES_NUM’ undeclared here (not in a function)
   30 | extern sys_startup_fn_t const g_startup_fn[SOC_CPU_CORES_NUM];
      |                                            ^~~~~~~~~~~~~~~~~
```
Seems this `SOC_CPU_CORES_NUM` is missing when `set-target linux`

### Steps to reproduce.

```
. ~/esp/v5.4.1/esp-idf/export.sh
idf.py --preview set-target linux
idf.py build
```

### Build or installation Logs.

```plain

```

KonstantinKondrashov, 2025-04-10:
Oh wait, it should not happen for Linux target. It has to single core.
https://github.com/espressif/esp-idf/blob/master/components/esp_system/include/esp_private/startup_internal.h#L29-L33.
Your settings are not correct. The Linux target has to have CONFIG_ESP_SYSTEM_SINGLE_CORE_MODE=y.

KonstantinKondrashov, 2025-04-10:
What app are you trying to build?

liyafe1997, 2025-04-10:
> Oh wait, it should not happen for Linux target. It has to single core. https://github.com/espressif/esp-idf/blob/master/components/esp_system/include/esp_private/startup_internal.h#L29-L33. Your settings are not correct. The Linux target has to have CONFIG_ESP_SYSTEM_SINGLE_CORE_MODE=y.

I have tried to add `CONFIG_ESP_SYSTEM_SINGLE_CORE_MODE=y` either to `sdkconfig` or `sdkconfig.defaults` but still doesn't work.
I think this should be done by the SDK. If I have to add this, what should I do if I want dualcore works when I target at esp32s3, any also the same codebase (with single core) works with target at linux.

Notice: Previous IDF versions (5.2.2 & 5.3.1) works well, I only has this problem after I upgraded to 5.4.1 recently.

fhrbata, 2025-04-11:
Hello @liyafe1997 ,

> I have tried to add CONFIG_ESP_SYSTEM_SINGLE_CORE_MODE=y either to sdkconfig or sdkconfig.defaults but still doesn't work.

The  [CONFIG_ESP_SYSTEM_SINGLE_CORE_MODE](https://github.com/espressif/esp-idf/blob/master/components/esp_system/Kconfig#L57) is a non-visible symbol, meaning its value cannot be directly changed by the user. It's set based on [CONFIG_FREERTOS_UNICORE](https://github.com/espressif/esp-idf/blob/master/components/freertos/Kconfig#L23), which selects it and  should be set for the linux target. 

`default "y" if IDF_TARGET_ESP32S2 || IDF_TARGET_LINUX`

Isn't the `CONFIG_FREERTOS_UNICORE` disabled in your configuration?

Thank you

liyafe1997, 2025-04-15:
> > Isn't the CONFIG_FREERTOS_UNICORE disabled in your configuration?
> 
> [@liyafe1997](https://github.com/liyafe1997) Did it help for your case? Please note, that the Linux target in the ESP-IDF project does not support dual-core configuration. See the comment above.
> 
> For Linux target, you have to see the sdkconfig file with the following options set: CONFIG_FREERTOS_UNICORE=y CONFIG_ESP_SYSTEM_SINGLE_CORE_MODE=y

No still doesn't work. 
I have confirmed `CONFIG_FREERTOS_UNICORE=y` is existing in `sdkconfig` while I am compiling (target linux), then it still has the exactly same error `SOC_CPU_CORES_NUM`.

KonstantinKondrashov, 2025-04-15:
@liyafe1997 

You can not just simply add these options in the `sdkconfig` file. It won't work. You can add them to `sdkconfig.defaults` and run `idf.py reconfigure`. Or make it manually via `idf.py menuconfig`. (Top) → Component config → FreeRTOS → Kernel -> `Run FreeRTOS only on first core`.

https://github.com/espressif/esp-idf/blob/e9bdd395995f60878e01cece10525987bacc8a5f/components/esp_system/include/esp_private/startup_internal.h#L29-L33

liyafe1997, 2025-04-15:
> @liyafe1997
>
> You can not just simply add these options in the `sdkconfig` file. It won't work. You can add them to `sdkconfig.defaults` and run `idf.py reconfigure`. Or make it manually via `idf.py [...]

Of course I have added to sdkconfig.defaults, and the sdkconfig is generated from that. That is why I say it is existing in sdkconfig while compiling.

KonstantinKondrashov, 2025-04-15:
Seems the freertos component is not part of your build this is why CONFIG_ESP_SYSTEM_SINGLE_CORE_MODE is left =n. While it is set to `y` in the freertos Kconfig.

liyafe1997, 2025-04-15:
> Can you share the sdkconfig file?

There are some private unreleased product brand name I don't want to share publicly, can you please give me an email or any other way I can share with you?

KonstantinKondrashov, 2025-04-16:
I do not want to share my email , but it is not even necessary. You can clean up the sdkconfig from the sensitive info. For now, it is not even a question. It is quite obvious that you have:
CONFIG_FREERTOS_UNICORE=y
# CONFIG_ESP_SYSTEM_SINGLE_CORE_MODE (is not set or `n`?)
# CONFIG_SOC_CPU_CORES_NUM is not set

The question is why your build does not include the esp_system (or freertos) component. Do you change `__COMPONENT_REQUIRES_COMMON` manually? If you have CONFIG_FREERTOS_UNICORE set it means that the freertos is included in your build and esp_system is not, I guess. Maybe the build log has some warnings related to this? Please double-check it or share it. Thanks.

closed 2025-04-20 by espressif-bot