espressif/esp-idf #15762: SPI slave configuration mismatch in mode 3

boborjan2, 2025-04-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.8

### Espressif SoC revision.

ESP32 rev3.1

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board using ESP32-WROOM-32E-N8R2

### Power Supply used.

External 3.3V

### What is the expected behavior?

In our setup, ESP32 acts as an SPI slave. SPI mode 3 is used as can be seen in this screenshot (The byte transmitted is 0x9E, case 1):
The signals are CLK and MOSI (_i.e this is what is received by ESP32!_)

[image]

Here is another tranasaction of the same 0x9E: (case 2)
[image]

We expect esp32 read the values correctly in both cases as 0x9E.
Oscilloscope signals do correspond to a mode 3 transaction.

### What is the actual behavior?

In the cases above, ESP32 reads input as: 0xDE (case 1) and 4E (case 2).

I checked Reference manual and spi_slave initialization code in spi_ll.h. I played with values in mode 3 init and for a surprise using the following init values made the issue go away (not only in the cases above but in general):
        hw->pin.ck_idle_edge = 0;
        hw->user.ck_i_edge = 1;               // <--------------- after having modified to 1, mode3 reception is fine.
        hw->ctrl2.miso_delay_mode = 1;
        hw->ctrl2.miso_delay_num = 0;
        hw->ctrl2.mosi_delay_mode = 0;
        hw->ctrl2.mosi_delay_num = 0;

Can you elaborate on that? Are we missing something?

What is also strange is that with mosi_delay_mode==0, this change should not make any difference (mosi_delay_mode==0 means zero delay for MOSI). It is miso_delay_mode that is 1, so according to the register description, this should effect MISO only.
Can it happen that in slave case, miso_delay_mode _applies to the data input line, rather than MISO signal_?

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

Alvin1Zhang, 2025-04-11:
thanks for reporting.

Icarus113, 2025-04-21:
v4.4 is end of life. Would you help try this issue on master branch or newer release versions to see if this still happens? Thanks. @boborjan2

Alvin1Zhang, 2025-05-07:
Thanks for reporting, will close due to inactivity, feel free to reopen if the issue still happens. Thanks.

closed 2025-05-07 by Alvin1Zhang

boborjan2, 2025-06-18:
@Alvin1Zhang Please repoen the issue, I do not have permissions to do so.