espressif/esp-idf #15763: Radio 40dBm weaker and unable to restore

JVKran, 2025-04-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi there,

We have used one ESP32-S3 as prototype with wireless charging in a PETG enclosure. PETG deforms at around 85 degrees celcius and I'm pretty sure the ESP32-S3 got hotter than that. There's one ESP32-S3 that has a worse performing radio than the others and not by a little, but by a lot; around 40dBm worse. 

I've disabled storing wifi calibration data in NVS to force a full calibration and also erased the flash with ```idf.py erase-flash``` and it doesn't change one bit. The same application on another PCB (of the same design) with the same external antenna reaches around -33dBm while the one PCB that became hot doesn't get any stronger than around -71dBm. I'm 100% sure they were connected to the same AP.

Could this difference be explained by the increased temperature? I think the PCB was heated for around a day or so. Is there anything else I can try?

Best regards,

Jochem

JVKran, 2025-04-14:
I just replaced the WROOM with another one and the RSSI of this one is correct, which indeed indicates damage to the WROOM module. What's the likelihood the heat caused the described behaviour?

EspressifTom, 2025-04-15:
Hi Jochem,

Based on your description, it seems that the RX may have been damaged, most likely due to ESD, which can easily occur in dry environments. We recommend implementing proper ESD protection measures to prevent this in the future.

If needed, we can help to verify the issue by testing the damaged module. Please feel free to send it to us for analysis.

Best regards,
Tom from Espressif Systems

JVKran, 2025-04-16:
Hi @EspressifTom, 

I'm afraid the module didn't survive the desoldering operation. How would we implement proper ESD protection? During assembly, the uFL connector is always exposed while it isn't in the finished product. 

It did pass our tests before delivery, in which we also take note of the RSSI. I'm assuming ESD damage would be instantly present, and not start occuring over a couple of weeks, right? Or could the RSSI degrade over time after damage has been dealt by ESD?

Best regards,

Jochem

EspressifTom, 2025-04-17:
Hi Jochem,

The ESD damage would be instantly present ,but the situation will get worse if ESD always happened.
The ESD protection including : Use ESD-safe workstations (mat, wrist strap, ionizer) during assembly and testing.
Ensure operators follow ESD-safe handling protocols, especially during manual testing or soldering.

Thanks,
Tom from Espressif Systems

closed 2025-05-07 by Alvin1Zhang