espressif/esp-idf #15766: There seems to be an error in the use of the 'chunked' judgment method in '_http_ event_ handler'(esp_http_client_example), with the opposite logic

JIAnengneng, 2025-04-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

//URL:https://github.com/espressif/esp-idf/blob/v5.4.1/examples/protocols/esp_http_client/main/esp_http_client_example.c
# 1. The following is a screenshot of the code call relationship explanation and problem reproduction
[image]
（The code for the image has been supplemented based on the example code）
# 2. The following is a screenshot of the problem fix after the code adjustment
[image]

nileshkale123, 2025-04-14:
Hello @JIAnengneng ,

Thanks for reporting the issue. 
I have verified that it's unable to read the chunked encoded response, for example, esp_http_client_example.

A fix for this has been raised internally and should be reflected in the master branch shortly. I will keep you updated on any relevant progress or changes related to the pull request.

In the meantime, I’m attaching the corresponding patch for your reference. Please let me know if you expect any additional changes or have further feedback.

[gh_issue_15766.txt](https://github.com/user-attachments/files/19733774/gh_issue_15766.txt)

nileshkale123, 2025-05-16:
Hello @JIAnengneng ,

Since this is an example intended for flexible usage, we will not be merging this change into the main repository. However, you are welcome to use the provided patch to capture and print chunked responses as needed.

We have already included a comment in the example to guide users on this, and based on our internal discussion, we’ve decided to close this merge request.

For reference, ESP-IDF already includes built-in support for handling chunked transfer encoding in HTTP responses.

nileshkale123, 2025-05-22:
I am closing this issue for now. Please feel free to re-open if needed.

closed 2025-05-22 by nileshkale123