espressif/esp-idf #15768: wifi process malloc fail

SMinin, 2025-04-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.4.1

### Espressif SoC revision.

ESP32 rev.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom

### Power Supply used.

External 3.3V

### What is the expected behavior?

It would be great if the device did not reboot and worked stably.

### What is the actual behavior?

Sometimes the device reboots because it is impossible to allocate memory in the wifi process. Below is the dump. There is no way to increase the stack for the wifi process via sdkconfig. Please add this feature.

Mem alloc fail. size 0x000006a0 caps 0x00001800

Backtrace: 0x40082152:0x3ffdd5e0 0x40090929:0x3ffdd610 0x400829c3:0x3ffdd640 0x40082a4e:0x3ffdd6a0 0x40082a8a:0x3ffdd6d0 0x4009a1a2:0x3ffdd700 0x40089cd2:0x3ffdd730 0x401d4491:0x3ffdd760 0x401d7984:0x3ffdd790 0x401d7c0f:0x3ffdd7c0 0x401d8147:0x3ffdd800 0x401d8203:0x3ffdd840 0x401d2462:0x3ffdd880 0x40090dbb:0x3ffdd8b0
--- 0x40082152: panic_abort at /home/sergey/esp/esp-idf/components/esp_system/panic.c:454
--- 0x40090929: _frxt_int_enter at /home/sergey/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:157
--- 0x400829c3: heap_caps_alloc_failed at /home/sergey/esp/esp-idf/components/heap/heap_caps.c:58
--- 0x40082a4e: heap_caps_malloc at /home/sergey/esp/esp-idf/components/heap/heap_caps.c:88
--- 0x40082a8a: heap_caps_malloc_default at /home/sergey/esp/esp-idf/components/heap/heap_caps.c:110
--- 0x4009a1a2: _malloc_r at /home/sergey/esp/esp-idf/components/newlib/heap.c:43
--- 0x40089cd2: wifi_malloc at /home/sergey/esp/esp-idf/components/esp_wifi/esp32/esp_adapter.c:82
--- 0x401d4491: ppRegressAmpdu at ??:?
--- 0x401d7984: esf_buf_alloc at ??:?
--- 0x401d7c0f: wDev_AppendRxBlocks at ??:?
--- 0x401d8147: wDev_ProcessRxSucData at ??:?
--- 0x401d8203: wDev_ProcessRxSucData at ??:?
--- 0x401d2462: ppTask at ??:?
--- 0x40090dbb: vPortAssertIfInISR at /home/sergey/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:456

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

SMinin, 2025-04-12:
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffdda50, name: 'wifi', GDB name: 'process 1073601104'
Crashed task is not in the interrupt context
Panic reason: Mem alloc fail. size 0x000006a0 caps 0x00001800

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x4008dcab
epc2           0x4000bff0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x60c20
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x40082155          0x40082155 <panic_abort+29>
lbeg           0x4000c349          1073791817
lend           0x4000c36b          1073791851
lcount         0xffffffff          4294967295
sar            0x8                 8
ps             0x60220             393760
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x800908b8          -2146891592
a1             0x3ffdd5e0          1073599968
a2             0x3ffdd64c          1073600076
a3             0x3ffdd61b          1073600027
a4             0x7                 7
a5             0x3ffdd67a          1073600122
a6             0x30                48
a7             0x30                48
a8             0x3ffb5888          1073436808
a9             0x1                 1
a10            0x3ffdd61b          1073600027
a11            0x1                 1
a12            0x29fff             172031
a13            0x40089cbc          1074306236
a14            0x3                 3
a15            0x0                 0

==================== CURRENT THREAD STACK =====================
#0  panic_abort (details=0x3ffdd64c "Mem alloc fail. size 0x000006a0 caps 0x00001800") at /home/sergey/esp/esp-idf/components/esp_system/panic.c:466
#1  0x400908b8 in esp_system_abort (details=0x3ffdd64c "Mem alloc fail. size 0x000006a0 caps 0x00001800") at /home/sergey/esp/esp-idf/components/esp_system/port/esp_system_chip.c:87
#2  0x400829c6 in heap_caps_alloc_failed (requested_size=1696, caps=6144, function_name=<optimized out>) at /home/sergey/esp/esp-idf/components/heap/heap_caps.c:58
#3  0x40082a51 in heap_caps_malloc (size=1696, caps=6144) at /home/sergey/esp/esp-idf/components/heap/heap_caps.c:88
#4  0x40082a8d in heap_caps_malloc_default (size=1696) at /home/sergey/esp/esp-idf/components/heap/heap_caps.c:110
#5  0x4009a129 in malloc (size=1696) at /home/sergey/esp/esp-idf/components/newlib/heap.c:24
#6  0x40089cd5 in wifi_malloc (size=1696) at /home/sergey/esp/esp-idf/components/esp_wifi/esp32/esp_adapter.c:82
#7  0x401d4600 in esf_buf_alloc_dynamic ()
#8  0x401d7af3 in esf_buf_alloc ()
#9  0x401d7d7e in wDev_IndicateFrame ()
#10 0x401d82b6 in wDev_ProcessRxSucData ()
[... 29 lines cut ...]
0x3ffaf630             ipc0      1/1         496/1028
0x3ffd7fc0        WebSocket      5/5         512/2548
0x3ffe82e8         CAN read      5/5         752/2316
0x3ffe74f0        CAN alert      5/5         608/2452
0x3ffafe38             ipc1    24/24         496/1036
0x3fff3c08          WebChat      2/2         656/3436
0x3ffce1b4        LED flash      0/0          512/496

==================== THREAD 1 (TCB: 0x3ffdda50, name: 'wifi') =====================
#0  panic_abort (details=0x3ffdd64c "Mem alloc fail. size 0x000006a0 caps 0x00001800") at /home/sergey/esp/esp-idf/components/esp_system/panic.c:466
#1  0x400908b8 in esp_system_abort (details=0x3ffdd64c "Mem alloc fail. size 0x000006a0 caps 0x00001800") at /home/sergey/esp/esp-idf/components/esp_system/port/esp_system_chip.c:87
#2  0x400829c6 in heap_caps_alloc_failed (requested_size=1696, caps=6144, function_name=<optimized out>) at /home/sergey/esp/esp-idf/components/heap/heap_caps.c:58
#3  0x40082a51 in heap_caps_malloc (size=1696, caps=6144) at /home/sergey/esp/esp-idf/components/heap/heap_caps.c:88
#4  0x40082a8d in heap_caps_malloc_default (size=1696) at /home/sergey/esp/esp-idf/components/heap/heap_caps.c:110
#5  0x4009a129 in malloc (size=1696) at /home/sergey/esp/esp-idf/components/newlib/heap.c:24
#6  0x40089cd5 in wifi_malloc (size=1696) at /home/sergey/esp/esp-idf/components/esp_wifi/esp32/esp_adapter.c:82
#7  0x401d4600 in esf_buf_alloc_dynamic ()
#8  0x401d7af3 in esf_buf_alloc ()
#9  0x401d7d7e in wDev_IndicateFrame ()
#10 0x401d82b6 in wDev_ProcessRxSucData ()
#11 0x401d8372 in wdevProcessRxSucDataAll ()
#12 0x401d25d1 in ppTask ()
#13 0x40090d4a in vPortTaskWrapper (pxCode=0x401d2464 <ppTask>, pvParameters=0x0) at /home/sergey/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

<---cut--->

Sajia96, 2025-04-21:
Hello @SMinin, Please share complete debug log and sdkconfig.

closed 2025-12-11 by Alvin1Zhang