espressif/esp-idf #15771: gpdma_register_rx_event_callback: on_recv_eof not in IRAM

SimonKagstrom, 2025-04-13:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

Waveshare esp32s3 with round 2.8" display

### Power Supply used.

USB

### What is the expected behavior?

I use esp_async_memcpy to fill the bounce buffer to draw the display. This improves performance measurably on my application.

### What is the actual behavior?

After upgrading from esp-idf 5.3 to 5.4.1, I now get an issue at startup:

```
E (135) gdma: gdma_register_rx_event_callbacks(500): on_recv_eof not in IRAM
E (135) async_mcp.gdma: esp_async_memcpy_install_gdma_template(163): failed to register RX EOF callback
```

and then a guru meditation when the actual memcpy is called.

### Steps to reproduce.

Rebuild my application (https://github.com/SimonKagstrom/maelir) with v5.4.1. The sdkconfig looks like this:

[sdkconfig.txt](https://github.com/user-attachments/files/19723376/sdkconfig.txt)

### Debug Logs.

```plain

```

### More Information.

Simply marking mcp_gdma_rx_eof_callback with the IRAM_ATTR resolves the issue for me, although I suspect there might be more to it.

```
diff --git a/components/esp_hw_support/dma/async_memcpy_gdma.c b/components/esp_hw_support/dma/async_memcpy_gdma.c
index 527e0ff79c..c450f76891 100644
--- a/components/esp_hw_support/dma/async_memcpy_gdma.c
+++ b/components/esp_hw_support/dma/async_memcpy_gdma.c
@@ -434,7 +434,8 @@ err:
     return ret;
 }

-static bool mcp_gdma_rx_eof_callback(gdma_channel_handle_t dma_chan, gdma_event_data_t *event_data, void *user_data)
+
+static bool IRAM_ATTR mcp_gdma_rx_eof_callback(gdma_channel_handle_t dma_chan, gdma_event_data_t *event_data, void *user_data)
 {
     bool need_yield = false;
     async_memcpy_gdma_context_t *mcp_gdma = (async_memcpy_gdma_context_t *)user_data;
```

suda-morris, 2025-04-14:
Thanks for the report. We're fixing it. @SimonKagstrom

closed 2025-04-18 by espressif-bot

SimonKagstrom, 2025-09-02:
@suda-morris I just tested with esp-idf 5.5.1, and I still see this issue.

The workaround I outlined above still works for me.