espressif/esp-idf #15772: Following the documentation leads to non-functioning SPI

bmink, 2025-04-13:
The documentation here:

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/spi_master.html#_CPPv416spi_bus_config_t

specifically mentions to set unused GPIOs to -1. This is incorrect, as the struct consists of unions so values will be overwritten. Eg. setting `mosi_io_num` to the desired value, and then setting `data0_io_num` to -1 will in fact overwrite `mosi_io_num` to be -1...

This is very confusing. The documentation should be fixed to make sure programmers don't overwrite the pin numbers they do want to use with -1.

wanckl, 2025-04-14:
@bmink Hello, SPI has different line mode to using different IOs, like `classic` mode using miso/mosi/clk, `dual` mode using data0~1/clk, and also `qaud, octal` mode,,

What config you want `mosi` together with `data0` ?

bmink, 2025-04-14:
I understand what you are saying. I don't have an issue with the way it works, but I do have an issue with the documentation. The documentation (that I linked above) specifically says to init `data0` to `-1` if it is not used. No indication is given that `mosi` and `data0` are members of the same union and *will overwrite each other*. The documentation should say to *only* either set `mosi` or `data0`. Same should be done for all the other unions in the struct.

wanckl, 2025-04-15:
@bmink Oh, I see, I'll more clear the documentation, you are right, thanks for report this 👍

closed 2025-04-15 by bmink