espressif/esp-idf #15777: ESP32P4 FB_DCDC function

PickaxeHit, 2025-04-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, espressif. I found that ESP32P4 has a pin named FB_DCDC, and in the example hardware design it connects to TLV62569's FB pin. 
I know esp32p4 can adjust its cpu voltage using this pin, but how it works? Is it just a current provider or consumer? And if the DCDC's FB ref voltage is not same to TLV62569(600mV), such as TPS82130's 800mV ref voltage, could esp32p4 adjust the output voltage?
I found that in the esp32 forum there is a [similar post](https://esp32.com/viewtopic.php?f=2&t=42480) but no reply there.
I'm designing a new hardware using smaller components (TPS82130) to reduce the size, so I'm finding an alternative solution for this function.
Thanks.

PickaxeHit, 2025-04-16:
I'm sorry but any update here?

Lindazhxy, 2025-04-29:
Hi,

Thank you for your question and for considering ESP32-P4 in your new design!

The FB_DCDC pin on the ESP32-P4 is used to support dynamic voltage scaling (DVS) by adjusting the feedback voltage of an external DC-DC converter. Internally, the ESP32-P4 controls a programmable resistor divider connected to this pin, and by setting the register PMU_HP_ACTIVE_DCM_VSET[4:0], it can modify the resistance ratio to influence the output voltage of the regulator.

Here are some key points:

The ESP32-P4 does not act as a current source or sink on the FB_DCDC pin. Instead, it behaves like a programmable voltage divider that outputs a scaled voltage to the external DC-DC converter’s FB (feedback) pin.

With converters like TLV62569 (reference voltage: 600 mV), the internal design matches well and allows fine control of the output voltage.

For converters like TPS82130, which use a different reference voltage (800 mV), it is still possible to connect the FB_DCDC pin. However, please note:

The internal divider of the ESP32-P4 is optimized for a 600 mV reference.

When using TPS82130, the minimum output voltage that can be achieved is limited to approximately 0.8 V. The chip cannot scale the voltage lower than that due to the higher reference voltage of the regulator.

In summary:

Yes, you can connect the FB_DCDC pin to the FB pin of the TPS82130.

However, dynamic voltage adjustment will be constrained — the minimum output voltage will be about 0.8 V, and you should check whether this meets your power requirements.

If you need a lower output voltage (e.g., down to 0.6 V), consider using a converter with a compatible FB reference voltage (like TLV62569).

PickaxeHit, 2025-04-29:
Thanks for your reply, it made me understand what FB_DCDC actually does. But I still have some questions:
> Instead, it behaves like a programmable voltage divider that outputs a scaled voltage to the external DC-DC converter’s FB (feedback) pin.

Does it mean that the voltage divider resistors connected to the dcdc chip are not necessary? Or if the resistors' value is too small, is it means P4 couldn't control the voltage? In the development board schematic v1.4, the tlv62569 has two 499k resistors connected, but in version 1.5, they are marked NC. Is my understanding correct?

> When using TPS82130, the minimum output voltage that can be achieved is limited to approximately 0.8 V. The chip cannot scale the voltage lower than that due to the higher reference voltage of the regulator.

I found that in the datasheet, the min value of VDD_HP is 0.9v. But according to your reply, can this voltage actually be lower? Can such a low voltage maintain the operation of the CPU?

> Internally, the ESP32-P4 controls a programmable resistor divider connected to this pin, and by setting the register PMU_HP_ACTIVE_DCM_VSET[4:0], it can modify the resistance ratio to influence the output voltage of the regulator.

Since the PMU part in the TRM is not yet public, I cannot understand the details. I guess that the current FB voltage is controlled by the IDF internally (or the control of FB is not implemented?). If it is the former, does it mean that I need to manually modify the IDF to support DCDC with 800mv Vref? If I do not make any adjustments, will the P4 burn out or work malfunction due to the excessively high core voltage?

My highest respect for your reply. Thanks!