espressif/esp-idf #15779: Clangd LSP and issues with Go-to definition

truck-iguana, 2025-04-14:
I have referenced other issues, and it has not solved my problem.
[Support a language server #6721](https://github.com/espressif/esp-idf/issues/6721)
[compile_commands.json is not complete #9468](https://github.com/espressif/esp-idf/issues/9468)

Following those threads did not solve the problem I've encountered. Before following those steps, my LSP (default `clangd`) had no problems parsing for function _declarations_. **My problem is that I cannot get my LSP to jump to the actual function definition**.

What I've accomplished following those steps in review.
1) Run `idf_tools.py install esp-clang`, now `which clangd` returns `/root/.espressif/tools/esp-clang/esp-18.1.2.2_20240912/esp-clang/bin/clangd`
2) added `.clangd` file to the root directory of `esp-idf` and my project directory for `hello_world`
.clangd is as defined
```
CompileFlags:
  Add: [-mlong-calls, -isysroot=/root/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf, -query-driver=/root/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp-elf-gcc]
  Remove: [
    -fno-tree-switch-conversion,
    -mtext-section-literals,
    -fstrict-volatile-bitfields,
    -mlong-calls,
    -fno-shrink-wrap,
    -march=rv32imc_zicsr_zifencei,
  ]
```

As for my esp-idf general setup, I followed the steps below.

[I followed along these steps in get-started for linux](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-macos-setup.html). I built the [hello_world application and flashed it to my esp32](https://github.com/espressif/esp-idf/tree/master/examples/get-started/hello_world).

As a test sample, I wanted to search for the definition of `esp_chip_info` (on this line)[https://github.com/espressif/esp-idf/blob/1c468f68259065ef51afd114605d9122f13d9d72/examples/get-started/hello_world/main/hello_world_main.c#L23]. My LSP could only bring me to the declaration [shown here](https://github.com/espressif/esp-idf/blob/1c468f68259065ef51afd114605d9122f13d9d72/components/esp_hw_support/include/esp_chip_info.h#L60). After parsing the codebase myself with `ripgrep`, I found the definition [here](https://github.com/espressif/esp-idf/blob/1c468f68259065ef51afd114605d9122f13d9d72/components/esp_hw_support/port/esp32/chip_info.c#L15).

I've inspected my `compile_commands.json` and highlighted the entry below

```
{
  "directory": "/workspace/esp_dev/esp-idf/examples/get-started/hello_world/build",
  "command": "/root/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DIDF_VER=\\\"v5.4.1-dirty\\\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/workspace/esp_dev/esp-idf/examples/get-started/hello_world/build/config -I/workspace/esp_dev/esp-idf/components/esp_hw_support/include .... lots of headers here ... -mlongcalls -Wno-frame-address -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function ... more options ... -o esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/chip_info.c.obj -c /workspace/esp_dev/esp-idf/components/esp_hw_support/port/esp32/chip_info.c"
  "file": "/workspace/esp_dev/esp-idf/components/esp_hw_support/port/esp32/chip_info.c",
  "output": "esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/chip_info.c.obj"
}
```

So in my `compile_commands.json` file, I do see the source file listed. But for some reason, the LSP is not actually jumping to this definition/implementation. `"No definition found."`

igrr, 2025-04-14:
Hi @truck-iguana, LSP is quite a bit out of scope of ESP-IDF project, so not sure how much we can help you.

I would recommend enabling verbose logging from clangd first, and then looking at clangd logs to see if there are any errors or warnings reported.

As for the settings: you haven't specified your IDF version, but at least for the `master` branch what works for me is:
- Clangd arguments: `--query-driver=** --log=verbose --compile-commands-dir=<path to your project's build directory>`
- `.clangd` file contents:
   ```
   CompileFlags:
    Remove: [-f*, -m*]
   ```

truck-iguana, 2025-04-15:
@igrr It looks like the `--compile-commands-dir=<path>` is exactly what was needed for my LSP to be able to go-to definition for `esp_chip_info`. I will do more experminting. Unfortunately for my text editor (helix), I set these `clangd` arguments inside a **global** `languages.toml` configuration. So I will have to see what else I can do. Still, thank you for pointing out the arguments you used. It's at least given me a verifiable solution to pursue

Alvin1Zhang, 2025-04-17:
Thanks for reporting, feel free to reopen.

closed 2025-04-17 by Alvin1Zhang

OrhanYigitDurmaz, 2025-05-05:
This is definitely relevant in zed editor too, will post my config later

igrr, 2025-05-05:
FWIW here is the zed config which works for me:

```
{
    "lsp": {
        "clangd": {
            "path_lookup": true,
            "binary": {
                "path": "<YOUR_HOME_DIR>/.espressif/tools/esp-clang/<VERSION>/esp-clang/bin/clangd",
                "arguments": [
                    "--header-insertion=never",
                    "--query-driver=**",
                    "--log=verbose",
                    "--compile-commands-dir=<YOUR_IDF_PATH>"
                ]
            }
        }
    }
}
```

This works in conjunction with a post-build step which copies the compile_commands.json file into $IDF_PATH. This way clangd can locate compile_commands.json also for IDF source files, should you need to use "go to definition" for IDF-provided functions.

If you don't use "go to definition" for IDF-provided functions often, you can point --compile-commands-dir to your project build dir, instead.

Another option, if both IDF and your project are under some common directory (let's say `~/esp/`), you can have the post-build step copy the compile_commands.json file there, and point --compile-commands-dir to that common top level directory.

OrhanYigitDurmaz, 2025-05-05:
> FWIW here is the zed config which works for me:
>
> ```
> {
>     "lsp": {
>         "clangd": {
>             "path_lookup": true,
>             "binary": {
>                 "path": [...]

this works nicely

the only downside is to manually set the dir names

BUT..

my project is in my home root folder, and only doing this config with compile commands set to the build dir inside project dir, it can still find esp headers!

[image]

I hope zed adds some more support into extensions so we can have our own esp-idf extension.

Thanks for the config again.