espressif/esp-idf #15781: Receiving of raw QoS data via promiscuous mode not possible

archef2000, 2025-04-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-S3 QFN56 rev 0.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3 Dev Kit dual type-c

### Power Supply used.

USB

### What is the expected behavior?

Get all packets send from other devices even when the qos data flag is set

### What is the actual behavior?

Missing/currupted wifi frames in promiscuous_rx_cb and cryptic wifi logs

### Steps to reproduce.

I can't find any way to send these QoS Data frames except for using two Apple devices during AirDrop. (Tried with esp (send_test in main.c) and linux)

In Wireshark they are printed as is, but in the terminal they never show up.

### Debug Logs.

```plain
Time of data transfer between Apple iPhones in AirDrop:

I (19446) wifi:reload(3fcb0c80, 3fcb0c98)
I (19446) wifi:19101896 (b0c38, 30b0c68, b0c74, b0c68, b0c50)
I (19446) wifi:(3fcb0ca4, 3fcb0c74, 0, 30098c3, 270)
I (19446) wifi:19101897 (b0c38, 30b0c68, b0c74, b0c68, b0c50)
I (19446) wifi:(3fcb0ca4, 3fcb0c74, 3fcb0c80, 30098c3, 270)
I (19456) wifi:19116915 (b0c38, 30b0c68, b0c74, b0c68, b0c50)
I (19466) wifi:(3fcb0ca4, 3fcb0c74, 3fcb0c80, 3539cc3, 270)
D4, 00
 ; rssi: -10
D4 00 00 00 62 50 4F 8E 9A 95 
D4, 00
 ; rssi: -10
D4 00 00 00 62 50 4F 8E 9A 95 
(dest) ADDR1=4e:ee:b0:28:44:0c
D0, 00
D0 00 30 00 4E EE B0 28 44 0C 
I (19496) wifi:reload(3fcb0c44, 3fcb0c5c)
I (19496) wifi:19157031 (b0c38, 30b0ca4, b0c38, b0ca4, b0c8c)
I (19496) wifi:(3fcb0c68, 3fcb0c38, 0, 300983e, a70)
I (19496) wifi:19157033 (b0c38, 30b0ca4, b0c38, b0ca4, b0c8c)
I (19506) wifi:(3fcb0c68, 3fcb0c38, 3fcb0c44, 300983e, a70)
I (19506) wifi:19172042 (b0c38, 30b0ca4, b0c38, b0ca4, b0c8c)
I (19516) wifi:(3fcb0c68, 3fcb0c38, 3fcb0c44, 365983e, a70)
B4, 00
 ; rssi: -11
B4 00 62 01 62 50 4F 8E 9A 95 
94, 00
 ; rssi: -26
94 00 00 00 4E EE B0 28 44 0C 
B4, 00
 ; rssi: -11
B4 00 1A 01 62 50 4F 8E 9A 95 
I (19546) wifi:reload(3fcb0c38, 3fcb0c50)
I (19546) wifi:19210516 (b0c38, 30b0c98, b0ca4, b0c98, b0c80)
I (19556) wifi:(3fcb0c5c, 3fcb0ca4, 0, 13009817, 270)
I (19556) wifi:19210518 (b0c38, 30b0c98, b0ca4, b0c98, b0c80)
I (19556) wifi:(3fcb0c5c, 3fcb0ca4, 3fcb0c38, 13009817, 270)
I (19566) wifi:19225527 (b0c38, 30b0c98, b0ca4, b0c98, b0c80)
I (19566) wifi:(3fcb0c5c, 3fcb0ca4, 3fcb0c38, 13539817, 270)
I (19586) wifi:reload(3fcb0c5c, 3fcb0c74)
I (19586) wifi:19250224 (b0c38, 30b0c44, b0c50, b0c44, b0ca4)
I (19586) wifi:(3fcb0c80, 3fcb0c50, 0, 3009877, 270)
I (19596) wifi:19250226 (b0c38, 30b0c44, b0c50, b0c44, b0ca4)
I (19596) wifi:(3fcb0c80, 3fcb0c50, 3fcb0c5c, 3009877, 270)
I (19606) wifi:19265235 (b0c38, 30b0c44, b0c50, b0c44, b0ca4)
I (19606) wifi:(3fcb0c80, 3fcb0c50, 3fcb0c5c, 3519877, 270)
I (19626) wifi:reload(3fcb0c80, 3fcb0c98)
I (19626) wifi:19289670 (b0c38, 30b0c68, b0c74, b0c68, b0c50)
I (19626) wifi:(3fcb0ca4, 3fcb0c74, 0, 23009cd4, a00270)
I (19636) wifi:19289672 (b0c38, 30b0c68, b0c74, b0c68, b0c50)
I (19636) wifi:(3fcb0ca4, 3fcb0c74, 3fcb0c80, 23009cd4, a00270)
I (19646) wifi:19304681 (b0c38, 30b0c68, b0c74, b0c68, b0c50)
I (19646) wifi:(3fcb0ca4, 3fcb0c74, 3fcb0c80, 236598d4, a00270)
```

### Diagnostic report archive.

code: https://github.com/archef2000/esp-owl/tree/wifi-dev

[diag-8da14cbb-fd4e-425c-8704-58984cfebd5a.zip](https://github.com/user-attachments/files/19742137/diag-8da14cbb-fd4e-425c-8704-58984cfebd5a.zip)

### More Information.

I sadly, can't provide a better example to send these wifi frames

zhangyanjiaoesp, 2025-04-15:
@archef2000 
Do you mean that the Qos data between two Apple devices can be captured by Wireshark, but can't be captured by the promiscuous mode on ESP32S3?
Can you provide the capture packets by the Wireshark?

zhangyanjiaoesp, 2025-04-15:
@archef2000 
1. Are you sure the Qos data is on channel 6?
`wifi_sniffer_set_channel(6);`
2. You can set the promiscuous filter to `WIFI_PROMIS_FILTER_MASK_DATA` if you only care the Qos data packets.
3. In the `wifi_sniffer_packet_handler`, you can check the Qos data like this:
```
static void wifi_sniffer_packet_handler(void *buf, wifi_promiscuous_pkt_type_t type) {
    const wifi_promiscuous_pkt_t *pkt = (wifi_promiscuous_pkt_t *)buf;
    const uint8_t *data = pkt->payload;

    // get Frame Control filed
    uint8_t frame_control_1 = data[0];
    uint8_t frame_control_2 = data[1];

    uint8_t type = (frame_control_1 >> 2) & 0x03;
    uint8_t subtype = (frame_control_1 >> 4) & 0x0F;

    if (type == 2 && subtype == 8) {
        printf("QoS Data frame captured!\n");
        ...
    }
}

```

archef2000, 2025-04-15:
The time delta from pcap to log is small (650ms) and the last two packets in the pcap match the log 

The filter applied to the wireshark pcap (this is also implemented in the [wifi-dev branch](https://github.com/archef2000/esp-owl/tree/5b70d4573b0d6c7405356810463bea9b87e05571))
`
wlan_radio.signal_dbm > -30 && !awdl && !wlan.fc.type_subtype == 0x1c
`

<details>
<summary>log</summary>

```
I (385) main_task: Calling app_main()
I (385) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
Interrupt configured. Press the button to restart the chip.
E (395) wifi_sniffer_init: [main/wifi/capture.c:269] awdl state set
I (405) pp: pp rom version: e7ae62f
I (405) net80211: net80211 rom version: e7ae62f
I (415) wifi:wifi driver task: 3fcae434, prio:23, stack:6656, core=0
I (415) wifi:wifi firmware version: 79fa3f41ba
I (415) wifi:wifi certification version: v7.0
I (415) wifi:config NVS flash: enabled
I (425) wifi:config nano formatting: disabled
I (425) wifi:Init data frame dynamic rx buffer num: 32
I (435) wifi:Init static rx mgmt buffer num: 5
I (435) wifi:Init management short buffer num: 32
I (445) wifi:Init dynamic tx buffer num: 32
I (445) wifi:Init static tx FG buffer num: 2
I (445) wifi:Init static rx buffer size: 1600
I (455) wifi:Init static rx buffer num: 10
I (455) wifi:Init dynamic rx buffer num: 32
I (465) wifi_init: rx ba win: 6
I (465) wifi_init: accept mbox: 6
I (465) wifi_init: tcpip mbox: 32
I (465) wifi_init: udp mbox: 6
I (475) wifi_init: tcp mbox: 6
I (475) wifi_init: tcp tx win: 5760
I (475) wifi_init: tcp rx win: 5760
I (485) wifi_init: tcp mss: 1440
I (485) wifi_init: WiFi IRAM OP enabled
I (485) wifi_init: WiFi RX IRAM OP enabled
I (495) wifi:(2.4G)set country: cc=US schan=1 nchan=13 policy=0

I (495) phy_init: phy_version 700,8582a7fd,Feb 10 2025,20:13:11
E (535) wifi:failed to post WiFi event=43 ret=259
I (535) wifi:mode : sta (80:65:99:c7:ae:98)
I (535) wifi:enable tsf
E (535) wifi:failed to post WiFi event=2 ret=259
I (535) wifi:ic_enable_sniffer
I (535) awdl: mac: 80:65:99:c7:ae:98
capture: aw_period:
I (545) awdl core: switch channel to 6 (slot 0)
psf_timer interval us: 112640
E (555) wifi:failed to post WiFi event=43 ret=259
I (555) awdl: (awdl_create) creating awdl driver
awdl_post_attach
awdl_netif_init
ipv6_addr: fe80:0000:0000:0000:8265:99ff:fec7:ae98
I (565) main_task: Returned from app_main()
I (4375) sniffer: type: 0; subtype: 13; rssi -6; len 334
D0 10 00 00 FF FF FF FF FF FF BA 32 81 B3 D1 14 00 25 00 FF 94 73 60 FF 7F 00 17 F2 08 10
I (6125) sniffer: type: 1; subtype: 11; rssi -6; len 16
B4 00 AA 00 62 50 4F 8E 9A 95 BA 32 81 B3 D1 14
I (6125) sniffer: type: 1; subtype: 11; rssi -6; len 16
B4 00 AA 00 62 50 4F 8E 9A 95 BA 32 81 B3 D1 14
I (6135) sniffer: type: 1; subtype: 11; rssi -6; len 16
B4 00 AA 00 62 50 4F 8E 9A 95 BA 32 81 B3 D1 14
I (6145) sniffer: type: 1; subtype: 11; rssi -6; len 16
B4 00 F6 01 62 50 4F 8E 9A 95 BA 32 81 B3 D1 14
I (6165) sniffer: type: 1; subtype: 13; rssi -6; len 10
D4 00 00 00 62 50 4F 8E 9A 95
I (6165) sniffer: type: 1; subtype: 13; rssi -6; len 10
D4 00 00 00 62 50 4F 8E 9A 95
I (6275) sniffer: type: 2; subtype: 0; rssi -6; len 132
08 00 00 00 33 33 80 00 00 FB BA 32 81 B3 D1 14 00 25 00 FF 94 73 E0 00 AA AA 03 00 17 F2
I (7195) sniffer: type: 2; subtype: 0; rssi -6; len 132
08 00 00 00 33 33 80 00 00 FB BA 32 81 B3 D1 14 00 25 00 FF 94 73 E0 01 AA AA 03 00 17 F2
```
</details>

The printed bytes are capped at 30 to fit in one line

[qos2.pcapng.txt](https://github.com/user-attachments/files/19761705/qos2.pcapng.txt)

archef2000, 2025-04-15:
> [@archef2000](https://github.com/archef2000) Do you mean that the Qos data between two Apple devices can be captured by Wireshark, but can't be captured by the promiscuous mode on ESP32S3? Can you provide the capture packets by the Wireshark?

This is a capture during the discovery of two iPhones in AirDrop. During that they are using QoS Data exchange on channel 6.
I don't only care for the QoS data, but need them to continue with my main projects.

zhangyanjiaoesp, 2025-04-16:
@archef2000 

> [qos2.pcapng.txt](https://github.com/user-attachments/files/19761705/qos2.pcapng.txt)

Is this capture same with the log？What we need is the capture packets by **Wireshark**, not the capture by ESP32S3. We want to use the Wireshark capture to check whether the QoS Data packets sent by the iPhone have any special encoding format (such as LDPC) that might prevent the ESP32-S3 from correctly parsing the contents

zhangyanjiaoesp, 2025-04-16:
@archef2000 
According to the capture packets, the encoding format for the Qos data is **LDPC**, the ESP32S3 can only record the length of this packets, in the `wifi_sniffer_packet_handler()`, the `wifi_promiscuous_pkt_type_t type` will be `WIFI_PKT_MISC` .

[image]

archef2000, 2025-04-16:
And why can't the esp record the data? Why is that format not supported?

zhangyanjiaoesp, 2025-04-16:
@archef2000 
Currently, the hardware does not support parsing packets encoded with LDPC

archef2000, 2025-04-16:
Is it a hardware limitation or wifi binary blob code that does not handle it? Are there any esp models that support paring it? Will there be support for it in the esp32C5?

archef2000, 2025-04-16:
I found this and I am using the esp32s3 https://docs.espressif.com/projects/esp-faq/en/latest/software-framework/wifi.html#does-esp32-support-ldpc so it should work

zhangyanjiaoesp, 2025-04-16:
> I found this and I am using the esp32s3 https://docs.espressif.com/projects/esp-faq/en/latest/software-framework/wifi.html#does-esp32-support-ldpc so it should work

This doc is wrong, will update it ASAP.

archef2000, 2025-04-16:
And for what is this flag if nothing is using it? https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s3/api-reference/network/esp_wifi.html#_CPPv4N18wifi_pkt_rx_ctrl_t10fec_codingE

zhangyanjiaoesp, 2025-04-16:
> And for what is this flag if nothing is using it? https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s3/api-reference/network/esp_wifi.html#_CPPv4N18wifi_pkt_rx_ctrl_t10fec_codingE

Currently the value is 0(BCC). In the future, when we support LDPC, this bit can be set based on the received packet.

archef2000, 2025-04-16:
So it is a software limitation?

zhangyanjiaoesp, 2025-04-16:
> So it is a software limitation?

The inability to parse LDPC is a hardware limitation. The `struct wifi_pkt_rx_ctrl_t` simply records some information extracted during packet reception. I don't understand why this would cause confusion for you.

zhangyanjiaoesp, 2025-04-16:
> > > And for what is this flag if nothing is using it? https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s3/api-reference/network/esp_wifi.html#_CPPv4N18wifi_pkt_rx_ctrl_t10fec_codingE
> > 
> > 
> > Currently the value is 0(BCC). In the future, when we support LDPC, this bit can be set based on the received packet.
> 
> The "In the future, when we support LDPC" sounded like it was a planned software feature.

Alright, I'm sorry for causing the misunderstanding.

zhangyanjiaoesp, 2025-04-17:
@archef2000 If there are no further questions, I think this issue can be closed.

archef2000, 2025-04-18:
Could you tell me what these repeated wifi logs are even when i have no wifi debug enabled and disabled any tx?
```
I (86086) wifi:reload(3fcb21e0, 3fcb21f8)
I (86086) wifi:85710111 (b21e0, 30b2240, b224c, b2240, b2228)
I (86086) wifi:(3fcb2204, 3fcb224c, 0, 13009845, 270)
I (86096) wifi:85710113 (b21e0, 30b2240, b224c, b2240, b2228)
I (86096) wifi:(3fcb2204, 3fcb224c, 3fcb21e0, 13009845, 270)
I (86106) wifi:85725123 (b21e0, 30b2240, b224c, b2240, b2228)
I (86106) wifi:(3fcb2204, 3fcb224c, 3fcb21e0, 13519845, 270)
I (86126) wifi:reload(3fcb2204, 3fcb221c)
I (86126) wifi:85750011 (b21e0, 30b21ec, b21f8, b21ec, b224c)
I (86126) wifi:(3fcb2228, 3fcb21f8, 0, 30098f1, a70)
I (86136) wifi:85750013 (b21e0, 30b21ec, b21f8, b21ec, b224c)
I (86136) wifi:(3fcb2228, 3fcb21f8, 3fcb2204, 30098f1, a70)
I (86146) wifi:85765023 (b21e0, 30b21ec, b21f8, b21ec, b224c)
I (86146) wifi:(3fcb2228, 3fcb21f8, 3fcb2204, 30098f1, a70)
```

archef2000, 2025-04-19:
Looks like the latest esp idf version has set the default wifi log level to INFO

zhangyanjiaoesp, 2025-04-21:
> Could you tell me what these repeated wifi logs are even when i have no wifi debug enabled and disabled any tx?
>
> ```
> I (86086) wifi:reload(3fcb21e0, 3fcb21f8)
> I (86086) wifi:85710111 (b21e0, [...]

@archef2000 This log related to the rx buffer reload. Is it easy for you to encounter this kind of log during your testing?

archef2000, 2025-04-21:
Yes it came up every few seconds. Until I disabled it by setting the log level to warnings for wifi even when I never had enabled info log.

archef2000, 2025-04-21:
It is 135/3 minutes in this log

<details>
<summary>log</summary>

```
Opened with baud rate: 115200
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2f00
entry 0x403c8928
I (27) boot: ESP-IDF v5.4.1-dirty 2nd stage bootloader
I (27) boot: compile time Apr 19 2025 20:48:34
I (27) boot: Multicore bootloader
I (28) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00177000
I (78) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=26d00h (158976) map
I (117) esp_image: segment 1: paddr=00036d28 vaddr=3fc9b700 size=050bch ( 20668) load
I (122) esp_image: segment 2: paddr=0003bdec vaddr=40374000 size=0422ch ( 16940) load
I (126) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=b21e8h (729576) map
I (257) esp_image: segment 4: paddr=000f2210 vaddr=4037822c size=133f0h ( 78832) load
I (274) esp_image: segment 5: paddr=00105608 vaddr=600fe100 size=0001ch (    28) load
I (284) boot: Loaded app from partition at offset 0x10000
I (284) boot: Disabling RNG early entropy source...
I (294) cpu_start: Multicore app
I (303) cpu_start: Pro cpu start user code
I (303) cpu_start: cpu freq: 240000000 Hz
I (303) app_init: Application information:
I (303) app_init: Project name:     owl
I (307) app_init: App version:      build-25-g653deac-dirty
I (312) app_init: Compile time:     Apr 20 2025 18:05:26
I (317) app_init: ELF file SHA256:  62fe3ad3a...
I (322) app_init: ESP-IDF:          v5.4.1-dirty
I (326) efuse_init: Min chip rev:     v0.0
I (330) efuse_init: Max chip rev:     v0.99 
I (334) efuse_init: Chip rev:         v0.2
I (338) heap_init: Initializing. RAM available for dynamic allocation:
I (344) heap_init: At 3FCA5370 len 000443A0 (272 KiB): RAM
I (349) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (354) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (359) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (366) spi_flash: detected chip: generic
I (368) spi_flash: flash io: dio
W (371) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (384) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (390) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (397) coexist: coex firmware version: e727207
I (415) coexist: coexist rom version e7ae62f
I (416) main_task: Started on CPU0
I (426) main_task: Calling app_main()
I (426) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
Interrupt configured. Press the button to restart the chip.
E (436) wifi_sniffer_init: [main/wifi/capture.c:270] awdl state set
dns_init: initializing
[... 3174 lines cut ...]
I (180676) wifi:(3fcb34b0, 3fcb34f8, 0, 300985c, 270)
I (180676) wifi:180294058 (b3498, 30b34ec, b34f8, b34ec, b34d4)
I (180686) wifi:(3fcb34b0, 3fcb34f8, 3fcb3504, 3009c5c, 270)
I (180686) wifi:180309073 (b3498, 30b34ec, b34f8, b34ec, b34d4)
I (180696) wifi:(3fcb34b0, 3fcb34f8, 3fcb3504, 300985c, 270)
180471369:178571855 0
180137981:178571855 0

181495952:179572670 0
181502881:179572670 0
awdl_action: unexpected bytes (343) at end of frameA9 F2 EE 26 08 8B 7E DE 7B CC 34 35 D6 34 6A 92 99 91 63 F0 67 DF A1 BE 6D A2 85 B1 61 22 44 DD 5E 92 B1 13 E0 A8 BE FF 72 30 13 BA EA 5B 6D 37 AE 2D 8E AF C9 77 4F A0 BE 7F 8B 99 99 83 30 13 A0 64 B7 D7 19 84 31 3D FC 10 AD 45 17 58 D7 4B 50 BE EA 8B CF 15 95 98 31 21 1D 78 20 EF 4E 8C E6 5B 6D 44 AC A9 A6 64 37 86 30 EB FA 7E 9E C0 BA 8C A1 DF DD DC 99 40 C3 55 90 F1 64 7F 9E CC 55 1A A4 21 25 09 6A 4A 11 08 13 30 BB C7 C3 00 06 30 B4 EA A1 DC 1E 1E 6D 42 FB 55 FA D6 B6 12 CA 06 16 A0 64 B7 98 E8 9C CE 4F 30 13 A4 E4 37 6A 2A E7 4E 36 81 55 87 C0 F8 BD E9 7E C4 7A D4 D3 84 72 C7 19 C8 00 14 55 F5 AC D9 E6 F1 9C 06 AE 81 59 34 3F 00 46 F8 C6 EE 94 DC ED 4E 10 A6 D7 21 C6 1E AA F0 83 3B FB 66 77 8F 80 7D C6 64 74 2B 87 AB 95 96 59 8A 98 95 F8 08 72 74 9F A9 C3 CD 99 FB 9E 53 B7 8B FD 71 22 B1 64 7B C0 F2 59 89 B1 1B DE 24 C4 12 01 15 E2 5A EB 67 37 4E F9 BD 7F 8B 17 97 26 9B 3F 72 35 6A 67 DC 42 69 75 DB E7 B8 64 3B 8F 91 6D 21 DD CC 0E 8B 59 44 CF 1C C8 D8 89 D3 4E AC C1 49 F9 A0 34 F7 FF 0D A4 A9 C2 00 00 00 00 
W (182496) awdl core: unhandled frame (-1)
I (182516) wifi:reload(3fcb34a4, 3fcb34bc)
I (182516) wifi:182136809 (b3498, 30b3504, b3498, b3504, b34ec)
I (182516) wifi:(3fcb34c8, 3fcb3498, 0, 100985f, 270)
I (182516) wifi:182136811 (b3498, 30b3504, b3498, b3504, b34ec)
I (182526) wifi:(3fcb34c8, 3fcb3498, 3fcb34a4, 100985f, 270)
I (182536) wifi:182151820 (b3498, 30b3504, b3498, b3504, b34ec)
I (182536) wifi:(3fcb34c8, 3fcb3498, 3fcb34a4, 100985f, 270)
182522386:180573483 0
182519533:180573483 0
183554315:181574297 0
183540126:181574297 
```
</details>