espressif/esp-idf #15783: I2C and I2C Communication Conflict

KCYoong, 2025-04-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.3.2

### Espressif SoC revision.

ESP32-S3-WROOM-1 N8R8

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

My application have use the LED Matrix Display which is used I2S Communication (detail on more information) and a number of I2C sensor. I am expected the LED Matrix Display and I2C sensor should be work together without problem.

### What is the actual behavior?

Previously, I use the LED Matrix Display with I2C Sensor (VEML-7700 & ADXL-345) without problem, but the problem occur after I replace the ADXL-345 by ICM-42688 sensor.
The system will occurs "I2C transaction unexpected nack detected" with ICM-42688 sensor.

I have did some analysis found the I2C able communicate without use of I2S.
So, I think most possible the I2C conflict with I2S.

### Steps to reproduce.

I am not ideal how this bug trigger, I would like to share my I2C configuration here

`// I2C_Master Bus Configuration
esp_err_t I2C_Master_Initial_BusConfiguration(int I2C_MASTER_NUM, int I2C_MASTER_SCL_IO, int I2C_MASTER_SDA_IO)
{
	i2c_master_bus_config_t i2c_bus_config = {
		.clk_source = I2C_CLK_SRC_DEFAULT,
		.i2c_port = I2C_MASTER_NUM,
		.scl_io_num = I2C_MASTER_SCL_IO,
		.sda_io_num = I2C_MASTER_SDA_IO,
		.glitch_ignore_cnt = 7,
		.flags.enable_internal_pullup = true,
	};

	// Initialize I2C Bus
	esp_err_t bus_error = i2c_new_master_bus(&i2c_bus_config, &bus_handle);
	if (bus_error == ESP_ERR_INVALID_ARG) {
		ESP_LOGE("I2C_BUS", "I2C Bus Configuration invalid!");
	} else if (bus_error == ESP_ERR_NOT_FOUND) {
		ESP_LOGE("I2C_BUS", "No available DMA channel!");
	} else if (bus_error == ESP_ERR_NO_MEM) {
		ESP_LOGE("I2C_BUS", "Out of memory!");
	} else if (bus_error != ESP_OK) {
        ESP_LOGE("I2C_BUS", "Error Code: %d \n", bus_error);
    } 
	return bus_error;
}

// I2C_Master Device Configuration
esp_err_t I2C_Master_Initial_DeviceConfiguration(i2c_master_dev_handle_t *dev_handle, uint8_t device_address)
{
	i2c_device_config_t dev_cfg = {
		.dev_addr_length = I2C_ADDR_BIT_LEN_7,
		.device_address = device_address,
		.scl_speed_hz = I2C_MASTER_FREQ_HZ,
	};
	
	esp_err_t device_error = i2c_master_bus_add_device(bus_handle, &dev_cfg, dev_handle);
	if (device_error == ESP_ERR_INVALID_ARG) {
		ESP_LOGE("I2C_DEVICE ", "I2C Device parameter is invalid!");
	} else if (device_error == ESP_ERR_NO_MEM) {
		ESP_LOGE("I2C_DEVICE", "Out of memory!");
	} else if (device_error != ESP_OK) {
		ESP_LOGE("I2C_DEVICE", "Error Code: %d \n", device_error);
	}
	return device_error;
}`

### Debug Logs.

```plain
I (27) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (27) boot: compile time Apr 15 2025 13:51:45
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot: efuse block revision: v1.3
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0f6d0h ( 63184) map
I (115) esp_image: segment 1: paddr=0001f6f8 vaddr=3fc93a00 size=00920h (  2336) load
I (116) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=237b0h (145328) map
I (147) esp_image: segment 3: paddr=000437d8 vaddr=3fc94320 size=027b8h ( 10168) load
I (149) esp_image: segment 4: paddr=00045f98 vaddr=40374000 size=0f934h ( 63796) load
I (173) boot: Loaded app from partition at offset 0x10000
I (173) boot: Disabling RNG early entropy source...
I (185) cpu_start: Multicore app
I (194) cpu_start: Pro cpu start user code
I (194) cpu_start: cpu freq: 160000000 Hz
I (194) app_init: Application information:
I (197) app_init: Project name:     app-template
I (202) app_init: App version:      1
I (207) app_init: Compile time:     Apr 15 2025 13:50:00
I (213) app_init: ELF file SHA256:  0b5bcc093fa5f61f...
I (219) app_init: ESP-IDF:          v5.3.1-dirty
I (224) efuse_init: Min chip rev:     v0.0
I (229) efuse_init: Max chip rev:     v0.99 
I (234) efuse_init: Chip rev:         v0.2
I (238) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FC97608 len 00052108 (328 KiB): RAM
I (252) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (258) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (264) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (271) spi_flash: detected chip: gd
I (274) spi_flash: flash io: dio
W (278) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (292) sleep: Configure to isolate all GPIO pins in sleep state
I (298) sleep: Enable automatic switching of GPIO sleep configuration
I (306) main_task: Started on CPU0
I (326) main_task: Calling app_main()
I (326) gpio: GPIO[1]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (326) begin(): Using GPIO 41 for R1_PIN
I (336) begin(): Using GPIO 21 for G1_PIN
I (336) begin(): Using GPIO 40 for B1_PIN
I (346) begin(): Using GPIO 39 for R2_PIN
I (346) begin(): Using GPIO 47 for G2_PIN
I (356) begin(): Using GPIO 42 for B2_PIN
I (356) begin(): Using GPIO 37 for A_PIN
I (366) begin(): Using GPIO 48 for B_PIN
I (366) begin(): Using GPIO 36 for C_PIN
I (366) begin(): Using GPIO -1 for D_PIN
I (376) begin(): Using GPIO -1 for E_PIN
I (376) begin(): Using GPIO 45 for LAT_PIN
I (386) begin(): Using GPIO 2 for OE_PIN
I (386) begin(): Using GPIO 35 for CLK_PIN
I (396) I2S-DMA: Free heap: 376228
I (396) I2S-DMA: Free SPIRAM: 0
I (406) I2S-DMA: allocating rowBitStructs with pixel_color_depth_bits of 8
I (406) I2S-DMA: Allocating 8192 bytes memory for DMA BCM framebuffer(s).
I (416) I2S-DMA: Minimum visual refresh rate (scan rate from panel top to bottom) requested: 60 Hz
W (426) I2S-DMA: lsbMsbTransitionBit of 0 gives 61 Hz refresh rate.
I (436) I2S-DMA: DMA has pixel_color_depth_bits of 8
I (436) I2S-DMA: Recalculated number of DMA descriptors per row: 128
I (446) I2S-DMA: 1024 DMA descriptors linked to buffer data.
I (456) S3: Clock divider is 16
I (456) I2S-DMA: DMA setup completed
D (966) i2c.common: new bus(0) at 0x3fca1e18
I (966) gpio: GPIO[14]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (966) gpio: GPIO[13]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
D (976) i2c.common: bus clock source frequency: 40000000hz
Transmit Data: 0 12
DataWrite: 0 0 12
I (986) VEML7700: ALS Configuration: 12 00
DataWrite: 11 1
E (1086) i2c.master: I2C transaction unexpected nack detected
E (1086) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (1086) i2c.master: i2c_master_transmit_receive(1097): I2C transaction failed
E (1096) I2C_READ: Error Code: 259
```

### More Information.

Software: Espressif-IDE

**I2C Components without problem**
1. VEML-7700 (Light Sensor)
2. ADXL-345 (Accelerometer)

**I2C Components had problem**
1. VEML-7700 (Light Sensor)
2. ICM-42688 (IMU Sensor)

**I2S Components**
1. LED Matrix Display with HUB75
Library: https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA

mythbuster5, 2025-04-15:
Connect logic analizer see what happened on line please. Seems changing sensor leads to this problem, a good way to check is to see the  online data and figure out why nack is given by the sensor.

BTW should your title be I2S AND I2C....?

KCYoong, 2025-04-15:
> Connect logic analizer see what happened on line please. Seems changing sensor leads to this problem, a good way to check is to see the online data and figure out why nack is given by the sensor.
> 
> BTW should your title be I2S AND I2C....?

Sorry about that, my side don't have the logic analyzer

mythbuster5, 2025-04-15:
Thet's strange because your previous sensor works and change sensor it can't work. So the thinking would be what's difference between two sensors, like i2c side, stretch? pins pull-up? etc. As you said without i2s, it works, plus i2s it doesn't work. But there is no source shared between i2s and i2c. If you can provide more direct evidence it would be thankful. (Actually see the data online is best)

mythbuster5, 2025-04-21:
Or @KCYoong Could you please help provide simplest code to reproduce? That's would be helpful to locate the issue.

KCYoong, 2025-04-23:
@mythbuster5 Sorry for late reply, I have draft the example code for reproduce problem, please have a look on it.
I have try another sensor with same code, but they are work fine, do it possible the ICM42688 sensor have problem? 

I unable to upload the project file, I share the google link for this project file
Link: [](https://drive.google.com/drive/folders/1Wj1GS7OlvhiA7Pqs71lbCRVoFhofvH96?usp=sharing)

Output Log: 

```
I (27) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (27) boot: compile time Apr 23 2025 10:53:12
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot: efuse block revision: v1.3
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0d670h ( 54896) map
I (114) esp_image: segment 1: paddr=0001d698 vaddr=3fc92f00 size=02980h ( 10624) load
I (117) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=203c0h (132032) map
I (144) esp_image: segment 3: paddr=000403e8 vaddr=3fc95880 size=00318h (   792) load
I (145) esp_image: segment 4: paddr=00040708 vaddr=40374000 size=0eec8h ( 61128) load
I (170) boot: Loaded app from partition at offset 0x10000
I (170) boot: Disabling RNG early entropy source...
I (182) cpu_start: Multicore app
I (191) cpu_start: Pro cpu start user code
I (191) cpu_start: cpu freq: 160000000 Hz
I (191) app_init: Application information:
I (194) app_init: Project name:     without-gfx
I (199) app_init: App version:      1
I (203) app_init: Compile time:     Apr 23 2025 13:12:50
I (209) app_init: ELF file SHA256:  7e3b668af...
I (215) app_init: ESP-IDF:          v5.3.2-dirty
I (220) efuse_init: Min chip rev:     v0.0
I (225) efuse_init: Max chip rev:     v0.99 
I (230) efuse_init: Chip rev:         v0.2
I (234) heap_init: Initializing. RAM available for dynamic allocation:
I (242) heap_init: At 3FC964C8 len 00053248 (332 KiB): RAM
I (248) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (254) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (260) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (267) spi_flash: detected chip: gd
I (271) spi_flash: flash io: dio
W (274) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (288) sleep: Configure to isolate all GPIO pins in sleep state
I (295) sleep: Enable automatic switching of GPIO sleep configuration
I (302) main_task: Started on CPU0
I (322) main_task: Calling app_main()
I (322) begin(): Using GPIO 41 for R1_PIN
I (322) begin(): Using GPIO 21 for G1_PIN
I (322) begin(): Using GPIO 40 for B1_PIN
I (332) begin(): Using GPIO 39 for R2_PIN
I (332) begin(): Using GPIO 47 for G2_PIN
I (342) begin(): Using GPIO 42 for B2_PIN
I (342) begin(): Using GPIO 37 for A_PIN
I (352) begin(): Using GPIO 48 for B_PIN
I (352) begin(): Using GPIO 36 for C_PIN
I (362) begin(): Using GPIO -1 for D_PIN
I (362) begin(): Using GPIO -1 for E_PIN
I (362) begin(): Using GPIO 45 for LAT_PIN
I (372) begin(): Using GPIO 2 for OE_PIN
I (372) begin(): Using GPIO 35 for CLK_PIN
I (382) I2S-DMA: Free heap: 385516
[... 44 lines cut ...]
E (2582) i2c.master: i2c_master_transmit_receive(1139): I2C transaction failed
I (2592) ICM42688: Data Read: 00 00
I (2692) VEML7700: Read data: 01 51

Display Blank.
E (2892) i2c.master: I2C transaction unexpected nack detected
E (2892) i2c.master: s_i2c_synchronous_transaction(888): I2C transaction failed
E (2892) i2c.master: i2c_master_transmit_receive(1139): I2C transaction failed
I (2902) ICM42688: Data Read: 00 00
I (3002) VEML7700: Read data: 01 52

Display Blank.
E (3202) i2c.master: I2C transaction unexpected nack detected
E (3202) i2c.master: s_i2c_synchronous_transaction(888): I2C transaction failed
E (3202) i2c.master: i2c_master_transmit_receive(1139): I2C transaction failed
I (3212) ICM42688: Data Read: 80 01
I (3312) VEML7700: Read data: 01 52

Display Blank.
E (3512) i2c.master: I2C transaction unexpected nack detected
E (3512) i2c.master: s_i2c_synchronous_transaction(888): I2C transaction failed
E (3512) i2c.master: i2c_master_transmit_receive(1139): I2C transaction failed
I (3522) ICM42688: Data Read: 00 00
I (3622) VEML7700: Read data: 01 51
```

mythbuster5, 2025-04-23:
No conclusion,, anyway, I will purchase one for debugging.

eriksl, 2025-04-23:
I have a similar problem. If just one sensor reacts with a NAK (because it's not ready yet), several other sensors will also yield a NAK and actually I don't think that's really happening, it's just the IDF code that's "seeing" NAKs, after the initial NAK.

I didn't see this behaviour on my ESP8266 (with my own I2C bit banging implementation). I'm not sure at which point of the rising state of the SCK line SDA is sampled? Maybe some sensors are a bit slow to react and for that reason it's preferable to sample SDA just a moment later than at the exact moment SCK comes up.

mythbuster5, 2025-04-23:
@eriksl  I‘m not sure you are facing the same issue. But your issue (one nack cause a normal device nack) I also noticed these days, will fix very soon.

eriksl, 2025-04-23:
I am experiencing this when no bus error state is going on (i.e. SDA and SCL are high and all devices are in HiZ). No slave is configured to do clock stretching (which, I think, is evil).

So what happens is that one slave does not ACK (so it does NAK) and after that, when polling, several slaves are seen as also not ACKing, during a short time. This might indicate some bus reset mechanism on the ESP32-S3 is initiated, but it's really not required at this stage.

But I also suspect the SDA line being sampled too eagerly, some slaves may need a few ns more to react.

pggh, 2025-04-24:
Just a thought: could it be a pure hardware problem? I don't know how your hardware looks like (neither logically nor the physical build). LED matrix display create huge current transients that can cause all sorts of problems. This might affect your I2C chips depending on the sensitivity of the them and the cables etc.
Could you retest with all software running but the LED display disconnected or somehow disabled?

mythbuster5, 2025-04-25:
[image]

Works fine on my side.

KCYoong, 2025-04-28:
> Just a thought: could it be a pure hardware problem? I don't know how your hardware looks like (neither logically nor the physical build). LED matrix display create huge current transients that can [...]

You are right, when i disconnect power of the LED Matrix Display, the I2C communication work properly, after i re-connect the power of LED Matrix Display, the I2C communication problem of ICM-42688 occurs again.

[image]

mythbuster5, 2025-05-08:
If it's hardware related and no more update, I will conclude this issue.

closed 2025-05-08 by mythbuster5

KCYoong, 2025-05-09:
Sorry for late reply, I also think it is hardware issues.
Thank you very much for support and helping to solve my problem.