espressif/esp-idf #15785: Required mesh information

gajananmaske1234, 2025-04-15:
### Is your feature request related to a problem?

I have used the library code from the "No Router Message Exchange" example (reference: https://github.com/espressif/esp-idf/issues/15665) as the basis for my development.

For the current implementation (reference: [Link2](https://github.com/nopnop2002/esp-mesh-lite-examples)), I would like to know:

Q 1 : What is the maximum number of devices that can be connected in the mesh network?

nopnop2002, 2025-04-15:
https://docs.espressif.com/projects/esp-techpedia/en/latest/esp-friends/solution-introduction/mesh/mesh-comparison.html

gajananmaske1234, 2025-04-16:
Thanks for quick response it will be helpful for me .

zhangyanjiaoesp, 2025-05-07:
@gajananmaske1234 If no more question, can we close this?

closed 2025-06-10 by Alvin1Zhang