espressif/esp-idf #15787: Request for official test reports for ESP32-S3-WROOM-1U-N4 (HTOL, EMC, SAR, etc.)

Maksym7716, 2025-04-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello Espressif team,

We are currently working on certification of our device for the Ukrainian market, which integrates the ESP32-S3-WROOM-1U-N4 module.

Could you please share the official test reports for the following standards mentioned in the datasheet?

- HTOL (High Temperature Operating Life)
- HTSL (High Temperature Storage Life)
- uHAST (Unbiased Highly Accelerated Stress Test)
- TCT (Temperature Cycling Test)
- ESD (Electrostatic Discharge)
- EMC: EN 301 489-1 and EN 301 489-17
- SAR (if applicable)
- EN 62368-1, RoHS, and REACH compliance reports

The official website contains basic documentation that talks about these test protocols, but there are no documents themselves. Without them we can't move forward, please help.

Best regards!

diplfranzhoepfinger, 2025-04-15:
did you try: 

sales@espressif.com ?

diplfranzhoepfinger, 2025-04-15:
maybe better: 

support@espressif.com 

they are quite reactive.

Maksym7716, 2025-04-15:
> did you try:
> 
> [sales@espressif.com](mailto:sales@espressif.com) ?

Yes, thank you, I sent the first letter there, although not much time has passed, I decided to ask here.

Alvin1Zhang, 2025-04-16:
@Maksym7716 Thanks for reporting, this is out of the scope of ESP-IDF project, will close the ticket and let us track in the mail you sent to sales@espressif.com, thanks.

closed 2025-04-16 by Alvin1Zhang