espressif/esp-idf #15789: ESP32-C3: BLE advertisements from a device with active connection are blocked when controller-based privacy is used

rojer, 2025-04-15:
When ESP32-C3 BLE central is paired, bonded and connected to a device, it no longer hears its advertisements.
This is not the case with the original ESP32 (uses host-based privacy) or ESP32-C6 (uses controller-based privacy but doesn't exhibit the bug).

Specifically, I tracked this down to controller-based RPA resolution: when it is enabled and device's IA (i.e. Public address) is present in the list,
advertisements form this address are inhibited while connected (securely or not).
Note that device itself does not use a random address, advertisements are sent from its identity/public address , yet they are filtered if it is present in the list.

The most specific statement of the problem is: On ESP32-C3, if controller-based RPA resolution is enabled (BLE_HCI_OCF_LE_SET_ADDR_RES_EN = 1, usually enabled during host init) and device's address is present in the RPA resolution list, advertisements from that address are inhibited while connected.

Relevant versions: ESP32-C3 chip revision v0.4, ESP-IDF 5.4.1, BT controller compile version [d74042a], phy_version 1200,2b7123f9,Feb 18 2025,15:22:21

Here's annotated log of a session illustrating it. Apart from the DUT, we have two BLE beacon devices (Shelly BLU Buttons): b0:c7:de:bc:49:a5 (button 1) and 38:39:8f:72:84:22 (button 2)

Initial scan - both buttons are visible in results:

```
[Apr 15 16:33:44.322] shos_rpc_inst.c:243     GATTC.Scan [18989774@shos-1416078] via HTTP_in POST 10.221.243.246:45042
[Apr 15 16:33:44.333] NimBLE                  GAP procedure initiated: discovery; own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=0 duration=9998ms
[Apr 15 16:33:44.344] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000b len=7
[Apr 15 16:33:44.350] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
[Apr 15 16:33:45.339] shos_rpc_service_gat:60 Scan result: 38:39:8f:72:84:22 14 0 0x27818d2a
[Apr 15 16:33:47.010] shos_rpc_service_gat:60 Scan result: b0:c7:de:bc:49:a5 14 0 0x640ec25f
[Apr 15 16:33:54.358] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
```

Establishing a connection to button 2 (no security):

```
[Apr 15 16:34:12.175] shos_rpc_inst.c:243     GATTC.Connect [31078507@shos-1416105] via HTTP_in POST 10.221.243.246:59434
[Apr 15 16:34:12.186] NimBLE                  GAP procedure initiated: stop advertising.
[Apr 15 16:34:12.192] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000a len=1
[Apr 15 16:34:12.202] NimBLE                  GAP procedure initiated: connect; peer_addr_type=0 peer_addr=38:39:8f:72:84:22 scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
[Apr 15 16:34:12.219] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000d len=25
[Apr 15 16:34:12.584] NimBLE                  ble_hs_hci_cmd_send: ogf=0x05 ocf=0x0005 len=2
[Apr 15 16:34:12.590] shos_bt_nimble_gatt:754 38:39:8f:72:84:22: CONNECT c 1 st 0 rssi 0
[Apr 15 16:34:12.596] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
[Apr 15 16:34:12.619] NimBLE                  ble_hs_hci_cmd_send: ogf=0x01 ocf=0x001d len=2
[Apr 15 16:34:12.626] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0022 len=6
[Apr 15 16:34:12.672] shos_rpc_service_ga:257 CONNECT 38:39:8f:72:84:22 ID 3 SEC 0: OK
```

Scan again:

```
[Apr 15 16:34:21.788] shos_rpc_inst.c:243     GATTC.Scan [19379840@shos-1416123] via HTTP_in POST 10.221.243.246:49066
[Apr 15 16:34:21.800] NimBLE                  GAP procedure initiated: discovery; own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=0 duration=9998ms
[Apr 15 16:34:21.811] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000b len=7
[Apr 15 16:34:21.818] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
[Apr 15 16:34:23.873] shos_rpc_service_gat:60 Scan result: 38:39:8f:72:84:22 14 0 0x5d41de4a
[Apr 15 16:34:24.599] shos_rpc_service_gat:60 Scan result: b0:c7:de:bc:49:a5 14 0 0xd5aaca21
[Apr 15 16:34:31.834] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
```

Both are still visible.

Add privacy list entry for button 2 (ble_hs_pvcy_add_entry):

```
[Apr 15 16:34:50.759] shos_rpc_inst.c:243     GATTC.PvcyAddEntry [31099451@shos-1416173] via HTTP_in POST 10.221.243.246:58004
[Apr 15 16:34:50.765] NimBLE                  GAP procedure initiated: stop advertising.
[Apr 15 16:34:50.771] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000a len=1
[Apr 15 16:34:50.777] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0027 len=39  <--- BLE_HCI_OCF_LE_ADD_RESOLV_LIST
[Apr 15 16:34:50.783] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x004e len=8
```

Scan again:

```
[Apr 15 16:35:08.373] shos_rpc_inst.c:243     GATTC.Scan [23142943@shos-1416195] via HTTP_in POST 10.221.243.246:40500
[Apr 15 16:35:08.385] NimBLE                  GAP procedure initiated: discovery; own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=0 duration=9998ms
[... 45 lines cut ...]
Button 2 disappeared.

Now disconnect from it:

```
[Apr 15 16:36:29.226] shos_rpc_inst.c:243     GATTC.Disconnect [28012367@shos-1416367] via HTTP_in POST 10.221.243.246:60108
[Apr 15 16:36:29.232] NimBLE                  GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19
[Apr 15 16:36:29.244] NimBLE                  ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3
[Apr 15 16:36:29.250] shos_bt_nimble_gatt:487 38:39:8f:72:84:22: Forced disconnect (0)
[Apr 15 16:36:29.269] shos_bt_nimble_gatt:767 38:39:8f:72:84:22: DISCONNECT c 1 reason 534 itvl 40 lat 0 svt 256
```

And scan:

```
[Apr 15 16:37:16.852] shos_rpc_inst.c:243     GATTC.Scan [20875043@shos-1416429] via HTTP_in POST 10.221.243.246:36850
[Apr 15 16:37:16.864] NimBLE                  GAP procedure initiated: discovery; own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=0 duration=9998ms
[Apr 15 16:37:16.875] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000b len=7
[Apr 15 16:37:16.882] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
[Apr 15 16:37:19.101] shos_rpc_service_gat:60 Scan result: 38:39:8f:72:84:22 14 0 0x972519c6
[Apr 15 16:37:26.540] shos_rpc_service_gat:60 Scan result: b0:c7:de:bc:49:a5 14 0 0x6fd562c5
[Apr 15 16:37:26.888] NimBLE                  ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
```

Button 2 reappears.

esp-zhp, 2025-04-17:
**Thanks for reporting the issue. I conducted the following tests:**

---

**Test 1**

- Address resolution is enabled.  
- Device 1 uses an RPA (Resolvable Private Address) for advertising.  
- Device 2 performs scanning and connects to Device 1.  
- After a successful connection, pairing and bonding are performed.  
- Device 1 continues advertising using its public address.  
- Device 2 scans again.

**Result:**  
- Device 2 **cannot** discover Device 1.  
- Other devices can still be discovered during scanning.

---

**Test 2**

- Address resolution is enabled.  
- Device 1 uses an RPA for advertising.  
- Device 2 performs scanning and connects to Device 1.  
- After a successful connection, **no** pairing or bonding is performed.  
- Device 1 continues advertising using its public address.  
- Device 2 scans again.

**Result:**  
- Device 2 **can** discover Device 1.  
- Other devices can still be discovered during scanning.

---

I believe I’ve successfully reproduced the issue as described.  
I will work on resolving this as soon as possible. Thank you for your patience.

esp-zhp, 2025-04-18:
@rojer 
Thanks for your patience. I’ve identified the root cause of the issue. To verify whether my fix addresses the same problem you encountered, I’m providing the library below for testing.  
Please check if it resolves your issue.  

**Kindly note:** This modification is not the final version and is intended for testing purposes only.  

Please replace the following file in your project:  
`esp-idf/components/bt/controller/lib_esp32c3_family/esp32c3/libbtdm_app.a`

rojer, 2025-04-18:
@esp-zhp thanks for the quick response but unfortunately i cannot confirm that controller build 2f09e80 fixes the issue, the behavior is still the same in my testing.

esp-zhp, 2025-04-21:
@rojer 
Sorry, I didn’t explain it clearly earlier.  
Please remove the line `r_lld_scan_start = 0x400012ac;` from `esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.bt_funcs.ld` (located at line 628), and use my lib, as the corresponding ROM function will no longer be used.  
My changes will take effect after this modification.

esp-zhp, 2025-04-21:
Looking forward to your further feedback. In the meantime, we will test and evaluate the final revised solution.

rojer, 2025-04-21:
yes, with the `r_lld_scan_start` ROM entry removed, the issue appears to be fixed. thank you!

esp-zhp, 2025-05-19:
Hello,

The issue has been fixed in the master branch. Other branches will be synchronized to GitHub gradually. The fix method is basically consistent with the previously provided library, with only minor modifications; the core changes are exactly the same.

The commit for the fix on master is: 244d4f8fa70036d88102ee05549808df198cf4c5. It will take some time to sync to GitHub. Meanwhile, you can continue to use the previously provided library directly.

Please feel free to reach out if you have further questions.

rojer, 2025-05-19:
ok, sounds good, thank you!
for us, a 5.4 backport is essential, the rest are optional.

esp-zhp, 2025-05-20:
Yes, all branches from v5.1 to v5.5 will be backported. However, v5.4 is currently in the testing phase, so it will take some time.

esp-zhp, 2025-05-28:
Hi,
This issue has been addressed and the fix is already included in **ESP-IDF v5.4**, specifically in commit [`[abb3b2d70c4f33b258395b0e158f211976f55f39](https://github.com/espressif/esp-idf/commit/abb3b2d70c4f33b258395b0e158f211976f55f39)`](https://github.com/espressif/esp-idf/commit/abb3b2d70c4f33b258395b0e158f211976f55f39). The fix ensures that advertisements from a peer device are no longer blocked when controller-based privacy is enabled and the device is actively connected.

Please update to ESP-IDF v5.4 or later to benefit from this fix.

closed 2025-06-30 by Alvin1Zhang