espressif/esp-idf #15792: WiFi Provisioning manager: Protocomm security failure under esp-idf v5.4.1 , ok with older say, v4.4 to v5.4.0

scholayil, 2025-04-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-C3 (QFN32) (version v0.4)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board with ESP32-C3-WROOM-02 module

### Power Supply used.

USB

### What is the expected behavior?

With ESP-IDF v5.4.0 or older, this code worked ok for last 2 years plus. I use BLE to provision WiFi connection, similiar to WiFi provisioning manager example with protocomm security2 used. Normally I connect to the BLE and get WiFi AP list from our IOS app.

### What is the actual behavior?

I do not get WiFi AP List  and gets security2 error when I try to wifi provision.

### Steps to reproduce.

1. Change ESP IDF version from e.g. 5.4.0 to 5.4.1 in VS Code ID.
2. Clean and build. 
3. Run code on device.
4. Try to provision from app.

### Debug Logs.

```plain
(101096) WIRELESS: EventBase: PROTOCOMM_TRANSPORT_BLE_EVENT  ID=0
I (101426) protocomm_nimble: mtu update event; conn_handle=1 cid=4 mtu=256
I (102016) security2: Using salt and verifier to generate public key...
E (103746) security2: Failed at mbedtls_gcm_auth_decrypt : -18

or

I (28614) WIRELESS: EventBase: PROTOCOMM_TRANSPORT_BLE_EVENT  ID=0
I (28924) protocomm_nimble: mtu update event; conn_handle=1 cid=4 mtu=256
I (29504) security2: Using salt and verifier to generate public key...
I (30184) WIRELESS: Received data: IDENTIFY?
E (30184) BLE_get_id: The SN is 1234444FABABC3CC
E (36884) security2: Failed at mbedtls_gcm_auth_decrypt : -18
E (36884) protocomm: Decryption of response failed for endpoint custom-data
E (36884) protocomm_nimble: Invalid content received, killing connection
```

scholayil, 2025-04-16:
Thank you @mahavirj 

I am not able to share code at this time. I compared example code (esp/v5.4.1/esp-idf/components/wifi_provisioning) to my code . They are very similar. 

I tried provisioning the example with 'ESP BLE Prov' IOS app v3.0.3, it provisions ok except that the 'ESP BLE Prov' app does not exercise the custom-data endpoint. Even custom-data works the same way in my code. 

Is there any way you can test the custom-data endpoint with your app with the example in v5.4.21 and update the example if needed? That is where I get the error on my esp device. 

Should I have our app developers look at if they need to update the app libraries?

hrushikesh430, 2025-04-22:
Hello @scholayil  Could you please close the issue, if your issue is clear?

closed 2025-04-22 by scholayil