espressif/esp-idf #15794: ESP32-C6 ULP wakeup cause

janav1, 2025-04-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-master

### Espressif SoC revision.

Chip is ESP32-C6 (QFN40) (revision v0.0)

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

The ulp_lp_core_get_wakeup_cause() function should return real wakeup cause, not return all set interrupt sources. Since I use both GPIO and timer wake-up, I need to distinguish which event happened.

### What is the actual behavior?

calling
`ulp_lp_core_update_wakeup_cause();
uint16_t LPWakeUpReason = ulp_lp_core_get_wakeup_cause();`
should return real wakeup reason (as on HP core), but it is returning all posible wakeup causes which happened. It looks like there has to be a function to clear wakeup cause, so it can be set by the HW acordingly.

### Steps to reproduce.

`#define LED1 2
#define LED2 3
#define INPUT_PIN 7`

HP core:
`
void app_main(void){

    /* Initialize selected GPIO as RTC IO, enable input, disable pullup and pulldown */
    rtc_gpio_init(INPUT_PIN);
    rtc_gpio_set_direction(INPUT_PIN, RTC_GPIO_MODE_INPUT_ONLY);
    rtc_gpio_pulldown_dis(INPUT_PIN);
    rtc_gpio_pullup_en(INPUT_PIN);
    ESP_ERROR_CHECK(rtc_gpio_wakeup_enable(INPUT_PIN, GPIO_INTR_ANYEDGE)); // GPIO_INTR_ANYEDGE
    
    rtc_gpio_init(LED1);
    rtc_gpio_set_direction(LED1, RTC_GPIO_MODE_OUTPUT_ONLY);
    rtc_gpio_pulldown_dis(LED1);
    rtc_gpio_pullup_dis(LED1);
    rtc_gpio_set_level(LED1, 1);

    rtc_gpio_init(LED2);
    rtc_gpio_set_direction(LED2, RTC_GPIO_MODE_OUTPUT_ONLY);
    rtc_gpio_pulldown_dis(LED2);
    rtc_gpio_pullup_dis(LED2);

    esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();
    printf("HP Reset reason: %d\r\n", (int)cause);

    if (cause == ESP_SLEEP_WAKEUP_ULP) {
    	printf("ULP woke up the main CPU! \r\n");
    	printf("Logged ULP WakeUp reason: %ld\r\n", ulp_LPWakeUpReason & 0xFFFF);
    	printf("Pin State: %04X\r\n", (uint16_t)(ulp_PINVal & 0xFFFF));
    }
    else {
    	printf("WakeUp %d, initializing ULP! \r\n", cause);
    	init_ulp_program();
    }

    /* Go back to sleep, only the ULP will run */
    printf("Entering deep sleep\r\n");
    uart_wait_tx_done(UART_NUM_0, 100);
    rtc_gpio_set_level(LED1, 0);
    ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());
    esp_deep_sleep_start();
}
`

LP main.c:
`
volatile uint8_t inputPinState = false;
volatile uint16_t LPWakeUpReason = 0;
volatile uint8_t PINVal = 0xFF;

int main (void)
{
	ulp_lp_core_gpio_set_level(LED2, 1);

	uint32_t NewPinval = ulp_lp_core_gpio_get_level(INPUT_PIN);

	if(PINVal == 0 && NewPinval){
		PINVal = 0xFF;
		inputPinState = true;
	}
	else if(PINVal && NewPinval == 0){
		PINVal = 0;
		inputPinState = false;
	}

	ulp_lp_core_update_wakeup_cause();
	LPWakeUpReason = ulp_lp_core_get_wakeup_cause();
	ulp_lp_core_gpio_clear_intr_status();

	ulp_lp_core_gpio_set_level(LED2, 0);
	ulp_lp_core_wakeup_main_processor();

	return 0;
}
`

### Debug Logs.

```plain
`HP Reset reason: 6
ULP woke up the main CPU! 
Logged ULP WakeUp reason: 16
Pin State: 0000
Entering deep sleep
HP Reset reason: 6
ULP woke up the main CPU! 
Logged ULP WakeUp reason: 16
Pin State: 0000
Entering deep sleep
...
HP Reset reason: 6
ULP woke up the main CPU! 
Logged ULP WakeUp reason: 20
Pin State: 00FF
Entering deep sleep
HP Reset reason: 6
ULP woke up the main CPU! 
Logged ULP WakeUp reason: 20
Pin State: 00FF
Entering deep sleep
...
HP Reset reason: 6
ULP woke up the main CPU! 
Logged ULP WakeUp reason: 20
Pin State: 0000
Entering deep sleep`
```

KonstantinKondrashov, 2025-04-16:
Hi @janav1!
Judging by the description of [ulp_lp_core_get_wakeup_cause](https://github.com/espressif/esp-idf/blob/e9bdd395995f60878e01cece10525987bacc8a5f/components/ulp/lp_core/lp_core/include/ulp_lp_core_utils.h#L19-L30), this is how it should be. This is not a bug, it is intended this way. We will consider adding a new API to get the LP wakeup cause in a way as the HP core does it. I do not know why the bitmap was chosen for this case.

Seems I see an issue, the lp_wakeup_cause var is not reset anywhere so it constantly accumulates the wakeup cause bits.

https://github.com/espressif/esp-idf/blob/e9bdd395995f60878e01cece10525987bacc8a5f/components/ulp/lp_core/lp_core/include/ulp_lp_core_utils.h#L19-L30

KonstantinKondrashov, 2025-04-16:
@janav1 You can try to reset the `lp_wakeup_cause` var at the start of the `ulp_lp_core_update_wakeup_cause()` functions. 
lp_wakeup_cause  = 0;

https://github.com/espressif/esp-idf/blob/e9bdd395995f60878e01cece10525987bacc8a5f/components/ulp/lp_core/lp_core/lp_core_utils.c#L43-L45

janav1, 2025-04-17:
@KonstantinKondrashov 
Thank you for the clarification, I tryed your sugestion, but now the ULP wakeup reason is zero all the time.
It would be great to have the same functionality for both processors.
Is there another way to verify whether the LP core was woken up by a timer or a GPIO interrupt?
As you can see from my example above, I am using both methods and need to distinguish which one occurred.

KonstantinKondrashov, 2025-04-17:
@janav1!
`ulp_lp_core_update_wakeup_cause()` is called at the start of the LP core (from `lp_core_startup()`).
https://github.com/espressif/esp-idf/blob/e9bdd395995f60878e01cece10525987bacc8a5f/components/ulp/lp_core/lp_core/lp_core_startup.c#L25

Calling it from the user code clears the wakeup reason. So, please remove `ulp_lp_core_update_wakeup_cause()` from your code. It should show the wakeup_cause correctly.

KonstantinKondrashov, 2025-04-18:
@janav1 Can you confirm that it ^^ works?

closed 2025-04-21 by espressif-bot

X-Ryl669, 2025-04-27:
I've tried your code and it works for me, albeit with the requirement of removing the call to `ulp_lp_core_update_wakeup_cause` from my code and the issue that interrupt going in on IO (one of the two selected wakeup source) while the LP core is running is "remembered" and triggers automatically when halting. See #15851 for details about this issue.